openwrt/asu #575: luci-app-attendedsysupgrade/auc: custom uci-defaults support

bam80, 2023-09-16:
`uci-defaults` scripts are essential for accessing device after firmware reset.

If user defined a custom UCI-Defaults script with [OpenWrt Firmware Selector](https://gitlab.com/openwrt/web/firmware-selector-openwrt-org#uci-defaults), it's feasible to preserve it upon firmware upgrade, at least optionally, and include it to new image.
_Otherwise, access to the device can be lost next time user performs firmware Reset._

As a simple approach to solve this, `luci-app-attendedsysupgrade/auc` utilities could check if `/rom/etc/uci-defaults/99-asu-defaults` file exists in current firmware and if so, include it to a new image:
https://github.com/openwrt/asu/blob/3ef4757b9fcc401b3d18ebb6954276b0210f07fb/asu/build.py#L202

@aparcar @mwarning

mwarning, 2023-09-17:
This is not really my area. ._.

bam80, 2023-09-17:
Please let me know if I need to report it separately for luci-app-attendedsysupgrade/auc utils.
Also, I would like to know if you agree with the idea overall.

efahl, 2023-09-17:
What happens if you simply add that file path to `/etc/sysupgrade.conf`?

bam80, 2023-09-17:
What path do you mean?
If it's in /rom, I doubt sysupgrade can handle it because it's r/o.
If you mean /etc/uci-defaults/, there are no more files there already because they are wiped on first boot.

bam80, 2023-09-17:
And again - it wouldn't help for the firmware reset case which is a main concern.

bam80, 2024-05-03:
@CTCD 
> And again - it wouldn't help for the firmware reset case which is a main concern.

I mean full reset here, when rw partition is formatted (the one triggered from LuCI works differently).

bam80, 2024-05-05:
@CTCD any solution we could merge for upstream?

JWvdVeer, 2026-11-10:
Any progress on this subject?

Since it's not only about defaults. Its also about other stuff you want to manage after each upgrade. I've got for example this script:
```
#!/bin/sh

sInject=$(
cat << 'INJECT'

# DUAL-RESET-START
if [ "$ACTION" = "pressed" ]; then date +"%s" > /tmp/reset_${BUTTON}; fi
if [ "$ACTION" = "released" ]; then
    rm -f "/tmp/reset_${BUTTON}"
    { [[ "$BUTTON" == "wlan" ]] && BUTTON="wps"; } || BUTTON="wlan"
    if [[ "$SEEN" -ge 15 ]] && [[ -f /tmp/reset_${BUTTON} ]] && [[ $(( $( date +"%s" ) - $( cat /tmp/reset_${BUTTON} ) )) -ge 15 ]]; then
        logger -t button "WPS + WiFi held > 15s. Factory Resetting!"
        firstboot -y && reboot &
        exit 0
    fi
fi
# DUAL-RESET-END

INJECT
)

for sAction in wlan wps; do
    echo "$sAction"
    sFile="/etc/rc.button/${sAction}"
    echo "$sFile"
    [[ ! -f "$sFile" ]] && echo -e "#!/bin/sh\n" > "$sFile"
    sed -i '/# DUAL-RESET-START/,/# DUAL-RESET-END/d' "$sFile"
    head -n 1 "$sFile" > "/tmp/new_$sAction"
    echo "$sInject" >> "/tmp/new_$sAction"
    tail -n +2 "$sFile" >> "/tmp/new_$sAction"
    chmod +x "/tmp/new_$sAction"
    mv "/tmp/new_$sAction" "$sFile"
done
```

This script manages the reset of a router, of which someone screwed up the reset button (and I actually prefer this method on the other routers that still have a functioning reset button, since the wlan/wps-buttons are easier accessible than the reset).

In a normal reset the script is still preserved. But in case of upgrade using the luci-environment, the script has been vanished afterwards.

In order to be as safe as possible, I would consider to first download the sysupgrade.itb. Then check whether the file `/rom/etc/uci-defaults/99-asu-defaults` exists. If so, unpack the itb-file, inject the uci-script and repack the itb and than do the upgrade.