openwrt/asu #579: Can’t build “auc” package with OpenWrt SDK: no json library found

bam80, 2023-09-23:
I'm trying to build `auc` package in OpenWrt SDK:
https://openwrt.org/packages/pkgdata/auc

Steps I'm doing:

* run container from latest OpenWrt SDK Docker image:
https://hub.docker.com/r/openwrt/sdk
* In the container, issue the commands:
  ```
  ./scripts/feeds update packages
  ./scripts/feeds install -a -p packages

  make package/auc/compile V=s
  # it runs "make menuconfig", exit with saving default configuration to continue
  ...
  ```
* got a CMake `json` error:
  ```
  ...
  CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
  Please set them or make sure they are set and tested correctly in the CMake files:
  json
      linked by target "auc" in directory /builder/build_dir/target-x86_64_musl/auc-0.3.2
  ```

dangowrt, 2023-09-23:
> ./scripts/feeds install -a -p packages

libjson-c is part of the core packages (which, if using the SDK, are also in a feed), so better just do
```
./scripts/feeds update -a
./scripts/feeds install -a
```
instead to make sure all dependencies are present.

bam80, 2023-09-23:
Thanks @dangowrt !
now I see it actually mentioned in the SDK usage Wiki:
https://openwrt.org/docs/guide-developer/toolchain/using_the_sdk#load_package_lists

closed 2023-09-23 by bam80

bam80, 2023-09-23:
The only issue it seem to pulls building of all kernel modules packages too and other unnecessary things, but at least it builds in the end.

dangowrt, 2023-09-23:
What I usually do (assumming freshly unpacked SDK):
```
./scripts/feeds update -a
./scripts/feeds install $mypackage
```
That will install only $mypackage *and all it's dependencies*.

bam80, 2023-09-23:
Thanks, `auc` (or it's dependencies) still pulls linux kernel but it's out of scope of this issue:
```
./scripts/feeds update -a
./scripts/feeds install auc

make package/auc/compile

 make[1] package/auc/compile
 make[2] -C package/toolchain clean-build
 make[2] -C package/toolchain compile
 make[2] -C feeds/base/package/libs/libjson-c clean-build
 make[2] -C feeds/base/package/libs/libjson-c compile
 make[2] -C feeds/base/package/utils/lua clean-build
 make[2] -C feeds/base/package/utils/lua compile
 make[2] -C feeds/base/package/libs/libubox clean-build
 make[2] -C feeds/base/package/libs/libubox compile
 make[2] -C feeds/base/package/libs/libnl-tiny clean-build
 make[2] -C feeds/base/package/libs/libnl-tiny compile
 make[2] -C feeds/base/package/system/ubus clean-build
 make[2] -C feeds/base/package/system/ubus compile
 make[2] -C feeds/base/package/system/uci clean-build
 make[2] -C feeds/base/package/system/uci compile
 make[2] -C feeds/base/package/network/utils/iwinfo clean-build
 make[2] -C feeds/base/package/network/utils/iwinfo compile
 make[2] -C feeds/base/package/utils/ucode clean-build
 make[2] -C feeds/base/package/utils/ucode compile
 make[2] -C feeds/base/package/system/rpcd clean-build
 make[2] -C feeds/base/package/system/rpcd compile
 make[2] -C feeds/base/package/libs/mbedtls clean-build
 make[2] -C feeds/base/package/libs/mbedtls compile
 make[2] -C package/kernel/linux clean-build
 make[2] -C package/kernel/linux compile
 make[2] -C feeds/base/package/kernel/cryptodev-linux clean-build
 make[2] -C feeds/base/package/kernel/cryptodev-linux compile
 make[2] -C feeds/base/package/libs/openssl clean-build
 make[2] -C feeds/base/package/libs/openssl compile
 make[2] -C feeds/base/package/libs/libtool clean-build
 make[2] -C feeds/base/package/libs/libtool compile
 make[2] -C feeds/base/package/libs/wolfssl clean-build
 make[2] -C feeds/base/package/libs/wolfssl compile
 make[2] -C feeds/base/package/libs/wolfssl clean-build
 make[2] -C feeds/base/package/libs/wolfssl compile
 make[2] -C feeds/base/package/libs/ustream-ssl clean-build
 make[2] -C feeds/base/package/libs/ustream-ssl compile
 make[2] -C feeds/base/package/libs/ustream-ssl clean-build
 make[2] -C feeds/base/package/libs/ustream-ssl compile
 make[2] -C feeds/base/package/libs/ustream-ssl clean-build
 make[2] -C feeds/base/package/libs/ustream-ssl compile
 make[2] -C feeds/base/package/libs/uclient clean-build
 make[2] -C feeds/base/package/libs/uclient compile
 make[2] -C feeds/packages/utils/attendedsysupgrade-common clean-build
 make[2] -C feeds/packages/utils/attendedsysupgrade-common compile
 make[2] -C feeds/packages/utils/auc clean-build
 make[2] -C feeds/packages/utils/auc compile