openwrt/asu #581: AttendedSysUpgrade - CORS Access-Control-Allow-Origin error?

robert-pajak, 2023-09-25:
I was using `AttendedSysUpgrade` feature for some time and I was happy with it.
Today it stopped working for me in Chrome and Edge browsers

The symptoms is: after pressing the green button "Search for firmware upgrade" I can see a white pop-up only, but it never proceeds to the next phase
[image]

After digging a bit, it seems like this issue is caused by the server returning multiple headers with the same name (Access-Control-Allow-Origin) but different values where only one value is allowed
[image]
 
 Could this be fixed please?

aparcar, 2023-09-25:
[image]

can't reproduce here sorry

robert-pajak, 2023-09-25:
Thanks for checking it...
Is it possible that SNAPSHOT and stable releases have different response headers from the server?

aparcar, 2023-09-25:
To me it looks like you're trying to open wrong URLs, so it's not a server issue but you're opening files which aren't there. I.e. `/api/v1/revision/S`. Not sure what could cause this.

robert-pajak, 2023-09-25:
Not really ... 
[image]
And if I open the URL that is erroring from within the LUCI interface (Luci web application) it does work
[image]

KevinMX, 2023-09-26:
Saw the same error message on [OpenWrt Firmware Selector](https://firmware-selector.openwrt.org/).

[image]

aparcar, 2023-09-26:
Curious, turns out both flask and caddy server would add a CORS header which is then rejected by web browsers. I disabled it on the flask side and now it works.

robert-pajak, 2023-09-26:
No idea what flask and caddy servers are :) 
But I can confirm now it works for me - Thanks!

szero, 2023-09-26:
was about to post about this aswell, very strange behavior because cors header was properly set to asterisk and it was still failing, the 302 redirect says it failed in the browser but it actually completed and resulted in failing on overview.json, some race condition between those two servers than

aparcar, 2023-09-26:
I added the CORS feature via https://github.com/openwrt/asu/commit/5481f9c9cb7bfc914da3292710d180e7dd244d7f and then it would duplicate, issue is that the web server hosts some files statically which have then missing CORS. Anyway, works now :)