openwrt/asu #595: Error building the firmware image Server response: Error: No JSON file found

hguresci, 2023-10-02:
"url": "https://sysupgrade.openwrt.org",
    "revision": "r16847-f8282da11e",
    "advanced_mode": "1",
    "branch": "21.02",
    "efi": null,
    "profile": "xiaomi,mi-router-4a-gigabit",
    "target": "ramips/mt7621",
    "version": "22.03.5",
    "packages": [
        "attendedsysupgrade-common",
        "base-files",
        "bind-host",
        "bind-libs",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dcwapd",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fping",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables",
        "iptables",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-mt7603",
        "kmod-mt76x2",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libc",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libopenssl",
        "libubox",
        "libubus",
[... 865 lines cut ...]
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
/builder/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/ntpdate: line 7: uci: command not found
./etc/init.d/ntpdate: line 7: uci: command not found
Pseudo file "/dev" exists in source filesystem "/builder/build_dir/target-mipsel_24kc_musl/root-ramips/dev".
Ignoring, exclude it (-e/-ef) to override.
5281+1 records in
5281+1 records out
2704084 bytes (2.7 MB, 2.6 MiB) copied, 0.0256195 s, 106 MB/s
25720+1 records in
25720+1 records out
13169054 bytes (13 MB, 13 MiB) copied, 0.0927085 s, 142 MB/s
padding image to 00f24000
padding image to 00f30000
padding image to 00f40000
bash: 8: Bad file descriptor
Failed to open firmware file
sha256sum: /builder/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-22.03.5-48ae00b3aa49-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin: No such file or directory
cp: cannot stat '/builder/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/tmp/openwrt-22.03.5-48ae00b3aa49-ramips-mt7621-xiaomi_mi-router-4a-gigabit-squashfs-sysupgrade.bin': No such file or directory

reukiodo, 2024-01-28:
[U35WF](http://u35wf.kirk11704.reukio.do/)

    [Status](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [Overview](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/status/overview)
        [Routing](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/status/routes)
        [Firewall](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/status/nftables)
        [System Log](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/status/logs)
        [Processes](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/status/processes)
        [Channel Analysis](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/status/channel_analysis)
        [Realtime Graphs](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/status/realtime)
    [System](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [System](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/system)
        [Administration](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/admin)
        [Software](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/opkg)
        [Startup](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/startup)
        [Scheduled Tasks](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/crontab)
        [Mount Points](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/mounts)
        [Attended Sysupgrade](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/attendedsysupgrade)
        [LED Configuration](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/leds)
        [Backup / Flash Firmware](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/flash)
        [Reboot](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/reboot)
    [Services](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [Transmission](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/services/transmission)
    [Network](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [Interfaces](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/network/network)
        [Wireless](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/network/wireless)
        [Switch](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/network/switch)
        [Routing](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/network/routes)
        [DHCP and DNS](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/network/dhcp)
        [Diagnostics](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/network/diagnostics)
        [Firewall](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/network/firewall)
    [Log out](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/logout)

Paused

    [Overview](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/attendedsysupgrade/overview)
    [Configuration](http://u35wf.kirk11704.reukio.do/cgi-bin/luci/admin/system/attendedsysupgrade/configuration)

Attended Sysupgrade

The attended sysupgrade service allows to easily upgrade vanilla and custom firmware images.

This is done by building a new firmware on demand via an online service.

Currently running: 23.05.2 - r23630-842932a63d
Powered by[ LuCI openwrt-23.05 branch (git-24.006.68745-9128656)](https://github.com/openwrt/luci) /[ OpenWrt 23.05.2 (r23630-842932a63d)](https://openwrt.org/)
Error building the firmware image

Server response: Error: No JSON file found
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23630-842932a63d",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "request_hash": "d4abfe5e67d35e7853d4338ebb5708b4",
    "profile": "kimax,u35wf",
    "target": "ramips/mt7620",
    "version": "23.05.2",
    "packages": [
        "base-files",
        "block-mount",
        "btop",
        "busybox",
        "ca-bundle",
[... 1000 lines cut ...]
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-mipsel_24kc_musl/root-ramips/dev".
Ignoring, exclude it (-e/-ef) to override.
4481+1 records in
4481+1 records out
2294345 bytes (2.3 MB, 2.2 MiB) copied, 0.013536 s, 169 MB/s
29627+1 records in
29627+1 records out
15169108 bytes (15 MB, 14 MiB) copied, 0.0908942 s, 167 MB/s
padding image to 010b0000
bash: line 1: 8: Bad file descriptor
Failed to open firmware file
sha256sum: /builder/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-23.05.2-03e6c0d5b704-ramips-mt7620-kimax_u35wf-squashfs-sysupgrade.bin: No such file or directory
cp: cannot stat '/builder/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/tmp/openwrt-23.05.2-03e6c0d5b704-ramips-mt7620-kimax_u35wf-squashfs-sysupgrade.bin': No such file or directory

Close

reukiodo, 2024-01-28:
seems that both issues occur with padding the file?
```
13169054 bytes (13 MB, 13 MiB) copied, 0.0927085 s, 142 MB/s
padding image to 00f24000
padding image to 00f30000
padding image to 00f40000
bash: 8: Bad file descriptor

padding image to 010b0000
bash: line 1: 8: Bad file descriptor
```