openwrt/asu #596: Bad message (77)

ShredRum, 2023-10-03:
Xiaomi Mi Router 4A (100M International Edition)
ramips/mt76x8
OpenWrt SNAPSHOT r24054-fe10f97439 / LuCI Master git-23.266.27574-7744ad0
5.15.132

I'm trying to update my router through the auc command and I get the following error:

root@OpenWrt:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r24054-fe10f97439 on ramips/mt76x8 (xiaomi,mi-router-4a-100m-intl)
Available: SNAPSHOT r24067-c354c069b3
Requesting package lists...
 libopenssl: 3.0.11-1 -> 3.0.10-1
 luci-app-opkg: git-23.267.29502-3e7fe9f -> git-23.009.82915-ec3aac4
 base-files: 1542-r24054-fe10f97439 -> 1535-r24002-e3559fb445
 netifd: 2023-09-19-7a58b995-2 -> 2023-09-19-7a58b995-1
 dnsmasq: 2.89-6 -> 2.89-4
 luci-i18n-base-ru: git-23.270.68722-cbe0748 -> git-23.263.30531-f59a194
 luci-app-firewall: git-23.267.29502-3e7fe9f -> git-23.212.18089-5d1e581
 luci-base: git-23.266.27574-7744ad0 -> git-23.236.53281-98e3743
 luci-app-attendedsysupgrade: git-23.265.60357-d7e905e -> git-23.223.40904-987ae1a
 luci-mod-network: git-23.264.73274-598b1eb -> git-23.233.29374-a6a428f
Are you sure you want to continue the upgrade process? [N/y] y
Requesting build...................................aircrack-ng - 1.6-3
airmon-ng - 1.6-3
attendedsysupgrade-common - 8
auc - 0.3.2-1
base-files - 1542-r24067-c354c069b3
busybox - 1.36.1-1
ca-bundle - 20230311-1
cgi-io - 2022-08-10-901b0f04-21
dnsmasq - 2.89-6
dropbear - 2022.82-5
ethtool - 6.4-1
firewall4 - 2023-09-01-598d9fbb-1
fstools - 2023-02-28-bfe882d5-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2022-08-13-4c7b720b-2
hostapd-common - 2023-09-08-e5ccbfc6-4
htop - 3.2.2-1
iw - 5.19-1
iwinfo - 2023-07-01-ca79f641-1
jansson - 2.14-3
jshn - 2023-05-23-75a3b870-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.15.132-1-993c4b695a7624047a1e8e8d6bad2b25
kmod-cfg80211 - 5.15.132+6.5-1
kmod-crypto-acompress - 5.15.132-1
kmod-crypto-aead - 5.15.132-1
kmod-crypto-ccm - 5.15.132-1
kmod-crypto-cmac - 5.15.132-1
kmod-crypto-crc32c - 5.15.132-1
kmod-crypto-ctr - 5.15.132-1
kmod-crypto-gcm - 5.15.132-1
kmod-crypto-gf128 - 5.15.132-1
kmod-crypto-ghash - 5.15.132-1
kmod-crypto-hash - 5.15.132-1
kmod-crypto-hmac - 5.15.132-1
kmod-crypto-manager - 5.15.132-1
kmod-crypto-null - 5.15.132-1
kmod-crypto-rng - 5.15.132-1
kmod-crypto-seqiv - 5.15.132-1
kmod-crypto-sha512 - 5.15.132-1
kmod-gpio-button-hotplug - 5.15.132-3
kmod-leds-gpio - 5.15.132-1
kmod-lib-crc-ccitt - 5.15.132-1
kmod-lib-crc32c - 5.15.132-1
kmod-lib-lzo - 5.15.132-1
kmod-mac80211 - 5.15.132+6.5-1
[... 137 lines cut ...]
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_kmods https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/kmods/5.15.132-1-993c4b695a7624047a1e8e8d6bad2b25). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_telephony https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony). Skipping.
Error: Impossible package selection: wpad-basic-mbedtls version not as requested: 2023-09-08-e5ccbfc6-3 vs. 2023-09-08-e5ccbfc6-4
Bad message (77)
root@OpenWrt:~#

aparcar, 2023-10-16:
Please try again and close this issue if it works now.

opty77, 2024-07-28:
```
root@tplink:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.3 r23809-234f1a2efa on ath79/generic (tplink,tl-wdr3600-v1)
Available: 23.05.4 r24012-d8dd03c46f
Requesting package lists...
[...]
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://downloads.openwrt.org/releases/23.05.4/targets/ath79/generic/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://downloads.openwrt.org/releases/23.05.4/packages/mips_24kc/base). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://downloads.openwrt.org/releases/23.05.4/packages/mips_24kc/luci). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://downloads.openwrt.org/releases/23.05.4/packages/mips_24kc/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://downloads.openwrt.org/releases/23.05.4/packages/mips_24kc/routing). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_telephony https://downloads.openwrt.org/releases/23.05.4/packages/mips_24kc/telephony). Skipping.
Error: Impossible package selection: liblua version not as requested: 5.3.5-5 vs. 5.1.5-11
Bad message (77)
```

opty77, 2024-08-07:
Succesfully upgraded yesterday.

@ShredRum: Ping.