openwrt/asu #599: Broken LUCi web inteface

robert-pajak, 2023-10-04:
When I tried to perform an Attended Sys Upgrade using Luci, after I pressed the green button for `requesting firmware image` I received the following error
[image]
It was working up to yesterday (or the day before yesterday) but now it causes an error
Could this be fixed please?

robert-pajak, 2023-10-05:
Thanks for checking @aparcar aparcar, but I can confirm the issue is still there
Looks like luci-attended-sysupgrade is performing a call to the https://sysupgrade.openwrt.org/api/v1/build?1696506154265 with the request payload
```
{
  "profile": "xiaomi,ax9000",
  "target": "qualcommax/ipq807x",
  "version": "SNAPSHOT",
  "packages": [
    "ath10k-firmware-qca9887-ct-full-htt",
    "ath11k-firmware-ipq8074",
    "ath11k-firmware-qcn9074",
    "auc",
    "base-files",
    "busybox",
    "ca-bundle",
    "collectd-mod-contextswitch",
    "collectd-mod-cpufreq",
    "collectd-mod-curl",
    "collectd-mod-dhcpleases",
    "collectd-mod-entropy",
    "collectd-mod-filecount",
    "collectd-mod-fscache",
    "collectd-mod-irq",
    "collectd-mod-modbus",
    "collectd-mod-ntpd",
    "collectd-mod-ping",
    "collectd-mod-processes",
    "collectd-mod-tcpconns",
    "ddns-scripts-route53",
    "dnsmasq",
    "dropbear",
    "e2fsprogs",
    "ethtool",
    "firewall4",
    "fstools",
    "fwtool",
    "getrandom",
    "hostapd-common",
    "htop",
    "inotifywait",
    "inotifywatch",
    "iperf3",
    "ipq-wifi-xiaomi_ax9000",
    "iptraf-ng",
    "iw",
    "iwinfo",
    "jansson",
    "jshn",
    "jsonfilter",
    "kernel",
    "kmod-ath",
    "kmod-ath10k-ct",
    "kmod-ath11k",
    "kmod-ath11k-ahb",
    "kmod-ath11k-pci",
    "kmod-cfg80211",
    "kmod-crypto-acompress",
    "kmod-crypto-aead",
    "kmod-crypto-ccm",
    "kmod-crypto-cmac",
    "kmod-crypto-crc32c",
    "kmod-crypto-ctr",
    "kmod-crypto-gcm",
    "kmod-crypto-gf128",
    "kmod-crypto-ghash",
    "kmod-crypto-hash",
    "kmod-crypto-hmac",
    "kmod-crypto-manager",
    "kmod-crypto-michael-mic",
    "kmod-crypto-null",
[... 934 lines cut ...]
    ],
    "request_hash": "d4a917af078e0bbfe12448e43a35c6ce",
    "target": "qualcommax/ipq807x",
    "version": "SNAPSHOT"
  },
  "request_hash": "d4a917af078e0bbfe12448e43a35c6ce",
  "source_date_epoch": 1696277590,
  "status": 200,
  "stderr": "Generate local signing keys...\nGenerate local certificate...\nPackage list missing or not up-to-date, generating it.\n\nBuilding package index...\nDownloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/Packages.gz\nUpdated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_core\nDownloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/Packages.sig\nSignature check passed.\nDownloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz\nUpdated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_base\nDownloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.sig\nSignature check passed.\nDownloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.1.55-1-0567810dfd9a1152272622ddb29d7220/Packages.gz\nUpdated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_kmods\nDownloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.1.55-1-0567810dfd9a1152272622ddb29d7220/Packages.sig\nSignature check passed.\nDownloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz\nUpdated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_luci\nDownloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.sig\nSignature check passed.\nDownloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz\nUpdated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_packages\nDownloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig\nSignature check passed.\nDownloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz\nUpdated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_routing\nDownloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig\nSignature check passed.\nDownloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz\nUpdated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_telephony\nDownloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig\nSignature check passed.\nDownloading file:packages/Packages\nUpdated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/imagebuilder\nDownloading file:packages/Packages.sig\nSignature check passed.\n/builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/etc/init.d/dnsmasq: line 4: /lib/functions.sh: No such file or directory\n/builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/etc/init.d/usteer: line 10: /lib/functions/network.sh: No such file or directory\n/builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/etc/init.d/usteer: line 11: /usr/share/libubox/jshn.sh: No such file or directory\n/builder/build_d [... cut ...] a53_musl/linux-qualcommax_ipq807x/tmp/openwrt-6e430445aa52-qualcommax-ipq807x-xiaomi_ax9000-initramfs-factory.ubi.tmp\"12627892 bytes (12.0 MiB)\n\nCalculating checksums...\n",
  "supported_devices": [
    "xiaomi,ax9000"
  ],
  "target": "qualcommax/ipq807x",
  "titles": [
    {
      "model": "AX9000",
      "vendor": "Xiaomi"
    }
  ],
  "version_code": "r24072-a181b9f0f9",
  "version_number": "SNAPSHOT"
}
```
That could explain why it works for some people as this depends on the device type

edrikk, 2023-10-05:
I noticed that @robert-pajak device was also an ipq807 device.  
So I wanted to add that I'm seeing similar behavior on mediatek as well:

> auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r24072-a181b9f0f9 on mediatek/mt7622 (xiaomi,redmi-router-ax6s)
Available: SNAPSHOT r24071-b7eea2db73
Requesting package lists...
 libopenssl: 3.0.11-1 -> 3.0.10-1
 luci-app-opkg: git-23.267.29502-3e7fe9f -> git-23.009.82915-ec3aac4
 base-files: 1542-r24072-a181b9f0f9 -> 1535-r24002-e3559fb445
 netifd: 2023-09-19-7a58b995-2 -> 2023-09-19-7a58b995-1
 libopenssl-legacy: 3.0.11-1 -> 3.0.10-1
 luci-app-firewall: git-23.267.29502-3e7fe9f -> git-23.212.18089-5d1e581
 wpad-openssl: 2023-09-08-e5ccbfc6-4 -> 2023-09-08-e5ccbfc6-3
 luci-app-dawn: git-23.267.29818-4c03f5f -> git-23.074.48419-2221110
 luci-base: git-23.266.27574-7744ad0 -> git-23.236.53281-98e3743
 libopenssl-conf: 3.0.11-1 -> 3.0.10-1
 hostapd-common: 2023-09-08-e5ccbfc6-4 -> 2023-09-08-e5ccbfc6-3
 dnsmasq-full: 2.89-6 -> 2.89-4

robert-pajak, 2023-10-05:
@edrikk wrote:
```
@aparcar after your comment I tried again from my ipq807 device. I found that the luci app this time responded and started to build the firmware. BUT, after about 5 minutes of it being "stuck" at 80% (generating image) I gave up (usually it takes a minute).
```
It happened to me as well, but it was very sporadic, I'd say maybe twice for 300 attempts?

aparcar, 2023-10-05:
So the LuCI app works fine for me
[image]

The CLI is currently broken due to missing JSON files on the server, I'm looking into that. There will be a full rewrite of the CLI so I'm currently a bit slow on implementing it. Sorry for the wait.

robert-pajak, 2023-10-05:
Thanks @aparcar for looking into this.
I think the existing mechanism is target/device specific and if something goes wrong on the server side (i.e. for ipq807x target) than all people who want's to use Attended Sys Upgrade on ipq807x won't be able to perform sys upgrade

aparcar, 2023-10-06:
I'm still in the dark why this issue happens. The code tries to find a suitable image by selecting whatever image has the correct filesystem (squashfs) and is either called *sysupgrade* or *combined*. This specific image exists and is shown in your posted request.

robert-pajak, 2023-10-06:
I'd recommend investigating and comparing the JSON structure you are getting with the one I attached.
It seems like yours has a property `name` which mine seems to miss. 
Because the responses for ipq807x and ramips are missing the expected property/attribute `name` the javascript code fails which triggers console error and `nothing happens after the green button is being pressed`.

khanjui, 2023-10-07:
` OpenWrt SNAPSHOT, r24002-e3559fb445
 -----------------------------------------------------
root@OpenWrt:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r24002-e3559fb445 on ramips/mt7621 (xiaomi,mi-router-4a-gigabit)
Available: SNAPSHOT r24071-b7eea2db73
Requesting package lists...
 luci-app-attendedsysupgrade: git-23.265.60357-d7e905e -> git-23.223.40904-987ae1a
 luci-mod-network: git-23.264.73274-598b1eb -> git-23.233.29374-a6a428f
Are you sure you want to continue the upgrade process? [N/y] y
Requesting build.........Current Target: "ramips/mt7621"
Current Architecture: "mipsel"
Current Revision: "r24084-3aef61060e"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-mbedtls logd mtd netifd opkg uci uclient-fetch urandom-seed urngd busybox procd procd-ujail procd-seccomp kmod-leds-gpio kmod-gpio-button-hotplug wpad-basic-mbedtls uboot-envtools kmod-crypto-hw-eip93 dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe
Available Profiles:

adslr_g7:
    ADSLR G7
    Packages: kmod-mt7615-firmware -uboot-envtools
    hasImageMetadata: 1
    SupportedDevices: adslr,g7
afoundry_ew1200:
    AFOUNDRY EW1200
    Packages: kmod-ata-ahci kmod-mt76x2 kmod-mt7603 kmod-usb3 kmod-usb-ledtrig-usbport -uboot-envtools
    hasImageMetadata: 1
    SupportedDevices: afoundry,ew1200 ew1200
alfa-network_ax1800rm:
    ALFA Network AX1800RM
    Packages: kmod-mt7915-firmware
    hasImageMetadata: 1
    SupportedDevices: alfa-network,ax1800rm
alfa-network_quad-e4g:
    ALFA Network Quad-E4G
    Packages: kmod-ata-ahci kmod-sdhci-mt7620 kmod-usb3 -wpad-basic-mbedtls
    hasImageMetadata: 1
    SupportedDevices: alfa-network,quad-e4g quad-e4g
ampedwireless_ally-r1900k:
    Amped Wireless ALLY-R1900K
    Packages: kmod-mt7615-firmware kmod-usb3
    hasImageMetadata: 1
    SupportedDevices: ampedwireless,ally-r1900k
ampedwireless_ally-00x19k:
    Amped Wireless ALLY-00X19K
    Packages: kmod-mt7615-firmware
    hasImageMetadata: 1
    SupportedDevices: ampedwireless,ally-00x19k
arcadyan_we420223-99:
    Arcadyan WE420223-99
    Packages: kmod-mt7615-firmware -uboot-envtools
    hasImageMetadata: 1
    SupportedDevices: arcadyan,we420223-99
asiarf_ap7621-001:
    AsiaRF AP7621-001
    Packages: kmod-sdhci-mt7620 kmod-mt76x2 kmod-usb3 -wpad-basic-mbedtls -uboot-envtools
    hasImageMetadata: 1
    SupportedDevices: asiarf,ap7621-001
asiarf_ap7621-nv1:
    AsiaRF AP7621-NV1
    Packages: kmod-sdhci-mt7620 kmod-mt76x2 kmod-usb3 -wpad-basic-mbedtls -uboot-envtools
    hasImageMetadata: 1
    SupportedDevices: asiarf,ap7621-nv1
asus_rp-ac56:
    ASUS RP-AC56
    Packages: kmod-mt7603 kmod-mt76x2 kmod-i2c-ralink kmod-sound-mt7620 -uboot-envtools
    hasImageMetadata: 1
    SupportedDevices: asus,rp-ac56
asus_rp-ac87:
    ASUS RP-AC87
    Packages: kmod-mt7615-firmware rssileds -uboot-envtools
[... 1089 lines cut ...]
    Packages: kmod-mt7603 kmod-usb3 kmod-usb-net-qmi-wwan kmod-usb-serial-option uqmi
    hasImageMetadata: 1
    SupportedDevices: zyxel,nr7101
zyxel_nwa50ax:
    ZyXEL NWA50AX
    Packages: kmod-mt7915-firmware zyxel-bootconfig
    hasImageMetadata: 1
    SupportedDevices: zyxel,nwa50ax
zyxel_nwa55axe:
    ZyXEL NWA55AXE
    Packages: kmod-mt7915-firmware zyxel-bootconfig
    hasImageMetadata: 1
    SupportedDevices: zyxel,nwa55axe
zyxel_wap6805:
    ZyXEL WAP6805
    Packages: kmod-mt7603 kmod-mt7621-qtn-rgmii -uboot-envtools
    hasImageMetadata: 1
    SupportedDevices: zyxel,wap6805
zyxel_wsm20:
    ZyXEL WSM20
    Packages: kmod-mt7915-firmware
    hasImageMetadata: 1
    SupportedDevices: zyxel,wsm20
Error: Received inncorrect version r24084-3aef61060e (requested r24071-b7eea2db73)
Bad message (77)`

edrikk, 2023-10-08:
Just FYI - a Raspberry Pi Model B (ARMv6-compatible processor rev 7 (v6l)  ---> bcm27xx/bcm2708) that I have is showing the same behavior.

I just tried a sysupgrade and no go...  Clicking "Request firmware image" does nothing.

> Uncaught (in promise) TypeError: image is null
    handle200 http://192.168.86.10/luci-static/resources/view/attendedsysupgrade/overview.js?v=git-23.236.53281-98e3743:4
    handleRequest http://192.168.86.10/luci-static/resources/view/attendedsysupgrade/overview.js?v=git-23.236.53281-98e3743:13
    promise callback*handleRequest http://192.168.86.10/luci-static/resources/view/attendedsysupgrade/overview.js?v=git-23.236.53281-98e3743:10
    pollFn http://192.168.86.10/luci-static/resources/view/attendedsysupgrade/overview.js?v=git-23.236.53281-98e3743:25
    step http://192.168.86.10/luci-static/resources/luci.js?v=git-23.236.53281-98e3743:92
    start http://192.168.86.10/luci-static/resources/luci.js?v=git-23.236.53281-98e3743:88
    add http://192.168.86.10/luci-static/resources/luci.js?v=git-23.236.53281-98e3743:83
    click http://192.168.86.10/luci-static/resources/view/attendedsysupgrade/overview.js?v=git-23.236.53281-98e3743:25
    promise callback*handleCheck/</</<.click< http://192.168.86.10/luci-static/resources/view/attendedsysupgrade/overview.js?v=git-23.236.53281-98e3743:25
    createHandlerFn http://192.168.86.10/luci-static/resources/ui.js?v=git-23.236.53281-98e3743:357

illumiN8i, 2023-10-11:
Same issue here on R7800 ipq806x

ascendbeing, 2023-10-11:
firmware selector was working until I tried today. id been using it to upgrade because asu has been broken 1-2+ weeks. but now the customized selector sysupgrade image links to an "undefined" subdir (literally the dir is /undefined/) on downloads openwrt org. so when I click the link it's 404, and the build log says calculating hashes at the end but it doesn't say it was successful at it. and when I provide the same list of packages it still gives me this "build" which still returns 404.

robert-pajak, 2023-10-11:
Just tried it again and today, after pressing the green button it started to build an image ... but got stuck on 80%
[image]
After some time new AJAX requests stopped to show in the browser network tab (looks like Luci javascipt stopped to check what's the progress on building an image)
I checked the periodically invoked URL calls, and then they gave me a 404 response
[image]

robert-pajak, 2023-10-11:
I tried again and building started again ... 
Not sure if this is red herring or not but it seems to me a bit strange
[image]
I am trying to build Xiaomi 9000 image, but I can see from the included stdout JSON response references to other devices like
`SupportedDevices: zyxel,nbg7815\n"`

This is the full JSON response I'm getting (while it works)
```
{
  "detail": "started",
  "enqueued_at": "2023-10-11T11:25:56.021744Z",
  "imagebuilder_status": "calculate_packages_hash",
  "request": {
    "arch": "aarch64_cortex-a53",
    "branch": "SNAPSHOT",
    "branch_data": {
      "enabled": true,
      "extra_keys": [],
      "extra_repos": {},
      "git_branch": "master",
      "name": "SNAPSHOT",
      "package_changes": [
        {
          "revision": 21994,
          "source": "libustream-wolfssl",
          "target": "libustream-mbedtls"
        },
        {
          "revision": 21994,
          "source": "px5g-wolfssl",
          "target": "px5g-mbedtls"
        },
        {
          "revision": 21994,
          "source": "wpad-basic-wolfssl",
          "target": "wpad-basic-mbedtls"
        },
        {
          "revision": 21994,
          "source": "libustream-wolfssl",
          "target": "libustream-mbedtls"
        },
        {
          "mandatory": true,
          "revision": 20282,
          "source": "kmod-nft-nat6"
        },
        {
          "revision": 18611,
          "source": "firewall",
          "target": "firewall4"
        }
      ],
      "path": "snapshots",
      "path_packages": "snapshots/packages",
      "pubkey": "RWS1BD5w+adc3j2Hqg9+b66CvLR7NlHbsj7wjNVj0XGt/othDgIAOJS+",
      "repos": [
        "base",
        "packages",
        "luci",
        "routing",
        "telephony"
      ],
      "snapshot": true,
      "updates": "dev",
      "versions": [
        "SNAPSHOT"
      ]
    },
    "build_cmd_packages": [
[... 369 lines cut ...]
      "urandom-seed",
      "urngd",
      "usign",
      "usteer",
      "wireless-regdb",
      "wpad-basic-mbedtls"
    ],
    "profile": "xiaomi_ax9000",
    "public_path": "/mnt/store/asu",
    "repository_allow_list": [
      "http://downloads.openwrt.org",
      "https://downloads.openwrt.org",
      "http://feed.libremesh.org",
      "https://feed.libremesh.org"
    ],
    "request_hash": "d4a917af078e0bbfe12448e43a35c6ce",
    "target": "qualcommax/ipq807x",
    "version": "SNAPSHOT"
  },
  "request_hash": "d4a917af078e0bbfe12448e43a35c6ce",
  "status": 202,
  "stderr": "",
  "stdout": "Current Target: \"qualcommax/ipq807x\"\nCurrent Architecture: \"aarch64\"\nCurrent Revision: \"r24111-66f6c20e45\"\nDefault Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-mbedtls logd mtd netifd opkg uci uclient-fetch urandom-seed urngd busybox procd procd-ujail procd-seccomp kmod-usb3 kmod-usb-dwc3 kmod-usb-dwc3-qcom kmod-leds-gpio kmod-gpio-button-hotplug kmod-qca-nss-dp kmod-ath11k-ahb wpad-basic-mbedtls uboot-envtools e2fsprogs kmod-fs-ext4 losetup kmod-phy-aquantia ath11k-firmware-ipq8074 dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe\nAvailable Profiles:\n\nbuffalo_wxr-5950ax12:\n    Buffalo WXR-5950AX12\n    Packages: ipq-wifi-buffalo_wxr-5950ax12\n    hasImageMetadata: 1\n    SupportedDevices: buffalo,wxr-5950ax12\ncompex_wpq873:\n    Compex WPQ873\n    Packages: ipq-wifi-compex_wpq873\n    hasImageMetadata: 1\n    SupportedDevices: compex,wpq873\ndynalink_dl-wrx36:\n    Dynalink DL-WRX36\n    Packages: ipq-wifi-dynalink_dl-wrx36\n    hasImageMetadata: 1\n    SupportedDevices: dynalink,dl-wrx36\nedgecore_eap102:\n    Edgecore EAP102\n    Packages: ipq-wifi-edgecore_eap102\n    hasImageMetadata: 1\n    SupportedDevices: edgecore,eap102\nedimax_cax1800:\n    Edimax CAX1800\n    Packages: ipq-wifi-edimax_cax1800\n    hasImageMetadata: 1\n    SupportedDevices: edimax,cax1800\nnetgear_rax120v2:\n    Netgear RAX120v2\n    Packages: ipq-wifi-netgear_rax120v2 kmod-spi-gpio kmod-spi-bitbang kmod-gpio-nxp-74hc164 kmod-hwmon-g761\n    hasImageMetadata: 1\n    SupportedDevices: netgear,rax120v2\nnetgear_wax620:\n    Netgear WAX620\n    Packages: kmod-spi-gpio kmod-gpio-nxp-74hc164 ipq-wifi-netgear_wax620\n    hasImageMetadata: 1\n    SupportedDevices: netgear,wax620\nnetgear_wax218:\n    Netgear WAX218\n    Packages: kmod-spi-gpio kmod-spi-bitbang kmod-gpio-nxp-74hc164 ipq-wifi-netgear_wax218\n    hasImageMetadata: 1\n    SupportedDevices: netgear,wax218\nprpl_haze:\n    prpl Foundation Haze\n    Packages: ath11k-firmware-qcn9074 ipq-wifi-prpl_haze kmod-ath11k-pci mkf2fs f2fsck kmod-fs-f2fs\n    hasImageMetadata: 1\n    SupportedDevices: prpl,haze\nqnap_301w:\n    QNAP 301w\n    Packages: ipq-wifi-qnap_301w\n    hasImageMetadata: 1\n    SupportedDevices: qnap,301w\nredmi_ax6:\n    Redmi AX6\n    Packages: ipq-wifi-redmi_ax6\n    hasImageMetadata: 1\n    SupportedDevices: redmi,ax6\nxiaomi_ax3600:\n    Xiaomi AX3600\n    Packages: ipq-wifi-xiaomi_ax3600 kmod-ath10k-ct-smallbuffers ath10k-firmware-qca9887-ct\n    hasImageMetadata: 1\n    SupportedDevices: xiaomi,ax3600\nxiaomi_ax9000:\n    Xiaomi AX9000\n    Packages: ipq-wifi-xiaomi_ax9000 kmod-ath11k-pci ath11k-firmware-qcn9074 kmod-ath10k-ct ath10k-firmware-qca9887-ct\n    hasImageMetadata: 1\n    SupportedDevices: xiaomi,ax9000\nzyxel_nbg7815:\n    ZYXEL NBG7815\n    Packages: ipq-wifi-zyxel_nbg7815 kmod-ath11k-pci kmod-hwmon-tmp103 kmod-bluetooth\n    hasImageMetadata: 1\n    SupportedDevices: zyxel,nbg7815\n"
}
```

edrikk, 2023-10-11:
So I just tried going from 9536446965 to 66f6c20 on IPQ807.
The process started, but got stuck on "80% generating image" for a good 10 minutes.  I closed the browser as it wasn't going anywhere clearly.

Going in again (private mode, different browser, etc) and clicking on "request firmware" button does nothing.

oxavelar, 2023-10-11:
> So I just tried going from 9536446965 to 66f6c20 on IPQ807. The process started, but got stuck on "80% generating image" for a good 10 minutes. I closed the browser as it wasn't going anywhere clearly.
> 
> Going in again (private mode, different browser, etc) and clicking on "request firmware" button does nothing.
> 
> EDIT: @aparcar It is now 4 hours later. Still build 66f6c20 shows as available, but clicking on the upgrade button does nothing. I seems that the original failure to generate has made it so that this build for this device is no longer retrievable. At least that's the symptom that I'm observing.

Hit 'cancel', then press the top right corner and hit 'paused'. Now click again the green button 'Search for firmware upgrade', what do you see?

edrikk, 2023-10-11:
Clicking on "pause" doesn't do anything, but hitting "the green button" does cause action.
The only problem is that due to [this bug in the luci-app](https://github.com/openwrt/luci/issues/6284), the result in error:

> Generate local signing keys...
> Generate local certificate...
> Package list missing or not up-to-date, generating it.
> 
> Building package index...
> Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/Packages.gz
> Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_core
> Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
> Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_base
> Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.1.56-1-0567810dfd9a1152272622ddb29d7220/Packages.gz
> Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_kmods
> Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.1.56-1-0567810dfd9a1152272622ddb29d7220/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
> Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_luci
> Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
> Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_packages
> Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
> Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_routing
> Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
> Signature check passed.
> Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
> Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_telephony
> Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
> Signature check passed.
> Downloading file:packages/Packages
> Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/imagebuilder
> Downloading file:packages/Packages.sig
> Signature check passed.
> Collected errors:
>  * opkg_install_cmd: Cannot install package 0.
>  * opkg_install_cmd: Cannot install package 1.
>  * opkg_install_cmd: Cannot install package 10.
>  * opkg_install_cmd: Cannot install package 11.
>  * opkg_install_cmd: Cannot install package 12.
>  * opkg_install_cmd: Cannot install package 13.
>  * opkg_install_cmd: Cannot install package 14.
>  * opkg_install_cmd: Cannot install package 15.
>  * opkg_install_cmd: Cannot install package 16.
>  * opkg_install_cmd: Cannot install package 17.
>  * opkg_install_cmd: Cannot install package 18.
>  * opkg_install_cmd: Cannot install package 19.
>  * opkg_install_cmd: Cannot install package 2.
>  * opkg_install_cmd: Cannot install package 20.
>  * opkg_install_cmd: Cannot install package 21.
>  * opkg_install_cmd: Cannot install package 22.
>  * opkg_install_cmd: Cannot install package 23.
>  * opkg_install_cmd: Cannot install package 24.
>  * opkg_install_cmd: Cannot install package 25.
>  * opkg_install_cmd: Cannot install package 26.
>  * opkg_install_cmd: Cannot install package 27.
>  * opkg_install_cmd: Cannot install package 28.
>  * opkg_install_cmd: Cannot install package 29.
>  * opkg_install_cmd: Cannot install package 3.
>  * opkg_install_cmd: Cannot install package 30.
>  * opkg_install_cmd: Cannot install package 31.
>  * opkg_install_cmd: Cannot install package 32.
>  * opkg_install_cmd: Cannot install package 33.
>  * opkg_install_cmd: Cannot install package 34.
[... 33 lines cut ...]
>  * opkg_install_cmd: Cannot install package 65.
>  * opkg_install_cmd: Cannot install package 66.
>  * opkg_install_cmd: Cannot install package 67.
>  * opkg_install_cmd: Cannot install package 68.
>  * opkg_install_cmd: Cannot install package 69.
>  * opkg_install_cmd: Cannot install package 7.
>  * opkg_install_cmd: Cannot install package 70.
>  * opkg_install_cmd: Cannot install package 71.
>  * opkg_install_cmd: Cannot install package 72.
>  * opkg_install_cmd: Cannot install package 73.
>  * opkg_install_cmd: Cannot install package 74.
>  * opkg_install_cmd: Cannot install package 75.
>  * opkg_install_cmd: Cannot install package 76.
[... cut ...]
>  * opkg_install_cmd: Cannot install package 79.
>  * opkg_install_cmd: Cannot install package 8.
>  * opkg_install_cmd: Cannot install package 80.
>  * opkg_install_cmd: Cannot install package 81.
>  * opkg_install_cmd: Cannot install package 82.
>  * opkg_install_cmd: Cannot install package 83.
>  * opkg_install_cmd: Cannot install package 9.
> make[2]: *** [Makefile:189: package_install] Error 255
> make[1]: *** [Makefile:154: _call_manifest] Error 2
> make: *** [Makefile:274: manifest] Error 2

edrikk, 2023-10-12:
Hi @aparcar 
Just wondering if you had any leads?
Both auc and asu being broken right now is well..  not ideal.  :)

robert-pajak, 2023-10-12:
I'd agree with Edrikk
Any progress @aparcar?
Any leads you need our help with testing?

aparcar, 2023-10-12:
Well let's make this a call for help. I'm not a JavaScript developer and don't plan to be anytime soon. I don't fully understand multiple parts of LuCIs framework design and therefore wired state things sadly happen in the app. Please step up and fix the LuCI app or find people to work on it, I'd very much appreciate that!

For now, I'm unable to reproduce the issue and don't know how to fix it.

edrikk, 2023-10-13:
Oof.  Well that's not good. :(
The good news is that (per your other ticket's note) I just tested auc and that is indeed looking fixed. :)

khanjui, 2023-10-13:
I can confirm that auc is working now.

robert-pajak, 2023-10-13:
:( Unfortunately `auc` doesn't work for me (qualcommax/ipq807x, Xiaomi ax9000)
```
Error: Received inncorrect version r24111-66f6c20e45 (requested r24124-518923178c)
Bad message (74)

dangowrt, 2023-10-14:
This ia caused by the server not having synced the version info after the latest buildbot run. Just retry, it should work after at most 1h after buildbot completed. @aparcar also had aome webhook in mind to let the buildbot completion trigger a refresh in the server, but apparently this is not in place yet.

edrikk, 2023-10-14:
My other devices appear to be working, but the old pi model b plus isn't.  It's been giving invalid argument 22 for the last couple of days that I've tried.

```
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r24002-e3559fb445 on bcm27xx/bcm2708 (raspberrypi,model-b-plus)
Invalid argument (22)

```

YaeMashiro, 2023-10-15:
```
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r24096-9536446965 on bcm27xx/bcm2708 (raspberrypi,model-zero-w)
Invalid argument (22)
```

YaeMashiro, 2023-10-18:
> Are you sure you want to continue the upgrade process? [N/y] y
Requesting build.....Current Target: "mediatek/filogic"
Current Architecture: "aarch64"
Current Revision: "r24154-6f31941d38"
xiaomi_redmi-router-ax6000-stock:
    Xiaomi Redmi Router AX6000 (stock layout)
    Packages: kmod-leds-ws2812b kmod-mt7986-firmware mt7986-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: xiaomi,redmi-router-ax6000-stock
xiaomi_redmi-router-ax6000-ubootmod:
    Xiaomi Redmi Router AX6000 (OpenWrt U-Boot layout)
    Packages: kmod-leds-ws2812b kmod-mt7986-firmware mt7986-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: xiaomi,redmi-router-ax6000-ubootmod
zyxel_ex5601-t0-stock:
    Zyxel EX5601-T0  (stock layout)
    Packages: kmod-mt7986-firmware mt7986-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: mediatek,mt7986a-rfb-snand
zyxel_ex5700-telenor:
    ZyXEL EX5700 (Telenor)
    Packages: kmod-mt7916-firmware kmod-ubootenv-nvram kmod-usb3 kmod-mt7986-firmware mt7986-wo-firmware
    hasImageMetadata: 1
    SupportedDevices: zyxel,ex5700-telenor
zyxel_nwa50ax-pro:
    ZyXEL NWA50AX Pro
    Packages: kmod-mt7981-firmware mt7981-wo-firmware zyxel-bootconfig
    hasImageMetadata: 1
    SupportedDevices: zyxel,nwa50ax-pro
Error: Received inncorrect version r24154-6f31941d38 (requested r24143-c9e6831a89)
Bad message (74)

edrikk, 2023-10-21:
Yes.  Neither auc nor ASU work for any of my models again.
Given all the other tickets clearly it’s widespread.

alienatedsec, 2023-10-21:
#624 The same issue

edrikk, 2023-10-21:
@aparcar I think one thing that would be very useful is if there was a flag for ASU and AUC to show the package list that one could simply paste into the online image builder to get the same build as currently running.

For people such as myself who have for example replaced mbedtls with openssl, etc.  Right now, it's a more involved task, so when AUC/ASU break for this long, there's no workaround.

edrikk, 2023-10-22:
A few devices:

**Xiaomi AX3600**

```
Running:   SNAPSHOT r24111-66f6c20e45 on qualcommax/ipq807x (xiaomi,ax3600)
Available: SNAPSHOT r24154-6f31941d38
...
Error: Received inncorrect version r24188-997bbc8a94 (requested r24154-6f31941d38)
Bad message (74)

```

**Xiaomi AX6S:**

```
Running:   SNAPSHOT r24124-518923178c on mediatek/mt7622 (xiaomi,redmi-router-ax6s)
Available: SNAPSHOT r24154-6f31941d38
...
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_kmods https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/kmods/5.15.135-1-230128744517abb106a3c2cd55fd0788). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_telephony https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony). Skipping.
Error: Impossible package selection: base-files version not as requested: 1542-r24124-518923178c vs. 1542-r24154-6f31941d38
Bad message (74)

```

**Pi Model B:**

```
Running:   SNAPSHOT r24002-e3559fb445 on bcm27xx/bcm2708 (raspberrypi,model-b-plus)
Invalid argument (22)

```

**Xiaomi Redmi AC2100:**

```
Running:   SNAPSHOT r24124-518923178c on ramips/mt7621 (xiaomi,redmi-router-ac2100)
Available: SNAPSHOT r24154-6f31941d38
...
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_kmods https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/kmods/5.15.135-1-1a9ece2a179bfc47e0ea97e5833514ea). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_telephony https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony). Skipping.
Error: Impossible package selection: base-files version not as requested: 1542-r24124-518923178c vs. 1544-r24192-4bdd1c1a13
Bad message (77)

```

khanjui, 2023-10-23:
Error: Received inncorrect version r24194-e1aaa1defd (requested r24154-6f31941d38)
Bad message (77)

Mi 4A Gigabit

edrikk, 2023-11-01:
The Raspberry Pi still shows the same error (uci and luci doesn't work):

> auc/0.3.2-1
> Server:    https://sysupgrade.openwrt.org
> Running:   SNAPSHOT r24002-e3559fb445 on bcm27xx/bcm2708 (raspberrypi,model-b-plus)
> Invalid argument (22)
> 

This has been the case for a few weeks.
auc for my other devices (e.g. mediatek/mt7622 and qualcommax/ipq807x) started to work I believe it was about a week ago.

edrikk, 2023-11-13:
The above target still gives the same error in auc, and ASU button click doesn’t do anything still.

edrikk, 2023-11-23:
@aparcar the above issue remains for the bcm27xx/bcm2708 target.
ASU button does nothing, and auc gives error 22.

I used the output of script:

`opkg list | awk 'BEGIN { ORS=" " }; {print $1}'`

in the imagebuilder to successfully build an updated image.