openwrt/asu #600: sysupgrade.openwrt.org/stats 502

webysther, 2023-10-07:


aparcar, 2023-10-07:
Yes I disabled it. Previously it would use a Prometheus instance to render nice graphs, however the monitored stats ended being to much and overloading the server. Prometheus used more CPU to calculate to calculate metrics than the build process took.

If you have an idea for better stats let me know.

Currently there are about 800 images generated per day, some days more than 1000.

webysther, 2023-10-08:
Hi @aparcar, I think you use the best ideia yet, but for my own experience using prometheus, you need to reduce the stats collected, this have a huge impact on memory and cpu usage, I have about 100 containers on my homelab and reducing to only metrics needed and the time of collected make a great difference.

Looking forward, I think collecting only every day or week the usage can change drastic the prometheus consuption, change the `scrape_interval` on prometheus configuration to every hour for instance.

What is used to collect data, node exporter?

aparcar, 2023-10-08:
You can scrape the metrics here https://sysupgrade.openwrt.org/metrics I may give this another shot but please feel free to implement something and I'll add it to the main instance.

webysther, 2023-10-08:
Do you have the prometheus configuration used? I don't find him on `misc` folder, only grafana.

aparcar, 2023-10-08:
I enabled it again and reduced the scrape interval http://sysupgrade.openwrt.org:9090/classic/config

aparcar, 2023-10-08:
Closing this as I enabled the stats again

closed 2023-10-08 by aparcar

webysther, 2023-10-08:
> I enabled it again and reduced the scrape interval http://sysupgrade.openwrt.org:9090/classic/config

I think you need to increase the scrape interval to reduce the impact in the server.