openwrt/asu #604: installed package libelf cannot be found in remote list!

hrw, 2023-10-12:
23.05 got released so I decided to upgrade my AP to latest 22.03 (from older build) and it went fine. 

But one thing worried me:
```
root@krolik:~# auc
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   22.03.5 r20134-5f15225c1e on mediatek/mt7622 (linksys,e8450-ubi)
Available: 22.03.5 r20134-5f15225c1e
Requesting package lists...
installed package libelf cannot be found in remote list!
installed package libncurses cannot be found in remote list!
Are you sure you want to continue the upgrade process? [N/y] n
```

System has libelf**1** nd libncurses**6** installed:
```
root@krolik:~# opkg list_installed|grep libelf ; opkg list_installed|grep libncurses
libelf1 - 0.186-1
libncurses6 - 6.3-2
```

And it is not even some reminder from past:
```
root@krolik:~# grep ^Package /usr/lib/opkg/status |grep libelf
Package: libelf1
root@krolik:~# grep ^Package /usr/lib/opkg/status |grep libncurses
Package: libncurses6
root@krolik:~#
```

Safe to ignore probably but surprised me.

edrikk, 2023-10-12:
I would not use that - in fact both auc and asu are broken right now..  So ticket just a couple below yours.

aparcar, 2023-10-12:
AUC should be fixed, please give it another go

hrw, 2023-10-13:
Looks good.

```
root@krolik:~# auc -b 23.05
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   22.03.5 r20134-5f15225c1e on mediatek/mt7622 (linksys,e8450-ubi)
Available: 23.05.0 r23497-6637af95aa
Requesting package lists...
 kmod-usb-storage: 5.10.176-1 -> 5.15.134-1
 terminfo: 6.3-2 -> 6.4-2
 iwinfo: 2022-12-15-8d158096-1 -> 2023-07-01-ca79f641-1
 luci-app-statistics: git-23.153.53801-38f5b55 -> git-23.184.27766-2c7b34b
 luci-lib-base: git-20.232.39649-1f6dc29 -> git-22.308.54612-9118452
 kmod-crypto-gf128: 5.10.176-1 -> 5.15.134-1
 opkg: 2022-02-24-d038e5b6-1 -> 2022-02-24-d038e5b6-2
 kmod-usb-core: 5.10.176-1 -> 5.15.134-1
 luci-app-opkg: git-23.093.42303-e16f620 -> git-23.009.82915-ec3aac4
 kmod-usb-storage-uas: 5.10.176-1 -> 5.15.134-1
 ubus: 2022-06-01-2bebf93c-1 -> 2023-06-05-f787c97b-1
 iw: 5.16-1 -> 5.19-1
 kmod-crypto-manager: 5.10.176-1 -> 5.15.134-1
 libuci: 2021-10-22-f84f49f0-6 -> 2023-08-10-5781664d-1
 rpcd: 2022-12-15-7de4820c-1 -> 2023-07-01-c07ab2f9-1
 busybox: 1.35.0-5 -> 1.36.1-1
 kmod-crypto-ctr: 5.10.176-1 -> 5.15.134-1
 ip-full: 5.15.0-4 -> 6.3.0-1
 kmod-nft-fib: 5.10.176-1 -> 5.15.134-1
 kmod-nfnetlink: 5.10.176-1 -> 5.15.134-1
 libubus-lua: 2022-06-01-2bebf93c-1 -> 2023-06-05-f787c97b-1
 kmod-crypto-hash: 5.10.176-1 -> 5.15.134-1
 kmod-nf-reject6: 5.10.176-1 -> 5.15.134-1
 kmod-crypto-aead: 5.10.176-1 -> 5.15.134-1
 libiwinfo-lua: 2022-12-15-8d158096-1 -> 2023-07-01-ca79f641-1
 luci-mod-system: git-23.118.78821-0c02883 -> git-23.118.78765-58f7b27
 openssl-util: 1.1.1w-1 -> 3.0.11-1
 kmod-nf-flow: 5.10.176-1 -> 5.15.134-1
 libnl-tiny: 2021-11-21-8e0555fb-1 -> 2023-07-27-bc92a280-1
 kmod-lib-crc-ccitt: 5.10.176-1 -> 5.15.134-1
 getrandom: 2021-08-03-205defb5-2 -> 2022-08-13-4c7b720b-2
 ucode-mod-ubus: 2022-12-02-46d93c9c-1 -> 2023-06-06-c7d84aae-1
 blockdev: 2.37.4-1 -> 2.39-2
 libustream-openssl: 2022-12-08-9217ab46-2 -> 2023-02-25-498f6e26-1
 luci-theme-bootstrap: git-23.093.42704-b47268a -> git-23.085.34270-d94a728
 kmod-pppoe: 5.10.176-1 -> 5.15.134-1
 libbpf: 2022-03-08-04c465fd-1 -> 1.2.2-1
 kmod-pppox: 5.10.176-1 -> 5.15.134-1
 kmod-nf-reject: 5.10.176-1 -> 5.15.134-1
 procd-ujail: 2022-06-01-7a009685-2 -> 2023-06-25-2db83655-2
 base-files: 1502-r20134-5f15225c1e -> 1545-r23497-6637af95aa
 kmod-lib-crc16: 5.10.176-1 -> 5.15.134-1
 kmod-nf-nat: 5.10.176-1 -> 5.15.134-1
 kmod-crypto-crc32c: 5.10.176-1 -> 5.15.134-1
 ucode-mod-uci: 2022-12-02-46d93c9c-1 -> 2023-06-06-c7d84aae-1
 kmod-mt7615e: 5.10.176+2022-09-06-d7054646-5 -> 5.15.134+2023-08-14-b14c2351-1
 netifd: 2022-08-25-76d2d41b-1 -> 2023-09-19-7a58b995-1
 libubus: 2022-06-01-2bebf93c-1 -> 2023-06-05-f787c97b-1
 firewall4: 2022-10-18-7ae5e14b-1 -> 2023-03-23-04a06bd7-1
 uboot-envtools: 2022.01-38 -> 2023.04-1
 kmod-usb-xhci-hcd: 5.10.176-1 -> 5.15.134-1
 dnsmasq: 2.86-16 -> 2.89-4
 procd: 2022-06-01-7a009685-2 -> 2023-06-25-2db83655-2
 kmod-crypto-hmac: 5.10.176-1 -> 5.15.134-1
 kmod-crypto-sha256: 5.10.176-1 -> 5.15.134-1
 kmod-mt7615-common: 5.10.176+2022-09-06-d7054646-5 -> 5.15.134+2023-08-14-b14c2351-1
 ubusd: 2022-06-01-2bebf93c-1 -> 2023-06-05-f787c97b-1
 hostapd-utils: 2022-01-16-cff80b4f-16.2 -> 2023-09-08-e5ccbfc6-4
 kmod-nf-nat6: 5.10.176-1 -> 5.15.134-1
 kmod-lib-crc32c: 5.10.176-1 -> 5.15.134-1
 libelf: 0.186-1 -> 0.189-1
[... 82 lines cut ...]
 kmod-crypto-rng: 5.10.176-1 -> 5.15.134-1
 luci: git-23.093.42303-d58cd69 -> git-23.051.66410-a505bb1
 kmod-nf-conntrack6: 5.10.176-1 -> 5.15.134-1
 libgcc: 11.2.0-4 -> 12.3.0-4
 libucode: 2022-12-02-46d93c9c-1 -> 2023-06-06-c7d84aae-1
 kmod-mt76-core: 5.10.176+2022-09-06-d7054646-5 -> 5.15.134+2023-08-14-b14c2351-1
 ubox: 2021-08-03-205defb5-2 -> 2022-08-13-4c7b720b-2
 kmod-crypto-crc32: 5.10.176-1 -> 5.15.134-1
 rpcd-mod-iwinfo: 2022-12-15-7de4820c-1 -> 2023-07-01-c07ab2f9-1
 luci-mod-network: git-23.170.81153-99d3815 -> git-23.233.52805-3a0ef93
 kmod-nft-core: 5.10.176-1 -> 5.15.134-1
 libltdl: 2.4.6-2 -> 2.4.7-1
 kmod-nls-base: 5.10.176-1 -> 5.15.134-1
 mtr-json: 0.95-2 -> 0.95-3
 hostapd-common: 2022-01-16-cff80b4f-16.2 -> 2023-09-08-e5ccbfc6-4
 iperf3: 3.11-1 -> 3.15-1
 collectd-mod-cpu: 5.12.0-34 -> 5.12.0-49
 kmod-ipt-offload: 5.10.176-1 -> 5.15.134-1
 wireless-regdb: 2023.05.03-1 -> 2023.09.01-1
 kmod-scsi-core: 5.10.176-1 -> 5.15.134-1
 kmod-slhc: 5.10.176-1 -> 5.15.134-1
 kmod-cfg80211: 5.10.176+5.15.92-1-1 -> 5.15.134+6.1.24-3
 ppp-mod-pppoe: 2.4.9.git-2021-01-04-3 -> 2.4.9.git-2021-01-04-4
Are you sure you want to continue the upgrade process? [N/y] n
```

closed 2023-10-13 by hrw