openwrt/asu #611: Error: Impossible package selection

melvin7822, 2023-10-15:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "request_hash": "9bb56d28f8547c476d4f912b6926e20d",
    "profile": "raspberrypi,4-model-b",
    "target": "bcm27xx/bcm2711",
    "version": "23.05.0",
    "packages": [
        "base-files",
        "bcm27xx-gpu-fw",
        "bcm27xx-userland",
        "brcmfmac-firmware-usb",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd-mod-thermal",
        "cypress-firmware-43455-sdio",
        "cypress-nvram-43455-sdio-rpi-4b",
        "dnsmasq-full",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "ip6tables-nft",
        "iptables-nft",
        "irqbalance",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-fs-vfat",
        "kmod-hid",
        "kmod-hid-generic",
        "kmod-input-core",
        "kmod-input-evdev",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-sound-arm-bcm2835",
        "kmod-sound-core",
        "kmod-usb-core",
        "kmod-usb-hid",
        "kmod-usb-net-lan78xx",
        "kmod-usb-net-rtl8152",
        "libblkid",
        "libblobmsg-json",
        "libc",
        "libcomerr",
        "libext2fs",
        "libf2fs",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libsmartcols",
        "libss",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libuuid",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-sqm",
        "luci-base",
        "luci-i18n-attendedsysupgrade-en",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "mkf2fs",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "partx-utils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-luci",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "urandom-seed",
        "usign",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/bcm27xx/bcm2711/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/bcm27xx/bcm2711/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a72/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a72/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a72/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a72/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a72/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a72/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a72/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a72/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a72/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/aarch64_cortex-a72/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package cypress-nvram-43455-sdio-rpi-4b.
 * opkg_install_cmd: Cannot install package luci-i18n-attendedsysupgrade-en.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

aparcar, 2023-10-16:
Please try again and close this issue if it works now.

melvin7822, 2023-10-16:
I received the exact same error.

closed 2023-11-17 by melvin7822