openwrt/asu #612: Error creating the firmware image for tplink,tl-wr1043nd-v4

Rotorwerk, 2023-10-16:
Here is the output during creation of the firmware image for tplink,tl-wr1043nd-v4. I want to upgrade from 23.05.0-rc4 - r23482-7fe85ce1f2 and tried to create the firmware image for 23.05 

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23482-7fe85ce1f2",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "tplink,tl-wr1043nd-v4",
    "target": "ath79/generic",
    "version": "23.05.0",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "kernel",
        "kmod-ath9k",
        "kmod-gpio-button-hotplug",
        "kmod-nft-offload",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb2",
        "libc",
        "libgcc",
        "libustream-mbedtls",
        "logd",
        "luci-app-firewall",
        "luci-i18n-attendedsysupgrade-de",
        "luci-i18n-base-de",
        "luci-i18n-opkg-de",
        "luci-i18n-privoxy-de",
        "luci-i18n-uhttpd-de",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-ssl",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssh-keygen",
        "openssh-server",
        "openssh-sftp-server",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "privoxy",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "swconfig",
        "uboot-envtools",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd",
        "wpad-basic-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/undefined"
}

Regards
Norbert

shaman79, 2023-10-16:
Similar problem with cudy x6, I think this is some server-side error as it throws http 500.

jchulce, 2023-10-16:
Same issue right now for all of my devices, looks like a server issue

aparcar, 2023-10-16:
Full storage, works again.

closed 2023-10-16 by aparcar