openwrt/asu #617: Error: Impossible package selection tplink,archer-c7-v2, 22.03.5 to 23.05

Nothing2024, 2023-10-17:
[OpenWrt](http://192.168.9.1/)

    [Status](http://192.168.9.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [Overview](http://192.168.9.1/cgi-bin/luci/admin/status/overview)
        [Routing](http://192.168.9.1/cgi-bin/luci/admin/status/routes)
        [Firewall](http://192.168.9.1/cgi-bin/luci/admin/status/iptables)
        [System Log](http://192.168.9.1/cgi-bin/luci/admin/status/logs)
        [Processes](http://192.168.9.1/cgi-bin/luci/admin/status/processes)
        [Channel Analysis](http://192.168.9.1/cgi-bin/luci/admin/status/channel_analysis)
        [Realtime Graphs](http://192.168.9.1/cgi-bin/luci/admin/status/realtime)
        [VnStat Traffic Monitor](http://192.168.9.1/cgi-bin/luci/admin/status/vnstat)
        [WireGuard](http://192.168.9.1/cgi-bin/luci/admin/status/wireguard)
    [System](http://192.168.9.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [System](http://192.168.9.1/cgi-bin/luci/admin/system/system)
        [Administration](http://192.168.9.1/cgi-bin/luci/admin/system/admin)
        [Software](http://192.168.9.1/cgi-bin/luci/admin/system/opkg)
        [Startup](http://192.168.9.1/cgi-bin/luci/admin/system/startup)
        [Scheduled Tasks](http://192.168.9.1/cgi-bin/luci/admin/system/crontab)
        [Attended Sysupgrade](http://192.168.9.1/cgi-bin/luci/admin/system/attendedsysupgrade)
        [LED Configuration](http://192.168.9.1/cgi-bin/luci/admin/system/leds)
        [Backup / Flash Firmware](http://192.168.9.1/cgi-bin/luci/admin/system/flash)
        [Reboot](http://192.168.9.1/cgi-bin/luci/admin/system/reboot)
    [Services](http://192.168.9.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [Dynamic DNS](http://192.168.9.1/cgi-bin/luci/admin/services/ddns)
        [Adblock](http://192.168.9.1/cgi-bin/luci/admin/services/adblock)
        [Wake on LAN](http://192.168.9.1/cgi-bin/luci/admin/services/wol)
        [Tinyproxy](http://192.168.9.1/cgi-bin/luci/admin/services/tinyproxy)
    [Network](http://192.168.9.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
        [Interfaces](http://192.168.9.1/cgi-bin/luci/admin/network/network)
        [Wireless](http://192.168.9.1/cgi-bin/luci/admin/network/wireless)
        [Switch](http://192.168.9.1/cgi-bin/luci/admin/network/switch)
        [Routing](http://192.168.9.1/cgi-bin/luci/admin/network/routes)
        [DHCP and DNS](http://192.168.9.1/cgi-bin/luci/admin/network/dhcp)
        [Diagnostics](http://192.168.9.1/cgi-bin/luci/admin/network/diagnostics)
        [Firewall](http://192.168.9.1/cgi-bin/luci/admin/network/firewall)
    [Logout](http://192.168.9.1/cgi-bin/luci/admin/logout)

Paused

    [Overview](http://192.168.9.1/cgi-bin/luci/admin/system/attendedsysupgrade/overview)
    [Configuration](http://192.168.9.1/cgi-bin/luci/admin/system/attendedsysupgrade/configuration)

Attended Sysupgrade

The attended sysupgrade service allows to easily upgrade vanilla and custom firmware images.

This is done by building a new firmware on demand via an online service.

Currently running: 22.03.5 - r20134-5f15225c1e
[Powered by LuCI openwrt-22.03 branch (git-23.119.80898-65ef406)](https://github.com/openwrt/luci) / OpenWrt 22.03.5 r20134-5f15225c1e
Error building the firmware image

Server response: Error: Impossible package selection
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "tplink,archer-c7-v2",
    "target": "ath79/generic",
    "version": "23.05.0",
    "packages": [
        "adblock",
        "ath10k-board-qca988x",
        "ath10k-firmware-qca988x-ct",
[... 167 lines cut ...]
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-i18n-ddns-en.
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-mips_24kc_musl/root-ath79/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

Close

beatstick, 2023-10-17:
I've got a similar error on different devices (mir3g and 860l):

Collected errors:
 * opkg_install_cmd: Cannot install package ksmbd-utils.
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-mipsel_24kc_musl/root-ramips/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

It's probablby because the of the change from wolfssl to mbedtls.

I tried deselcting the wolfssl package in the advaced options, but this did not help.

beatstick, 2023-10-17:
I fixed the error by deselecting everything related to wolfssl among the existing packages when starting (advanced mode) attended sysupdate. I also had to deselect the  ksmbd-utils package.

Now 23.03 installed properly.

beatstick, 2023-10-17:
Oh, on both routers wireless stopped working now. 
I am not sure if this is a related issue: I get this error from logread:

e 'radio0' is now up
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122): Syntax error: Unable to compile module '/usr/share/hostap/common.uc':
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   | Syntax error: Unable to resolve path for module 'nl80211'
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   | In /usr/share/hostap/common.uc, line 1, byte 35:
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   |
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   |  `import * as nl80211 from "nl80211";`
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   |   Near here ------------------------^
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   |
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   |
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   | Syntax error: Unable to resolve path for module 'rtnl'
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   | In /usr/share/hostap/common.uc, line 2, byte 29:
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   |
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   |  `import * as rtnl from "rtnl";`
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):   |   Near here ------------------^
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122): In /usr/share/hostap/wdev.uc, line 3, byte 101:
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):
Tue Oct 17 17:44:53 2023 daemon.notice netifd: radio0 (2122):  `import { vlist_new, is_equal, wdev_create, wdev_remove, phy_open } from "/usr/share/hostap/common.uc";`