openwrt/asu #633: Linksys EA8500 Impossible packages upgrade to 23.05.0

OntarioJim, 2023-10-24:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "linksys,ea8500",
    "target": "ipq806x/generic",
    "version": "23.05.0",
    "packages": [
        "ath10k-firmware-qca99x0-ct",
        "base-files",
        "block-mount",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "https-dns-proxy",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ata-ahci",
        "kmod-ata-ahci-platform",
        "kmod-ata-core",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-fs-ext4",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-phy-qcom-ipq806x-usb",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-scsi-core",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-ehci",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb-ohci",
        "kmod-usb-storage",
        "kmod-usb-storage-uas",
        "kmod-usb-xhci-hcd",
        "kmod-usb2",
        "kmod-usb3",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-nlbwmon",
        "luci-app-opkg",
        "luci-app-samba4",
        "luci-app-sqm",
        "luci-app-watchcat",
        "luci-base",
        "luci-i18n-attendedsysupgrade-en",
        "luci-i18n-nlbwmon-en",
        "luci-i18n-samba4-en",
        "luci-i18n-sqm-en",
        "luci-i18n-watchcat-en",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "swconfig",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}
STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/ipq806x/generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/ipq806x/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/arm_cortex-a15_neon-vfpv4/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/arm_cortex-a15_neon-vfpv4/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/arm_cortex-a15_neon-vfpv4/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/arm_cortex-a15_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-i18n-attendedsysupgrade-en.
 * opkg_install_cmd: Cannot install package luci-i18n-nlbwmon-en.
 * opkg_install_cmd: Cannot install package luci-i18n-samba4-en.
 * opkg_install_cmd: Cannot install package luci-i18n-sqm-en.
 * opkg_install_cmd: Cannot install package luci-i18n-watchcat-en.
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

aparcar, 2023-10-24:
@dangowrt do you mind coming up with a solution which I could implement?

OntarioJim, 2023-10-24:
> @dangowrt do you mind coming up with a solution which I could implement?

Wolfssl has been replaced with mbedtls. I'm not sure if simply de-selecting wolfssl packages prior to building the image will solve the issue or if there are dependencies that will fail.

potchin, 2023-10-25:
Getting the same issue on a raspi 4. Even after removing/replacing wolfssl related packages..

`libustream-wolfssl` -> removed
`px5g-wolfssl` -> `px5g-mbedtls`
`wpad-basic-wolfssl` -> `wpad-basic-mbedtls `

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "raspberrypi,4-model-b",
    "target": "bcm27xx/bcm2711",
    "version": "22.03.5",
    "packages": [
        "adguardhome",
        "base-files",
        "bcm27xx-gpu-fw",
        "bcm27xx-userland",
        "brcmfmac-firmware-usb",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd",
        "collectd-mod-cpu",
        "collectd-mod-interface",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-memory",
        "collectd-mod-network",
        "collectd-mod-rrdtool",
        "collectd-mod-uptime",
        "curl",
        "cypress-firmware-43455-sdio",
        "cypress-nvram-43455-sdio-rpi-4b",
        "ddns-scripts",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iperf3",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-brcmfmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-fs-vfat",
        "kmod-hid",
        "kmod-hid-generic",
        "kmod-input-core",
        "kmod-input-evdev",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-nat6",
        "kmod-nf-reject",
[... 139 lines cut ...]
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a72/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a72/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a72/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a72/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a72/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package px5g-mbedtls wants to install file /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/usr/sbin/px5g
	But that file is already provided by package  * px5g-wolfssl
 * opkg_install_cmd: Cannot install package px5g-mbedtls.
 * opkg_install_cmd: Cannot install package wpad-basic-mbedtls.
make[2]: *** [Makefile:170: package_install] Error 255
make[1]: *** [Makefile:135: _call_manifest] Error 2
make: *** [Makefile:255: manifest] Error 2

Close
```