openwrt/asu #639: ASU finds update for 22.03.3, but not for 22.03.5

richb-hanover, 2023-10-29:
I'm using OpenWrt on two Netgear WNDR3800 routers as dumb APs in my home. They have substantially the same additional packages but one's running OpenWrt 22.03.3 r20028-43d71ad93e  - the other's running OpenWrt 22.03.5 r20134-5f15225c1e 

When using AttendedSysupgrade:

* The one running 22.03.3 CAN see a new version (23.05.0)
* The one running 22.03.5 says "The device runs the latest firmware version 22.03.5 - r20134-5f15225c1e"

I have note updated either router, but would be willing to provide additional troubleshooting info. Thanks for this terrific tool!

aparcar, 2023-10-30:
Is it possible that the one running on 22.03.3 has the advanced mode enabled? By default no branch upgrades are suggested but only point releases (i.e. 22.03.4 to 22.03.5).

richb-hanover, 2023-10-30:
Good thought, but no. Both Configuration tabs point to https://sysupgrade.openwrt.org, and both checkboxes are not checked. 

Here's the config info for both: what other info could I provide? Thanks.

```
root@MediaRoom-WNDR3800:/etc/config# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.3'
DISTRIB_REVISION='r20028-43d71ad93e'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.3 r20028-43d71ad93e'
DISTRIB_TAINTS=''
root@MediaRoom-WNDR3800:/etc/config# cat /etc/config/attendedsysupgrade

config server 'server'
	option url 'https://sysupgrade.openwrt.org'

config client 'client'
	option upgrade_packages '1'
	option auto_search '0'
	option advanced_mode '0'
```

```
root@FrontRoom-WNDR3800:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.5'
DISTRIB_REVISION='r20134-5f15225c1e'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.5 r20134-5f15225c1e'
DISTRIB_TAINTS=''
root@FrontRoom-WNDR3800:~# cat /etc/config/attendedsysupgrade

config server 'server'
	option url 'https://sysupgrade.openwrt.org'

config client 'client'
	option upgrade_packages '1'
	option auto_search '0'
	option advanced_mode '0'
```

aparcar, 2023-10-30:
Are you using LuCI or CLI?

richb-hanover, 2023-10-30:
@aparcar I'm sorry - I screwed up the version numbers. See the original post for the correct question. Thanks.

eehmke, 2023-11-03:
This seems to be the same issue as 
https://github.com/openwrt/asu/issues/618
I tried both LuCI and CLI.

eehmke, 2023-11-03:
When activating advanced mode, there are conflicts regarding the package
libustream-wolfssl20201210

eehmke, 2023-11-04:
I ended up installing the sysupgrade image, reinstalled my apps and restored my backup. That worked.
I go the impression that attended sysupgrade does not handle well changes like moving from  wolfssl to embedssl.

richb-hanover, 2023-11-05:
I am closing this issue. My real question is, "Why doesn't attendedsysupgrade want to upgrade 22.03.x to 23.05.x?"

I'll submit a new issue ...

closed 2023-11-05 by richb-hanover