openwrt/asu #647: Error building the firmware image - 22.03.05 -> 23.05.0

richb-hanover, 2023-11-05:
As detailed in #639, I am trying to upgrade my 22.03.05 router to 23.05.0. I checked the "Advance" box in the Configuration tab, selected 23.05.0, then got the error message below.

@eehmke @aparcar I see this looks like the same problem as https://github.com/openwrt/asu/issues/639#issuecomment-1793200005. Is there a workaround? What other troubleshooting information can I provide? 

```
Error building the firmware image

Server response: Error: Impossible package selection
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "request_hash": "1515dddf392f726d2cc6c74f72f71cf7",
    "profile": "netgear,wndr3800",
    "target": "ath79/generic",
    "version": "23.05.0",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd-mod-ethstat",
        "collectd-mod-ipstatistics",
        "collectd-mod-irq",
        "collectd-mod-load",
        "collectd-mod-ping",
        "collectd-mod-powerdns",
        "collectd-mod-sqm",
        "collectd-mod-thermal",
        "collectd-mod-wireless",
        "curl",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables-nft",
        "iptables-nft",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-cfg80211",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-leds-reset",
        "kmod-lib-crc-ccitt",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-owl-loader",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb-ohci",
        "kmod-usb2",
        "libblobmsg-json",
        "libc",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libustream-wolfssl",
        "libwolfssl",
        "libxtables",
        "logd",
        "lua",
        "lua-argparse",
        "lua-bit32",
        "lualanes",
        "luaposix",
        "luarocks",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-snmpd",
        "luci-app-sqm",
        "luci-app-statistics",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "netperf",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "swconfig",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}

STDERR:

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-mips_24kc_musl/root-ath79/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

eehmke, 2023-11-05:
@richb-hanover 
In one case I could do the upgrade when using advanced mode. I disabled all wolfssl packages before the generation of the upgrade. This seemed to work, but I could not repeat the procedure on another device. Finally I did a complete sysupgrade using the package from the firmware selector. Of course, I had to reinstall all needed packages and restore the configuration backup.

As I have more devices  to upgrade, I step back for now and wait for a solution.

eehmke, 2023-11-05:
We have now several related issues:
https://github.com/openwrt/asu/issues/633

akostadinov, 2024-09-02:
Another duplicate #938 and I also hit this issue. Would be really useful to fix or provide a reliable workaround. Now I see things could be easy I can't do it the hard way anymore!!! :sunglasses: