openwrt/asu #653: Error building the firmware image: Impossible package selection

MilesBHuff, 2023-11-15:
# Error building the firmware image
**Server response:** `Error: Impossible package selection`

*[Please report the error message and request](https://github.com/openwrt/asu/issues)*

## Request Data:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "zbtlink,zbt-wg3526-16m",
    "target": "ramips/mt7621",
    "version": "23.05.0",
    "packages": [
        "adblock",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables-nft",
        "iptables-nft",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ata-ahci",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-mt7603",
        "kmod-mt76x2",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-sdhci-mt7620",
        "kmod-slhc",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb3",
        "libblobmsg-json",
        "libc",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libopenssl",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libustream-wolfssl",
        "libxtables",
        "logd",
        "lua",
        "luci",
        "luci-app-adblock",
        "luci-app-attendedsysupgrade",
        "luci-app-commands",
        "luci-app-firewall",
        "luci-app-openvpn",
        "luci-app-opkg",
        "luci-app-upnp",
        "luci-base",
        "luci-compat",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openvpn-openssl",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "tcpdump",
        "ubi-utils",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}
```

## STDERR:
```
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.0/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.0/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-mipsel_24kc_musl/root-ramips/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

score76, 2023-11-16:
Me too, also trying to upgrade from r20134...

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "tplink,archer-c6-v3",
    "target": "ramips/mt7621",
    "version": "23.05.2",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-gpio-button-hotplug",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mt7603",
        "kmod-mt7615e",
        "kmod-mt7663-firmware-ap",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
[... 84 lines cut ...]
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-i18n-attendedsysupgrade-en.
 * opkg_install_cmd: Cannot install package luci-i18n-dashboard-en.
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-mipsel_24kc_musl/root-ramips/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

florian-reus, 2023-12-06:
Me too. 
main not-default package is wpad-wolfssl. I tried to deselect it, with same outcome.

Server response: Error: Impossible package selection
Please report the error message and request

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "request_hash": "82655cf9bbf5f7ce1284c53994b4a1a7",
    "profile": "ubnt,unifi-6-lr-v2",
    "target": "mediatek/mt7622",
    "version": "23.05.2",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7615-common",
        "kmod-mt7615-firmware",
        "kmod-mt7615e",
        "kmod-mt7915e",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
[... 114 lines cut ...]
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

florian-reus, 2023-12-13:
my workaround for my device upgrades from 22 to 23:
- use https://firmware-selector.openwrt.org for custom image
- add the non-standard, needed packages there
- update
--- worked for unifi 6 LR with 4 non-standard packages (used as APs)
--- next test will be Asus AX53U. used as router, so there are ~ 20 packages to add