openwrt/asu #654: Error building the firmware image

wdpi, 2023-11-16:
Server response: Error: Impossible package selection

**Request Data:**
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23497-6637af95aa",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "tplink,archer-c6-v2",
    "target": "ath79/generic",
    "version": "23.05.2",
    "packages": [
        "ath10k-board-qca9888",
        "ath10k-firmware-qca9888-ct",
        "attendedsysupgrade-common",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "https-dns-proxy",
        "iperf3",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-random-core",
        "kmod-slhc",
        "libblobmsg-json",
        "libc",
        "libcap-ng",
        "libcares",
        "libcurl",
        "libev",
        "libgcc",
        "libiperf3",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnghttp2",
        "libnl-tiny",
        "libpthread",
        "librt",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "libuuid",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-https-dns-proxy",
        "luci-app-opkg",
        "luci-app-upnp",
        "luci-base",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "miniupnpd-nftables",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rpcsys",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "swconfig",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.132.65998-fa9fb2f"
}
```

**STDERR:**

```
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.2/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.2/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/base/netifd_2023-11-10-35facc83-1_mips_24kc.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download netifd. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package base-files.
 * opkg_install_pkg: Checksum or size mismatch for package netifd. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package netifd.
 * opkg_install_pkg: Checksum or size mismatch for package netifd. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package netifd.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

user8446, 2023-11-16:
Mine failed as well, server response for me was: init

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23497-6637af95aa",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "request_hash": "e5bf8f80736985c947ee7489e4bbc4f7",
    "profile": "extreme-networks,ws-ap3825i",
    "target": "mpc85xx/p1020",
    "version": "23.05.2",
    "packages": [
        "ath10k-firmware-qca988x",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dropbear",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "hostapd-mbedtls",
        "htop",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-button-hotplug",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-input-core",
        "kmod-input-gpio-keys",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nls-base",
        "kmod-random-core",
        "kmod-slhc",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "openwrt-keyring",
        "opkg",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "swconfig",
        "tc-tiny",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.306.39494-6f7b129"
}
```

wdpi, 2023-12-03:
Today ( Dec. 3, 2023) it worked flawlessly.

closed 2023-12-03 by wdpi