openwrt/asu #659: Error building the firmware image - Server response: init

mepley1, 2023-11-19:
Error building the firmware image
Server response: init

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20123-38ccc47687",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "extreme-networks,ws-ap3825i",
    "target": "mpc85xx/p1020",
    "version": "23.05.2",
    "packages": [
        "ath10k-firmware-qca988x-ct",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "htop",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-button-hotplug",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-input-core",
        "kmod-input-gpio-keys",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-ehci",
        "kmod-usb2",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "luci-theme-material",
        "mtd",
        "nano",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "swconfig",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}

jacekjaros, 2023-11-20:
i have similar issue on the same hardware

```
Error building the firmware image

Server response: init
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "extreme-networks,ws-ap3825i",
    "target": "mpc85xx/p1020",
    "version": "23.05.2",
    "packages": [
        "ath10k-firmware-qca988x-ct",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "htop",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-button-hotplug",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-input-core",
        "kmod-input-gpio-keys",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-ehci",
        "kmod-usb2",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "lldpd",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-i18n-attendedsysupgrade-en",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssl-util",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "procps-ng",
        "procps-ng-ps",
        "prometheus-node-exporter-lua-hostapd_stations",
        "prometheus-node-exporter-lua-openwrt",
        "prometheus-node-exporter-lua-uci_dhcp_host",
        "prometheus-node-exporter-lua-wifi",
        "prometheus-node-exporter-lua-wifi_stations",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "swconfig",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "vim-full",
        "wireless-regdb",
        "wpad-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}
```

upgrade to latest 22.03.5 are working fine

jacekjaros, 2023-11-20:
upgrade 23.05 to 23.05.2 are also failing

```
Error building the firmware image

Server response: init
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23497-6637af95aa",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "extreme-networks,ws-ap3825i",
    "target": "mpc85xx/p1020",
    "version": "23.05.2",
    "packages": [
        "ath10k-firmware-qca988x-ct",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "htop",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-button-hotplug",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-input-core",
        "kmod-input-gpio-keys",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-random-core",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-ehci",
        "kmod-usb2",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "lldpd",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-ttyd",
        "luci-base",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "swconfig",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.306.39494-6f7b129"
}
```

mepley1, 2023-11-22:
[This comment](https://github.com/openwrt/asu/issues/654#issuecomment-1813748902) under #654 looks like the same issue on another AP3825i.

tcharewicz, 2023-11-23:
The same issue when you try build image on web page with default list of packages.

[image]

jacekjaros, 2023-12-21:
i think this issue can be closed - i don't face error described here any more

closed 2023-12-27 by mepley1