openwrt/asu #660: Error building the firmware image Server response: Error: Impossible package selection

CapXilinx, 2023-11-23:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19803-9a599fee93",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "tplink,archer-a7-v5",
    "target": "ath79/generic",
    "version": "22.03.5",
    "packages": [
        "ath10k-board-qca988x",
        "base-files",
        "busybox",
        "ca-certificates",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-batman-adv",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb2",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libopenssl",
        "libopenssl-conf",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libustream-openssl",
        "logd",
        "lua",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssl-util",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "swconfig",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "urandom-seed",
        "usign",
        "wpad-mesh-openssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}
STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/22.03.5/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/22.03.5/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package libustream-wolfssl20201210 wants to install file /builder/build_dir/target-mips_24kc_musl/root-ath79/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-openssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:170: package_install] Error 255
make[1]: *** [Makefile:135: _call_manifest] Error 2
make: *** [Makefile:255: manifest] Error 2

Znerox, 2023-11-26:
I got the same error, upgrading a Pi 4 from 22.03.5 to 23.05.2

Collected errors:
 * opkg_install_cmd: Cannot install package cypress-nvram-43455-sdio-rpi-4b.
 * opkg_install_cmd: Cannot install package luci-i18n-pbr-en.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

fbrendel, 2023-12-01:
I had similar errors with luci-ssl dependencies libustream-mbedtls vs. libustream-wolfssl.
I solved that by removing wolfssl and installing mbedtls.
Here are my steps:
1. download libustream-mbedtls, ca-bundle and ca-certificates packages to your PC (NOT TO THE ROUTER) https&#58;//downloads.openwrt.org/releases/23.05.2/packages/mips_24kc/base/ (in **my** case _mips_24kc_)
2. run LuCI asu and deselect all *wolfssl* packages
3. after the sysupgrade copy the downloaded *.ipk from step 1 to /tmp
4. install these packages `opkg install /tmp/libustream-mbedtls20201210_2023-02-25-498f6e26-1...`
5. run `opkg update` and install the deselected packages from step 2 with the correct ssl vendor (in my case mbedtls)

Please keep in mind that you need a working network connection to your router to copy the *.ipk after the sysupgrade for an offline install. So be carefull what you remove in step 2 :smirk:

MartinMelikMerkumians, 2024-04-17:
Thanks, that solved it for me

closed 2025-04-20 by MartinMelikMerkumians