openwrt/asu #662: Could not reach API at "https://sysupgrade.openwrt.org/json/v1/overview.json". Please try again later.

DaniilMir, 2023-11-28:
On both OenWRT 23.05.2 devices I got the following error in UI: 
```
Error connecting to upgrade server
Could not reach API at "https://sysupgrade.openwrt.org/json/v1/overview.json". Please try again later.
```
auc command results : 
> root@router:/etc/scripts# auc -n
> auc/0.3.2-1
> Server:    https://sysupgrade.openwrt.org
> Running:   23.05.2 r23630-842932a63d on ipq806x/generic (netgear,r7800)
> No data available (61)

Is it a server-side issue or a local one?

bplab-IN, 2023-11-28:
I'm experiencing this issue as well, with a Belkin RT3200. I made sure all my packages are updated, and I do not have banip installed.

nigelterry, 2023-11-28:
I have the same issue. It looks likehttps://sysupgrade.openwrt.org/json/v1/overview.json is missing or protected from being read by the web server

DaniilMir, 2023-11-28:
> I have the same issue. It looks like https://sysupgrade.openwrt.org/json/v1/overview.json is missing or protected from being read by the web server

[image]

Yes, Looks like it is a server-side issue.

nigelterry, 2023-11-28:
[image]

lemur1111, 2023-11-28:
Same here:
Error connecting to upgrade server
Could not reach API at "https://sysupgrade.openwrt.org/json/v1/overview.json". Please try again later.

metaphys, 2023-11-29:
Problem for me too:
```sh
~# auc -c
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   22.03.4 r20123-38ccc47687 on x86/64 (generic)
No data available (61)
```

babstar99, 2023-11-29:
Also experiencing this issue;
    # auc -c
    auc/0.3.2-1
    Server:    https://sysupgrade.openwrt.org
    Running:   23.05.2 r23630-842932a63d on mediatek/mt7622 (netgear,wax206)
   No data available (61)

aparcar, 2023-11-30:
Please try again. There is something wrong with the updating logic so a file was missing. I'll see what's the issue and let you know. Sorry for the inconvenience

DaniilMir, 2023-11-30:
@aparcar  Thank you for updating. 
I just tried and on bth devices got the following error : 

> Error building the firmware image
Server response: The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

aparcar, 2023-11-30:
I rested the server to have it pick the latest code changes. I see now people using it again:

```
default      |████████ 8, 4 executing, 2444 finished, 123 failed
1 queues, 8 jobs total

7cdf038894454b42975044aee5eb344b (03cb44a0621b 10.89.0.25:38844 1): busy default. jobs: 3 finished, 1 failed
c4c97d59bfdf42f883448aa81113fb44 (e48b99690c80 10.89.0.24:40316 1): busy default. jobs: 4 finished, 0 failed
85662540cda34ba2b6bcd7c6cb386eb9 (d274ee88ffdc 10.89.0.26:48284 1): busy default. jobs: 3 finished, 0 failed
23529d0b430f479b98214b8c724edc35 (94ef57d13cc3 10.89.0.23:38148 1): busy default. jobs: 4 finished, 0 failed
4 workers, 1 queues

Updated: 2023-11-30 09:56:00.605556
```

DaniilMir, 2023-11-30:
@aparcar For me it is still failing in both UI and shell options. 

>  auc -n
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on ipq806x/generic (netgear,r7800)
Invalid argument (22)

[image]

akoskoronka, 2023-11-30:
Luci version works as the http error 404 is resolved, auc still reports Invalid argument (22)

akoskoronka, 2023-12-01:
Now auc seems to be listing updates properly, however did not yet try to request the update.

akoskoronka, 2023-12-01:
Thanks! If all fine, I suppose this issue can be closed.

brad112358, 2024-09-04:
I'm not sure why this wasn't closed, but since it's not, I'll report I'm seeing the exact same issue currently.  Is there again a problem with this service?

Eldarock, 2024-09-06:
Looks like it's back online.

ShayBox, 2024-09-07:
This service seems to crash a lot, maybe a secondary instance on another server/port would help with reliability?

shampeon, 2024-09-18:
The service is back offline.

alpha30, 2024-10-27:
Unfortunately, it seems to be not working again, I just tried several ways of performing an attended sysupgrade and always got "http 404 error" "server reply invalid"

wildnet, 2024-12-06:
Unfortunately, it seems to be not working again, I can't even ping it.

IrfDev, 2024-12-06:
Service is down again :(

UAVXP, 2025-02-06:
[This URL](https://sysupgrade.openwrt.org/json/v1/overview.json) returns `Internal Server Error` now.
I cannot perform attended sysupgrade

ayubwaheed, 2025-02-21:
Service down again!!!

Whole API service is down getting the error message

Could not reach API at "https://sysupgrade.openwrt.org/json/v1/overview.json". Please try again later.

Internal Server Error

scroc-ha, 2025-04-18:
Hi,

[This URL](https://sysupgrade.openwrt.org/json/v1/overview.json) returns `Internal Server Error` again.

Is there an alternative server to upgrade from 23.05 ?

maxinfo337, 2025-05-18:
same ! internal server error :( how to upgrade from 23.05 ?

BUG-reporter, 2025-05-20:
The same problem.

Soromeister, 2025-05-21:
Same problem appears now.

s0ftice, 2025-05-22:
https://forum.openwrt.org/t/the-openwrt-firmware-selector/81721/1375
maintainer is on vacation 🏝️🤷🏼

Soromeister, 2025-05-25:
Started working again. Thanks to whomever fixed it.