openwrt/asu #671: Error building the firmware image on qualcommax/ipq807

robert-pajak, 2023-12-01:
Could this issue be investigated please?
I am trying to upgrade using Luci Attended Sysupgrade but I'm getting the following error

Error building the firmware image
Server response: Error: Impossible package selection

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24450-944d3526a3",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "request_hash": "739fa5fe7c5ff119899efd7a42cffbc5",
    "profile": "xiaomi,ax9000",
    "target": "qualcommax/ipq807x",
    "version": "SNAPSHOT",
    "packages": [
        "ath10k-firmware-qca9887-ct-full-htt",
        "ath11k-firmware-ipq8074",
        "ath11k-firmware-qcn9074",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "collectd-mod-contextswitch",
        "collectd-mod-cpufreq",
        "collectd-mod-curl",
        "collectd-mod-dhcpleases",
        "collectd-mod-entropy",
        "collectd-mod-filecount",
        "collectd-mod-fscache",
        "collectd-mod-irq",
        "collectd-mod-modbus",
        "collectd-mod-mqtt",
        "collectd-mod-ntpd",
        "collectd-mod-openvpn",
        "collectd-mod-ping",
        "collectd-mod-processes",
        "collectd-mod-tcpconns",
        "ddns-scripts-route53",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "ethtool",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "htop",
        "inotifywait",
        "inotifywatch",
        "iperf3",
        "ipq-wifi-xiaomi_ax9000",
        "iptraf-ng",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath11k",
        "kmod-ath11k-ahb",
[... 326 lines cut ...]
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20231130 for mtd
 * pkg_hash_fetch_best_installation_candidate: Packages for mtd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package mtd.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mtd:
 *      libubox20231130
 * opkg_install_cmd: Cannot install package mtd.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
Error: Impossible package selection
Bad message (74)
`

rennewpeter, 2023-12-01:
I have the same error

robert-pajak, 2023-12-05:
Issue is now resolved

closed 2023-12-05 by robert-pajak