openwrt/asu #677: Attended Upgraded Error: Error building the firmware image Server response: Unsupported profile: glinet,gl-mt6000

diggity801, 2023-12-05:
Error building the firmware image
Server response: Unsupported profile: glinet,gl-mt6000

Please report the error message and request
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23497-6637af95aa",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "glinet,gl-mt6000",
    "target": "mediatek/filogic",
    "version": "23.05.2",
    "packages": [
        "attr",
        "avahi-dbus-daemon",
        "bark-router-client",
        "base-files",
        "blkid",
        "block-mount",
        "busybox",
        "ca-bundle",
        "ca-certificates",
        "cgi-io",
        "chat",
        "comgt",
        "comgt-ncm",
        "conntrack",
        "curl",
        "dbus",
        "ddns-scripts",
        "ddns-scripts-services",
        "dhcpdiscover",
        "dnscrypt-proxy2",
        "dnsmasq-full",
        "dosfstools",
        "dropbear",
        "e2fsprogs",
        "eip197-mini-firmware",
        "ethtool",
        "f2fsck",
        "fcgi",
        "firewall4",
        "fstools",
        "fwtool",
        "getdns",
        "getrandom",
        "gl-line-graph",
        "gl-oui-rpc",
        "gl-sdk4-acl",
        "gl-sdk4-adguardhome",
        "gl-sdk4-bark",
        "gl-sdk4-base-files",
        "gl-sdk4-cable",
        "gl-sdk4-carrier-monitor",
        "gl-sdk4-clients",
        "gl-sdk4-cloud",
        "gl-sdk4-ddns",
        "gl-sdk4-dns",
        "gl-sdk4-edgerouter",
        "gl-sdk4-firewall",
        "gl-sdk4-fullconenat",
        "gl-sdk4-igmp",
        "gl-sdk4-init-portal",
        "gl-sdk4-ipv6",
        "gl-sdk4-lan",
        "gl-sdk4-led",
[... 424 lines cut ...]
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "unzip",
        "uqmi",
        "urandom-seed",
        "urngd",
        "usb-modeswitch",
        "usbmuxd",
        "usign",
        "vsftpd-tls",
        "wireguard-tools",
        "wireless-regdb",
        "wpad-openssl",
        "wwan",
        "xtables-nft",
        "zerotier",
        "zlib"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.132.65998-fa9fb2f"

aparcar, 2023-12-05:
How do you run that device on a stable branch? I can only find the image for snapshots?

diggity801, 2023-12-05:
> How do you run that device on a stable branch? I can only find the image for snapshots?

GL.iNet sells them preloaded with OpenWRT, but they add their own UI in addition to the Luci UI and some of the settings that they preloaded (SMB for example) are causing me issues, so I'm trying to load a vanilla version of OpenWRT.

aparcar, 2023-12-05:
I think right now you have to use the official snapshots or stick with their custom version.