openwrt/asu #685: Build Servers Unavailable

jking-cutter, 2023-12-12:
The https://sysupgrade.openwrt.org and http://asu.aparcar.org:8000 build servers appear to be non-functional at the moment. 

https://sysupgrade.openwrt.org
Running:   22.03.5 r20134-5f15225c1e on octeon/generic (erlite)
Invalid argument (22)

Server:    http://asu.aparcar.org:8000
Running:   22.03.5 r20134-5f15225c1e on octeon/generic (erlite)
No data available (61)

Server:    https://asu.aparcar.org
Running:   22.03.5 r20134-5f15225c1e on octeon/generic (erlite)
Connection error: Invalid SSL certificate
No data available (61)

Stats on the https://sysupgrade.openwrt.org server show builds are don't appear to be running currently: https://sysupgrade.openwrt.org/stats/?orgId=1&refresh=1m&from=now-7d&to=now

jturn08, 2023-12-13:
Confirm I've had the same issue for the last day

Soromeister, 2023-12-13:
Could be related to #663

lars18th, 2023-12-13:
Hi @Soromeister ,

> In the meantime, you could use [asu.hauke-m.de](http://asu.hauke-m.de:8000/) instead.

The server seems the be running, but it fails:
```
http://asu.hauke-m.de:8000/stats

{
  "detail": "The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.",
  "status": 500,
  "title": "Internal Server Error",
  "type": "about:blank"
}
```

aparcar, 2023-12-13:
Should be running again, Podman keeps having wired hiccups were it either slows down massively or refuses to work at all. I ~~can't~~ don't want to automate the hard reset of Podman every now and then. I might drop the entire Podman approach again as nothing really got better, sadly.

hyperreal64, 2023-12-13:
I have this issue as well. No matter which server I try, it says "No data available."

jking-cutter, 2023-12-13:
> I have this issue as well. No matter which server I try, it says "No data available."

@hyperreal64 Appears to be working fine now on https://sysupgrade.openwrt.org/ at least, stats show a number of builds going through still.