openwrt/asu #687: Error building the firmware image - xiaomi,mi-router-4a-gigabit

webysther, 2023-12-14:
Currently running: 23.05.2 - r20265-f85a79bcb4
Board Name / Profile: xiaomi,mi-router-4a-gigabit

```json
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20265-f85a79bcb4",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "xiaomi,mi-router-4a-gigabit",
    "target": "ramips/mt7621",
    "version": "23.05.2",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "curl",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "htop",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-gpio-button-hotplug",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-zstd",
        "kmod-mt7603",
        "kmod-mt76x2",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libpcap",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-i18n-attendedsysupgrade-en",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-rpc",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoa",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "procps-ng-watch",
        "prometheus-node-exporter-lua",
        "prometheus-node-exporter-lua-nat_traffic",
        "prometheus-node-exporter-lua-netstat",
        "prometheus-node-exporter-lua-openwrt",
        "prometheus-node-exporter-lua-uci_dhcp_host",
        "prometheus-node-exporter-lua-wifi",
        "prometheus-node-exporter-lua-wifi_stations",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "tcpdump",
        "ubi-utils",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl",
        "zram-swap",
        "zstd"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}
```

STDERR

```sh
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.2/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.2/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-i18n-attendedsysupgrade-en.
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-mipsel_24kc_musl/root-ramips/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

rany2, 2023-12-17:
Please see this comment: https://github.com/openwrt/asu/issues/688#issuecomment-1859301228

webysther, 2023-12-18:
After removing `luci-ssl` from selection worked, thanks

closed 2023-12-18 by webysther