openwrt/asu #688: Error building the firmware image

jfjoly, 2023-12-17:
Upgrading from OpenWrt 22.03.6 r20265-f85a79bcb4

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20265-f85a79bcb4",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "request_hash": "d4115bba62279d7bd6ebd5ceeffaf685",
    "profile": "raspberrypi,4-compute-module",
    "target": "bcm27xx/bcm2711",
    "version": "23.05.2",
    "packages": [
        "base-files",
        "bash",
        "bcm27xx-gpu-fw",
        "bcm27xx-userland",
        "brcmfmac-firmware-usb",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd",
        "collectd-mod-cpu",
        "collectd-mod-interface",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-memory",
        "collectd-mod-network",
        "collectd-mod-ping",
        "collectd-mod-rrdtool",
        "collectd-mod-sensors",
        "collectd-mod-thermal",
        "cypress-firmware-43455-sdio",
        "cypress-nvram-43455-sdio-rpi-4b",
        "diffutils",
        "dnscrypt-proxy2",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "fdisk",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "httping",
        "iperf",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-brcmfmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-fs-vfat",
        "kmod-hid",
        "kmod-hid-generic",
        "kmod-input-core",
        "kmod-input-evdev",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mdio-devres",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-phy-realtek",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-r8169",
        "kmod-slhc",
        "kmod-sound-arm-bcm2835",
        "kmod-sound-core",
        "kmod-usb-core",
        "kmod-usb-hid",
        "kmod-usb-net-lan78xx",
        "libblkid",
        "libblobmsg-json",
        "libc",
        "libcomerr",
        "libext2fs",
        "libf2fs",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libopenssl",
        "libsmartcols",
        "libss",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libuuid",
        "libwolfssl",
        "logd",
        "losetup",
        "lsblk",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-openvpn",
        "luci-app-opkg",
        "luci-app-statistics",
        "luci-base",
        "luci-compat",
        "luci-i18n-attendedsysupgrade-en",
        "luci-i18n-openvpn-en",
        "luci-i18n-statistics-en",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mkf2fs",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openvpn-openssl",
        "openwrt-keyring",
        "opkg",
        "partx-utils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "r8169-firmware",
        "resize2fs",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "screen",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "ext4",
    "client": "luci/git-23.093.42303-58b861d"
}
```

STDERR:

```
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.2/targets/bcm27xx/bcm2711/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.2/targets/bcm27xx/bcm2711/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package cypress-nvram-43455-sdio-rpi-4b.
 * opkg_install_cmd: Cannot install package luci-i18n-attendedsysupgrade-en.
 * opkg_install_cmd: Cannot install package luci-i18n-openvpn-en.
 * opkg_install_cmd: Cannot install package luci-i18n-statistics-en.
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

rany2, 2023-12-17:
See https://github.com/openwrt/luci/commit/7550ad69aff5e57718ee9bc33a2078eccd13fc7c. `luci-ssl` now uses mbedtls, just don't install it as `libustream-wolfssl` simply being installed should be enough.

adavidzh, 2024-01-02:
@rany2, thanks for the link.

I am not sure what you mean with 
>  just don't install it as `libustream-wolfssl` simply being installed should be enough for SSL support.

For instance, I am on OpenWrt 22.03.6 and `admin/system/opkg` tells me that
Package name | Version | Size (.ipk) | Description
-- | -- | -- | --
luci-ssl | git-20.244.36115-e10f954 | 974 B | LuCI with HTTPS support (WolfSSL as SSL backend)

This `luci-ssl` is greyed out, not something I can replace/reinstall, at least not from the LuCI GUI.

So, what is the upgrade path for those in this situation?

adavidzh, 2024-01-02:
Ok, problem solved by following the instructions [here](https://forum.openwrt.org/t/snapshot-update-issue-with-libustream-ssl-so/150784/7).

rany2, 2024-01-02:
> I am not sure what you mean with
> 
> > just don't install it as `libustream-wolfssl` simply being installed should be enough for SSL support.
> 
> For instance, I am on OpenWrt 22.03.6 and `admin/system/opkg` tells me that

Even though you solved it I'll just clarify, you don't need to have `luci-ssl` installed. If you uninstall it and just generate your firmware as usual, it will work. This is because from 22.03 to 23.05, `luci-ssl` went from depending on `wolfssl` to `mbedtls`. The issue here is that because you are requesting `libustream-wolfssl` (as you had it installed from the previous luci-ssl dependency in 22.03) and `libustream-mbedtls` (as this is what luci-ssl in 23.05 depends on now) at the same time; this causes the impossible package selection error.

jfjoly, 2024-01-05:
Thanks for the help! 
I got passed the issue thanks to [those instructions](https://forum.openwrt.org/t/snapshot-update-issue-with-libustream-ssl-so/150784/7). 
Now I have a new set of errors: 
```
Collected errors:
 * opkg_install_cmd: Cannot install package cypress-nvram-43455-sdio-rpi-4b.
 * opkg_install_cmd: Cannot install package luci-i18n-attendedsysupgrade-en.
 * opkg_install_cmd: Cannot install package luci-i18n-openvpn-en.
 * opkg_install_cmd: Cannot install package luci-i18n-statistics-en.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

The issue look similar to https://github.com/openwrt/asu/issues/645

jfjoly, 2024-01-05:
Here is the complete output. 
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20265-f85a79bcb4",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "request_hash": "b0cfc4c06d48827b5c1f45eb7a9c5e81",
    "profile": "raspberrypi,4-compute-module",
    "target": "bcm27xx/bcm2711",
    "version": "23.05.2",
    "packages": [
        "auc",
        "base-files",
        "bash",
        "bcm27xx-gpu-fw",
        "bcm27xx-userland",
        "brcmfmac-firmware-usb",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd",
        "collectd-mod-cpu",
        "collectd-mod-interface",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-memory",
        "collectd-mod-network",
        "collectd-mod-ping",
        "collectd-mod-rrdtool",
        "collectd-mod-sensors",
        "collectd-mod-thermal",
        "cypress-firmware-43455-sdio",
        "cypress-nvram-43455-sdio-rpi-4b",
        "diffutils",
        "dnscrypt-proxy2",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "fdisk",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "httping",
        "iperf",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-brcmfmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-fs-vfat",
        "kmod-hid",
        "kmod-hid-generic",
        "kmod-input-core",
        "kmod-input-evdev",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mdio-devres",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
[... 140 lines cut ...]
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_cortex-a72/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package cypress-nvram-43455-sdio-rpi-4b.
 * opkg_install_cmd: Cannot install package luci-i18n-attendedsysupgrade-en.
 * opkg_install_cmd: Cannot install package luci-i18n-openvpn-en.
 * opkg_install_cmd: Cannot install package luci-i18n-statistics-en.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

jfjoly, 2024-05-06:
I solved it by removing these packages from the selection 
```
cypress-nvram-43455-sdio-rpi-4b
luci-i18n-attendedsysupgrade-en
luci-i18n-openvpn-en
luci-i18n-statistics-en
```

And adding this one
```
brcmfmac-nvram-43455-sdio
```

The `-en` packages are not necessary anymore and `brcmfmac-nvram-43455-sdio` replaces `cypress-nvram-43455-sdio-rpi-4b`.

Source: https://forum.openwrt.org/t/error-with-package-on-23-05-2-firmware-selector-cypress-nvram-43455-sdio-rpi-4b/177882

closed 2024-05-06 by jfjoly