openwrt/asu #706: build failure: error: ext4_allocate_best_fit_partial: failed to allocate 804 blocks, out of space

joshenders, 2024-01-13:
`22.03` => `23.05.2` upgrade on x86 ext4-combined-efi install.

Repro: Removed all wolfssl packages to prevent a package conflict. 

Additional info: `/dev/sda2` is 10GiB.
Build log: [build.log](https://github.com/openwrt/asu/files/13927806/build.log)

Request:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20028-43d71ad93e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": {
        "path": "/sys/firmware/efi",
        "type": "directory",
        "size": 0,
        "mode": 16877,
        "atime": 1705095733,
        "mtime": 1705095733,
        "ctime": 1705095733,
        "inode": 5704,
        "uid": 0,
        "gid": 0
    },
    "request_hash": "0a55691ec30beadd927286ff07958d41",
    "profile": "x86",
    "target": "x86/64",
    "version": "23.05.2",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "curl",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "fdisk",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "grub2",
        "grub2-bios-setup",
        "grub2-efi",
        "iperf3",
        "iptables-nft",
        "iputils-arping",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-amazon-ena",
        "kmod-amd-xgbe",
        "kmod-bnx2",
        "kmod-button-hotplug",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-e1000",
        "kmod-e1000e",
        "kmod-forcedeth",
        "kmod-fs-vfat",
        "kmod-igb",
        "kmod-igc",
        "kmod-input-core",
        "kmod-ixgbe",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-r8169",
        "kmod-slhc",
        "kmod-tg3",
        "libblkid",
        "libblobmsg-json",
        "libcomerr",
        "libext2fs",
        "libf2fs",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libsmartcols",
        "libss",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libuuid",
        "libwolfssl",
        "logd",
        "losetup",
        "lsblk",
        "lua",
        "luci",
        "luci-app-acme",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-ntpc",
        "luci-app-opkg",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mkf2fs",
        "mosh-server",
        "mtd",
        "netcat",
        "netifd",
        "nftables-json",
        "nmap",
        "ntpd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "partx-utils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "python3-speedtest-cli",
        "resize2fs",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "screen",
        "tailscale",
        "tcpdump",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usbutils",
        "usign"
    ],
    "diff_packages": true,
    "filesystem": "ext4",
    "client": "luci/git-23.093.42303-58b861d"
}
```

STDERR:
```
STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.2/targets/x86/64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.2/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-x86_64_musl/root-x86/dev".
Ignoring, exclude it (-e/-ef) to override.
error: ext4_allocate_best_fit_partial: failed to allocate 804 blocks, out of space?
make[3]: *** [/builder/include/image.mk:350: /builder/build_dir/target-x86_64_musl/linux-x86_64/root.ext4] Error 1
make[2]: *** [Makefile:208: build_image] Error 2
make[1]: *** [Makefile:146: _call_image] Error 2
make: *** [Makefile:262: image] Error 2
```

skug67, 2025-07-27:
Same issue trying to upgrade NanoPI R5S from 24.10.1 to 24.10.2

skug67, 2025-08-07:
Update -- was able to upgrade successfully by changing the rootfs size to 256MB

> Same issue trying to upgrade NanoPI R5S from 24.10.1 to 24.10.2