openwrt/asu #711: Revert "Allow custom UCI defaults: False"

hashcan, 2024-01-22:
Hello world, 
I would like to request to revert the commit that changed the "Allow custom UCI defaults" option from True to False for the https://sysupgrade.openwrt.org website. This option allows us to use a custom uci-default script to configure the image after flashing. This is very useful for setting up things like network, firewall, wireless, and other settings that are specific to our devices and environments.
as I use the "don't keep the current configurations" option to flash the image, I rely on the custom uci-default script to set up the device automatically. Without this option, we have to manually configure the device after flashing, which is tedious and error-prone.

Rinat84, 2024-01-22:
[ImmortalWrt](https://firmware-selector.immortalwrt.org/) this feature works

aparcar, 2024-01-24:
I set the according env variable which I forgot in the last deployment, enjoy.

closed 2024-01-24 by aparcar

AlanFinotty, 2024-07-20:
https://github.com/openwrt/asu/blob/main/asu/config.py#L14

hashcan, 2024-07-21:
@aparcar can you please switch it on again ?

hansenjt91, 2024-07-22:
How does this keep getting turned off?