openwrt/asu #726: `/api/v1/revision` returns different version than `/api/v1/build`

felschr, 2024-01-31:
When I run `auc` it currently fails with "Received inncorrect version":
```console
root@openwrt:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r24939-0b498c3c3f on mediatek/filogic (glinet,gl-mt6000)
Available: SNAPSHOT r24944-906595c262
Requesting package lists...
 base-files: 1560-r24939-0b498c3c3f -> 1560-r24944-906595c262
 libmbedtls: 2.28.5-2 -> 2.28.7-2
 luci-mod-network: git-24.026.05172-0c27557 -> git-24.028.55456-75a2fd2
Are you sure you want to continue the upgrade process? [N/y] y
Requesting build......Current Target: "mediatek/filogic"
Current Architecture: "aarch64"
Current Revision: "r24939-0b498c3c3f"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-mbedtls logd mtd netifd opkg uci uclient-fetch urandom-seed urngd busybox procd procd-ujail procd-seccomp kmod-leds-gpio kmod-gpio-button-hotplug kmod-crypto-hw-safexcel kmod-mt7915e wpad-basic-mbedtls uboot-envtools dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe
Available Profiles:
# [REMOVED LIST OF PROFILES]
Error: Received inncorrect version r24939-0b498c3c3f (requested r24944-906595c262)
Bad message (74)
```

When I tried `luci-attended-sysupgrade` instead, it shows `r24944-906595c262` in the selection and in the following step it'll just show `r24939-0b498c3c3f` as selected.

Here are the specific API requests I made:

```sh
curl 'https://sysupgrade.openwrt.org/api/v1/revision/SNAPSHOT/mediatek/filogic?1706718651673'
```

```sh
curl 'https://sysupgrade.openwrt.org/api/v1/build?1706718731791' -X POST -H 'Content-Type: application/json' --data-raw '{"profile":"glinet,gl-mt6000","target":"mediatek/filogic","version":"SNAPSHOT","packages":["auc","base-files","busybox","ca-bundle","dawn","dnsmasq","dropbear","e2fsprogs","f2fsck","firewall4","fstools","kernel","kmod-crypto-hw-safexcel","kmod-gpio-button-hotplug","kmod-leds-gpio","kmod-mt7915e","kmod-mt7986-firmware","kmod-nft-offload","kmod-usb3","libc","libgcc","libustream-mbedtls","logd","luci","luci-app-attendedsysupgrade","luci-app-dawn","luci-ssl","mkf2fs","mt7986-wo-firmware","mtd","netifd","nftables-json","odhcp6c","odhcpd-ipv6only","opkg","ppp","ppp-mod-pppoe","procd","procd-seccomp","procd-ujail","tailscale","uboot-envtools","uci","uclient-fetch","urandom-seed","urngd","wpad-basic-mbedtls"],"diff_packages":true,"filesystem":"squashfs","client":"luci/git-23.292.39411-021c439"}'
```

felschr, 2024-01-31:
Seems those endpoints are out-of-sync for some reason (caching?).
Perhaps `/api/v1/build` should allow specifying the version in the request so it can try to reevaluate caches or at least return an error if the version can't be found.

boecko, 2024-02-02:
Hi,

i've experienced the same behavior on 3 different devices
```
root@device1:~# auc -n                                                                                    
auc/0.3.2-1                                                                                              
Server:    https://sysupgrade.openwrt.org                                                                
Running:   23.05-SNAPSHOT r23642-6264d12ed8 on ipq806x/generic (zyxel,nbg6817)                           
Available: 23.05-SNAPSHOT r23737-be7d9da7be         
Requesting package lists...                                                                              
..
Error: Received inncorrect version r23728-b5c728948c (requested r23737-be7d9da7be)
Bad message (74)
```

```
root@device2:~# auc -n
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05-SNAPSHOT r23728-b5c728948c on ath79/generic (tplink,archer-a7-v5)
Available: 23.05-SNAPSHOT r23737-be7d9da7be
Requesting package lists...
...
Error: Received inncorrect version r23728-b5c728948c (requested r23737-be7d9da7be)
Bad message (77)
```
```
root@device3:~# auc -n                 
auc/0.3.2-1                                                                                                                                                                                                       
Server:    https://sysupgrade.openwrt.org                                                                                                                                                                         
Running:   23.05-SNAPSHOT r23711-cdc907abed on mediatek/mt7622 (linksys,e8450-ubi)
Available: 23.05-SNAPSHOT r23737-be7d9da7be                                                              
Requesting package lists...       
...
Error: Received inncorrect version r23728-b5c728948c (requested r23737-be7d9da7be)
Bad message (74)
```

device2 was updated successfully three days ago.
The buildserver seems to be broken.

Soromeister, 2024-02-04:
I am getting the same issue. Seems like it's stuck on `r24939-0b498c3c3f`

vwghub, 2024-02-04:
Same error, I'm stuck on `SNAPSHOT r24939-0b498c3c3f` as well.

cherinyy, 2024-02-05:
I suspect the server has stopped syncing.

https://github.com/openwrt/asu/issues/730

janumix, 2024-02-06:
Also stuck on r24939-0b498c3c3f with no chance to install other pkg. Is there any workaround ?

cherinyy, 2024-02-06:
> Also stuck on r24939-0b498c3c3f with no chance to install other pkg. Is there any workaround ?

No, you might consider compiling the firmware yourself, or setting up the asu service yourself.

vwghub, 2024-02-07:
> Also stuck on r24939-0b498c3c3f with no chance to install other pkg. Is there any workaround ?

I am working around the issue by downloading the latest base image file from the firmware selector page then logging with ssh and editing /etc/config/network with vi .  You can get a copy from your backup archive.  This gets me connected to the internet.  Then I can add back my packages with opkg and use LuCI to restore my settings.  

I'm doing this in Mac OS X using terminal.  I think you use putty in Windows.  It's not too difficult but you need to manually configure your IP a couple of times, and find and delete the known_hosts file to regain access to 192.168.1.1. after the initial flash then run commands  until you can get back to LuCI.  

You can always experiment and recover if you have saved a prior custom build image.  Just unplug the router and hold down the reset key for 10 seconds while plugging the power back in (until the router light turns white) which will take you to GL.inet's uboot web interface. 

Hope this helps.

cherinyy, 2024-02-07:
https://github.com/openwrt/asu/releases/tag/v0.8.0

This seems to be a problem caused by developers who are trying to switch services to run inside containers.

It is not clear if the problem is because it has not been migrated, is in the process of being migrated or has occurred after migration.

aparcar, 2024-02-12:
Should be working again, Docker containers were updated from an outdated mirror. Please reopen if issue still exists.

Yax, 2024-05-25:
@aparcar I think this issue is happening again:
```
auc -n -b 23.05 -B 23.05-SNAPSHOT
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.3 r23809-234f1a2efa on ramips/mt7621 (ubnt,edgerouter-x-sfp)
Available: 23.05-SNAPSHOT r23880-28bf0137b4
Requesting package lists...
 kmod-nft-fib: 5.15.150-1 -> 5.15.158-1
[................]
 kmod-slhc: 5.15.150-1 -> 5.15.158-1
Are you sure you want to continue the upgrade process? [N/y] y
Current Target: "ramips/mt7621"
Current Architecture: "mipsel"
Current Revision: "r23893-08becaf62f"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-mbedtls logd mtd netifd opkg uci uclient-fetch urandom-seed urngd busybox procd procd-ujail procd-seccomp kmod-leds-gpio kmod-gpio-button-hotplug wpad-basic-mbedtls uboot-envtools dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe
Available Profiles:

adslr_g7:
    ADSLR G7
    Packages: kmod-mt7615-firmware -uboot-envtools
    hasImageMetadata: 1
    SupportedDevices: adslr,g7
[................]
zyxel_wsm20:
    ZyXEL WSM20
    Packages: kmod-mt7915-firmware
    hasImageMetadata: 1
    SupportedDevices: zyxel,wsm20
Error: Received inncorrect version r23893-08becaf62f (requested r23880-28bf0137b4)
Bad message (77)
```