openwrt/asu #732: Can't upgrade from SNAPSHOT - r24936-60ffcfdabc to SNAPSHOT - r25004-b463737826

floatingpurr, 2024-02-05:
As described in the title, I can't upgrade from SNAPSHOT - r24936-60ffcfdabc to SNAPSHOT - r25004-b463737826

Here's the request:
```json
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24936-60ffcfdabc",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "zyxel,ex5601-t0-ubootmod",
    "target": "mediatek/filogic",
    "version": "SNAPSHOT",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "curl",
        "ddns-scripts",
        "ddns-scripts-services",
        "dnsmasq",
        "dropbear",
        "eip197-mini-firmware",
        "ethtool-full",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-authenc",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-des",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-hw-safexcel",
        "kmod-crypto-lib-chacha20",
        "kmod-crypto-lib-chacha20poly1305",
        "kmod-crypto-lib-curve25519",
        "kmod-crypto-lib-poly1305",
        "kmod-crypto-manager",
        "kmod-crypto-md5",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha1",
        "kmod-crypto-sha256",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-i2c-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
[... 279 lines cut ...]
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ppp:
 * 	kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa)
 * opkg_install_cmd: Cannot install package kmod-ppp.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-pppox
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-pppoe
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-pppoe:
 * 	kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa)
 * opkg_install_cmd: Cannot install package kmod-pppoe.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-sfp
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-tun
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-udptunnel4
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-udptunnel6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-usb-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-usb-xhci-hcd
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-usb-xhci-mtk
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-usb3
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-wireguard
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-nf-ipt6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-ip6tables
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

PreK, 2024-02-05:
Also cannot upgrade from SNAPSHOT - r24947-4ae2f43b3a

`
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24947-4ae2f43b3a",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "glinet,gl-mt6000",
    "target": "mediatek/filogic",
    "version": "SNAPSHOT",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "ca-certificates",
        "ddns-scripts-cloudflare",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "eip197-mini-firmware",
        "f2fsck",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-authenc",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-des",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-hw-safexcel",
        "kmod-crypto-manager",
        "kmod-crypto-md5",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha1",
        "kmod-crypto-sha256",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7915e",
        "kmod-mt7986-firmware",
[... 188 lines cut ...]
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-mt7915e
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-mt7915e:
 * 	kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa)
 * opkg_install_cmd: Cannot install package kmod-mt7915e.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-mt7986-firmware
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-slhc
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-ppp
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ppp:
 * 	kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa)
 * opkg_install_cmd: Cannot install package kmod-ppp.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-pppox
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-pppoe
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-pppoe:
 * 	kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa)
 * opkg_install_cmd: Cannot install package kmod-pppoe.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-usb-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-usb3
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb3:
 * 	kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa)
 * opkg_install_cmd: Cannot install package kmod-usb3.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-13427c9c106e234e854403dc29a267fa) for kmod-ipt-core
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2`

jseplae, 2024-02-05:
```javascript
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24699-b196a9f6ce",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "yuncore,ax820",
    "target": "ramips/mt7621",
    "version": "SNAPSHOT",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-authenc",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-des",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-hw-eip93",
        "kmod-crypto-manager",
        "kmod-crypto-md5",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha1",
        "kmod-crypto-sha256",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7915-firmware",
        "kmod-mt7915e",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
[... 213 lines cut ...]
 * opkg_install_cmd: Cannot install package kmod-mt76-connac.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587) for kmod-thermal
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587) for kmod-mt7915e
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587) for kmod-mt7915-firmware
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-mt7915-firmware:
 * 	kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587)
 * opkg_install_cmd: Cannot install package kmod-mt7915-firmware.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587) for kmod-slhc
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587) for kmod-ppp
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ppp:
 * 	kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587)
 * opkg_install_cmd: Cannot install package kmod-ppp.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587) for kmod-pppox
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587) for kmod-pppoe
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-pppoe:
 * 	kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587)
 * opkg_install_cmd: Cannot install package kmod-pppoe.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-b6238e83684a403c67e49fa937e9e587) for kmod-ipt-core
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

</p>
</details>

aparcar, 2024-02-06:
I'm trying to figure out the issue...

floatingpurr, 2024-02-07:
I see. What would be the proper way to upgrade in this scenario?

Crav88, 2024-02-09:
still with the issue?

khanjui, 2024-02-09:
Still.

`Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-nf-reject6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-nf-log
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-nf-log6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-nft-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-nft-offload
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for firewall4:
 * 	kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2)
 * opkg_install_cmd: Cannot install package firewall4.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-crypto-aead
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-crypto-md5
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-crypto-sha1
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-crypto-sha256
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-crypto-hw-eip93
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-crypto-hw-eip93:
 * 	kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2)
 * opkg_install_cmd: Cannot install package kmod-crypto-hw-eip93.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-gpio-button-hotplug
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-leds-gpio
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-crypto-hmac
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-crypto-sha512
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-crypto-rng
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-crypto-gf128
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-mac80211
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-mt7603
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-mt7603:
 * 	kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2)
 * opkg_install_cmd: Cannot install package kmod-mt7603.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-mt76x02-common
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-mt76x2
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-mt76x2:
 * 	kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2)
 * opkg_install_cmd: Cannot install package kmod-mt76x2.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2) for kmod-lib-crc-ccitt
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ppp:
 * 	kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2)
 * opkg_install_cmd: Cannot install package ppp.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ppp-mod-pppoe:
 * 	kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2)
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for qosify:
 * 	kernel (= 5.15.148-1-4516eb38306297a872583f9d591f8ce2)
 * opkg_install_cmd: Cannot install package qosify.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2`

aparcar, 2024-02-12:
Should be working again, Docker containers were updated from an outdated mirror. Please reopen if issue still exists.

closed 2024-02-12 by aparcar