openwrt/asu #735: Error updating bcm27xx/bcm2711 From SNAPSHOT - r24944-906595c262 to SNAPSHOT - r25017-04a8f553e0

jaromanda, 2024-02-06:
Request data:

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24944-906595c262",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "request_hash": "29ef36c590d8539efae568bd519c534e",
    "profile": "raspberrypi,4-compute-module",
    "target": "bcm27xx/bcm2711",
    "version": "SNAPSHOT",
    "packages": [
        "auc",
        "base-files",
        "bcm27xx-eeprom",
        "bcm27xx-gpu-fw",
        "bcm27xx-utils",
        "bcp38",
        "blkid",
        "brcmfmac-nvram-43455-sdio",
        "busybox",
        "ca-bundle",
        "cfdisk",
        "cypress-firmware-43455-sdio",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "ethtool",
        "fdisk",
        "firewall4",
        "fstools",
        "htop",
        "i2c-tools",
        "irqbalance",
        "kernel",
        "kmod-crypto-acompress",
        "kmod-crypto-sha256",
        "kmod-fs-vfat",
        "kmod-gpio-button-hotplug",
        "kmod-i2c-bcm2835",
        "kmod-lib-crc-ccitt",
        "kmod-lib-lzo",
        "kmod-nft-offload",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-r8169",
        "kmod-sound-arm-bcm2835",
        "kmod-sound-core",
        "kmod-usb-hid",
        "kmod-usb-net-lan78xx",
        "kmod-usb-net-rtl8152",
        "kmod-usb-storage-uas",
        "libc",
        "libgcc",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-bcp38",
        "luci-ssl",
        "luci-theme-material",
        "mkf2fs",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
[... 97 lines cut ...]
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d) for kmod-usb-hid
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-hid:
 * 	kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d)
 * opkg_install_cmd: Cannot install package kmod-usb-hid.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d) for kmod-usb-net
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d) for kmod-usb-net-lan78xx
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-net-lan78xx:
 * 	kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d)
 * opkg_install_cmd: Cannot install package kmod-usb-net-lan78xx.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d) for kmod-usb-net-cdc-ether
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d) for kmod-usb-net-rtl8152
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-net-rtl8152:
 * 	kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d)
 * opkg_install_cmd: Cannot install package kmod-usb-net-rtl8152.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d) for kmod-usb-storage
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d) for kmod-usb-storage-uas
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-storage-uas:
 * 	kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d)
 * opkg_install_cmd: Cannot install package kmod-usb-storage-uas.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.75-1-277bb553c83430f1da5027800bd52d8d) for kmod-ipt-core
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

jaromanda, 2024-02-07:
Same issue SNAPSHOT - r25026-1c02b874fa

cherinyy, 2024-02-07:
https://github.com/openwrt/asu/issues/732#issuecomment-1928547622

jaromanda, 2024-02-08:
FYI I did an upgrade about a week ago from a 6+ month old snapshot (r22503-3b212db232) to the current version without an issue. So, it seems something broke in the last week or so

jaromanda, 2024-02-10:
Same issue SNAPSHOT - r25104-f0e732f488

aparcar, 2024-02-12:
Should be working again, Docker containers were updated from an outdated mirror. Please reopen if issue still exists.

closed 2024-02-12 by aparcar