openwrt/asu #740: error building snapshot - server response 'init'

ghoffman9, 2024-02-18:
same error with auc and luci-attendedsysupgrade, for past 24 hours

Error building the firmware image
Server response: init

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r25219-6e2962d4c5",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "request_hash": "e1065c8d3569ddf8f6343aaf41ba85c8",
    "profile": "askey,rt4230w-rev6",
    "target": "ipq806x/generic",
    "version": "SNAPSHOT",
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/undefined"
}

ghoffman9, 2024-02-18:
also same error with firmware-selector

aparcar, 2024-02-18:
Odd I'll have a look later. I keep on getting reports of this error...

briney83, 2024-02-19:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r25233-6da308f4de",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "xiaomi,redmi-router-ax6000-stock",
    "target": "mediatek/filogic",
    "version": "SNAPSHOT",
    "packages": [],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/undefined"
}

ghoffman9, 2024-02-20:
i now have a second router that boot-looped after an apparently-successful build on auc.
It’s a snapshot build, for rt4230w.  possibly of interest, I have been using openssl not mbedtls.
I created the original config on my router by downloading from downloads.openwrt.org, adding packages, and then create an update using auc:

first, i downloaded r25242:
## install updated basic image
cd /tmp  && wget https://downloads.openwrt.org/snapshots/targets/ipq806x/generic/openwrt-ipq806x-generic-askey_rt4230w-rev6-squashfs-sysupgrade.bin && sysupgrade openwrt-ipq806x-generic-askey_rt4230w-rev6-squashfs-sysupgrade.bin

##use openssl and customize
opkg update && opkg install --force-overwrite libustream-openssl openssl-util px5g-standalone libuhttpd-openssl && opkg remove *mbed*  && opkg install wpad-openssl && opkg install luci kmod-usb-printer luci-app-p910nd luci-app-uhttpd auc luci-app-attendedsysupgrade irqbalance luci-app-vnstat2 luci-app-statistics 

this worked to get r25242 running on one of my access points. 
Then doing auc,  an r25250 image was built without reported error. I installed that using auc -y.  this led to boot loop. that built file is attached.
[openwrt-r25250-d4f272ccf3b6-ipq806x-generic-askey_rt4230w-rev6-squashfs-sysupgrade.zip](https://github.com/openwrt/asu/files/14337845/openwrt-r25250-d4f272ccf3b6-ipq806x-generic-askey_rt4230w-rev6-squashfs-sysupgrade.zip)

I really appreciate the tools you have built and I’m only concerned that others not fall into the same problem, since there have been other reports of boot-loop in the past few days.