openwrt/asu #749: Help: Question: Use custom DTS for ASU

NovaPixell, 2024-02-24:
This is probably the wrong channel to ask such a question.

My router model is Zyxel EX5601-T0 (ubootmod version).

I am currently using a OpenWRT firmware with custom-made DTS.
I want to ask if it is possible to create an ASU that supports my custom DTS, so that I can do the sysupgrade in-place.

Thank you for your support in advance.

aparcar, 2024-02-24:
Any reason not to upstream your device support?

NovaPixell, 2024-02-24:
> Member

My device is already supported. I have currently created a custom DTS to be able to enable the swap between RJ45 WAN - SFP WAN. Something that OpenWRT does not natively support.

We are waiting for changes to be made on the PHY Numbering, and then modify the official DTSs

NovaPixell, 2024-02-28:
> @dangowrt do you have a comment on this? Seems a bit niche

I would like to understand precisely, how ASU works.

The device makes a request to the sysupgrade service, and firmware is compiled that matches the identified board-name. is this correct?

In my case, since I use a custom DTS, how can I adapt it? is it feasible?