openwrt/asu #777: Error building the firmware image on Friendly R4S

rvdheuvel72, 2024-03-26:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23630-842932a63d",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "request_hash": "0a8d2e9a4517a6c8226ea37cfdfacc93",
    "profile": "friendlyarm,nanopi-r4s",
    "target": "rockchip/armv8",
    "version": "23.05.2",
    "packages": [
        "acme",
        "acme-acmesh",
        "acme-acmesh-dnsapi",
        "acme-common",
        "alsa-lib",
        "alsa-ucm-conf",
        "alsa-utils",
        "arptables-nft",
        "attendedsysupgrade-common",
        "auc",
        "avahi-dbus-daemon",
        "base-files",
        "bind-check",
        "bind-client",
        "bind-ddns-confgen",
        "bind-dig",
        "bind-dnssec",
        "bind-host",
        "bind-libs",
        "bind-nslookup",
        "bind-rndc",
        "bind-tools",
        "blkid",
        "bmon",
        "bridge",
        "btrfs-progs",
        "busybox",
        "ca-bundle",
        "ca-certificates",
        "cfdisk",
        "cgi-io",
        "cgroupfs-mount",
        "collectd",
        "collectd-mod-cpu",
        "collectd-mod-ethstat",
        "collectd-mod-interface",
        "collectd-mod-irq",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-memory",
        "collectd-mod-network",
        "collectd-mod-rrdtool",
        "collectd-mod-sensors",
        "collectd-mod-thermal",
        "confuse",
        "conntrack",
        "conntrackd",
        "containerd",
        "coremark",
        "coreutils",
        "coreutils-sort",
        "curl",
        "dbus",
        "dbus-utils",
        "ddns-scripts",
        "ddns-scripts-cloudflare",
        "ddns-scripts-cnkuai",
        "ddns-scripts-digitalocean",
[... 1430 lines cut ...]
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://downloads.openwrt.org/releases/23.05.2/targets/rockchip/armv8/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/base). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/luci). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/routing). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_telephony https://downloads.openwrt.org/releases/23.05.2/packages/aarch64_generic/telephony). Skipping.

edgean, 2024-03-27:
Waiting for the answer as well.

aparcar, 2024-03-27:
I'm on vacation will check this soonish

rvdheuvel72, 2024-04-01:
> I had to remove every packages related to containerd and docker for the build to pass, you could probably try that?

Dockers is already removed. Tried removing containerd as well, but still same error.