openwrt/asu #778: Attended Sysupgrade Firmware upgrade issue

AUT-TOM, 2024-03-28:
Request Data:

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": {
        "path": "/sys/firmware/efi",
        "type": "directory",
        "size": 0,
        "mode": 16877,
        "atime": 1711612098,
        "mtime": 1711612098,
        "ctime": 1711612098,
        "inode": 9049,
        "uid": 0,
        "gid": 0
    },
    "request_hash": "c5dd72842a9dcc9530259a45468f2c5c",
    "profile": "default-string-default-string",
    "target": "x86/64",
    "version": "23.05.3",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "ca-certificates",
        "cgi-io",
        "collectd-mod-ping",
        "ddns-scripts",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "fdisk",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "grub2",
        "grub2-bios-setup",
        "grub2-efi",
        "htop",
        "iptables-nft",
        "irqbalance",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-amazon-ena",
        "kmod-amd-xgbe",
        "kmod-bnx2",
        "kmod-button-hotplug",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-e1000",
        "kmod-e1000e",
        "kmod-forcedeth",
        "kmod-fs-vfat",
        "kmod-igb",
        "kmod-igc",
        "kmod-input-core",
        "kmod-ixgbe",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-r8169",
        "kmod-sched-cake",
        "kmod-slhc",
        "kmod-tcp-bbr",
        "kmod-tg3",
        "kmod-tun",
        "kmod-veth",
        "libblkid",
        "libblobmsg-json",
        "libc",
        "libcomerr",
        "libext2fs",
        "libf2fs",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libsmartcols",
        "libss",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libuuid",
        "libwolfssl",
        "logd",
        "losetup",
        "lsblk",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-ddns",
        "luci-app-firewall",
        "luci-app-nlbwmon",
        "luci-app-opkg",
        "luci-app-statistics",
        "luci-app-wireguard",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mkf2fs",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "partx-utils",
        "powertop",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "qrencode",
        "resize2fs",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "tailscale",
        "tc-tiny",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}
```

STDERR:
```
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/x86/64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/x86_64/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-x86_64_musl/root-x86/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

thanks in advance <3

AUT-TOM, 2024-04-03:
hmm, is there a way to see the dependencies on the deprecated wolfssl ? maybe I can kick the packages out, so the upgrade works again

dangowrt, 2024-04-03:
Can you try to update `auc` using `opkg update ; opkg install auc` and retry? Support for client-side automatic package replacements (such as libustream-wolfssl -> libustream-mbedtls) was added a while ago, but maybe your version of `auc` doesn't support that yet. If `auc` is up-to-date (please share version info of `auc` as well) and it happens anyway, lets look into it more.

AUT-TOM, 2024-04-04:
`Package auc (0.3.1-1) installed in root is up to date.`

I also found the commands for checking dependencies:

```
root@OpenWrt:~# opkg whatdepends libwolfssl5.6.3.ee39414e
Root set:
  libwolfssl5.6.3.ee39414e
What depends on root set
        libustream-wolfssl20201210 2022-12-08-9217ab46-2        depends on libwolfssl5.6.3.ee39414e
        px5g-wolfssl 6.2        depends on libwolfssl5.6.3.ee39414e
        luci-ssl git-20.244.36115-e10f954       depends on libustream-wolfssl20201210
root@OpenWrt:~# opkg whatdepends libustream-wolfssl20201210
Root set:
  libustream-wolfssl20201210
What depends on root set
        luci-ssl git-20.244.36115-e10f954       depends on libustream-wolfssl20201210
```

So luci-ssl git-20.244.36115-e10f954 is the troublemaker, that keeps me from upgrading... right now I am using luci with http access. 

Any suggestions how to proceed here without breaking stuff?

AUT-TOM, 2024-04-04:
What I tried:

- install mbedtls-util package
- uninstall *wolfssl* packages (libwolfssl5.6.3.ee39414e & libustream-wolfssl20201210 & px5g-wolfssl)
- Erorr message:
```
[...] 
No packages removed.
Collected errors:
 * pkg_run_script: Internal error: libwolfssl5.6.3.ee39414e has a NULL tmp_unpack_dir.
 * opkg_remove_pkg: not removing package "libwolfssl5.6.3.ee39414e", prerm script failed
The opkg remove command failed with code 255.
```
...but the packages have been uninstalled anyway
- Attended sysupgrade 
But then it hangs with the "Installing the sysupgrade. Do not unpower device!" message

In the syslogs I can only see:
```
Thu Apr  4 10:13:49 2024 user.info upgrade: Image metadata not present
Thu Apr  4 10:13:49 2024 user.info upgrade: Unable to determine upgrade device
```
That's where I am stuck right now :(

AUT-TOM, 2024-09-22:
Sadly I still can't upgrade with auc

here is my auc -c

```
auc/0.3.1-1
Requesting package lists...
 luci-app-statistics: [1;32mgit-23.153.53801-38f5b55 -> git-23.353.71249-ba52db0[0m
 kmod-nft-compat: [1;32m5.10.176-1 -> 5.10.221-1[0m
 luci-app-opkg: [1;32mgit-23.093.42303-e16f620 -> git-24.045.79537-9f5b5dd[0m
 kmod-ixgbe: [1;32m5.10.176-1 -> 5.10.221-1[0m
 ddns-scripts-services: [1;32m2.8.2-25 -> 2.8.2-26[0m
 luci-lib-ip: [1;32mgit-20.250.76529-62505bd -> git-23.311.79310-704a335[0m
 kmod-nft-fib: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-nfnetlink: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-crypto-hash: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-nf-reject6: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-tun: [1;32m5.10.176-1 -> 5.10.221-1[0m
 luci-mod-system: [1;32mgit-23.118.78821-0c02883 -> git-24.067.02339-de5734b[0m
 kmod-nf-flow: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-lib-crc-ccitt: [1;32m5.10.176-1 -> 5.10.221-1[0m
 luci-theme-bootstrap: [1;32mgit-23.093.42704-b47268a -> git-24.208.65546-18f3914[0m
 kmod-pppoe: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-pppox: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-veth: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-nf-reject: [1;32m5.10.176-1 -> 5.10.221-1[0m
 procd-ujail: [1;32m2022-06-01-7a009685-2 -> 2022-06-01-7a009685-3[0m
 base-files: [1;32m1502-r20134-5f15225c1e -> 1506-r20341-591b7e93d3[0m
 kmod-nf-nat: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-input-core: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-crypto-crc32c: [1;32m5.10.176-1 -> 5.10.221-1[0m
 ddns-scripts: [1;32m2.8.2-25 -> 2.8.2-26[0m
 kmod-r8169: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-mii: [1;32m5.10.176-1 -> 5.10.221-1[0m
 procd: [1;32m2022-06-01-7a009685-2 -> 2022-06-01-7a009685-3[0m
 kmod-lib-crc32c: [1;32m5.10.176-1 -> 5.10.221-1[0m
 powertop: [1;32m2.14-2 -> 2.15-1[0m
 luci-mod-status: [1;32mgit-23.158.78816-1c26abc -> git-24.212.61256-910cf4d[0m
 kmod-i2c-core: [1;32m5.10.176-1 -> 5.10.221-1[0m
 liboping: [1;32m1.10.0-3 -> 1.10.0-4[0m
 libmbedtls: [1;32m2.28.7-1 -> 2.28.8-1[0m
 kmod-fs-vfat: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-nft-nat: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-phy-realtek: [1;32m5.10.176-1 -> 5.10.221-1[0m
 luci-app-firewall: [1;32mgit-23.093.42704-1c01c64 -> git-24.067.02332-0337416[0m
 kmod-nf-ipt: [1;32m5.10.176-1 -> 5.10.221-1[0m
 lua: [1;32m5.1.5-10 -> 5.1.5-11[0m
 dropbear: [1;32m2022.82-2 -> 2022.82-4[0m
 kmod-hwmon-core: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-nls-utf8: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-nls-cp437: [1;32m5.10.176-1 -> 5.10.221-1[0m
 procd-seccomp: [1;32m2022-06-01-7a009685-2 -> 2022-06-01-7a009685-3[0m
 kmod-nf-log: [1;32m5.10.176-1 -> 5.10.221-1[0m
 luci-proto-ppp: [1;32mgit-21.158.38888-88b9d84 -> git-24.135.45225-e7a8d09[0m
 luci-base: [1;32mgit-23.119.80898-65ef406 -> git-24.208.65546-18f3914[0m
 mbedtls-util: [1;32m2.28.7-1 -> 2.28.8-1[0m
 kmod-nf-log6: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-tg3: [1;32m5.10.176-1 -> 5.10.221-1[0m
 luci-proto-ipv6: [1;32mgit-21.148.48881-79947af -> git-23.355.78888-e047387[0m
 kmod-nf-conntrack-netlink: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-libphy: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-ipt-core: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-e1000: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-ppp: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-nft-offload: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-pps: [1;32m5.10.176-1 -> 5.10.221-1[0m
 kmod-nf-conntrack: [1;32m5.10.176-1 -> 5.10.221-1[0m
 liblua: [1;32m5.1.5-10 -> 5.1.5-11[0m
 htop: [1;32m3.2.2-1 -> 3.3.0-1[0m
 kmod-tcp-bbr: [1;32m5.10.176-1 -> 5.10.221-1[0m
[... 1339 lines cut ...]
+ [ -d /builder/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-22.03.7-5d4c2bdbd5a2-x86-64-generic-ext4-combined-efi.img.gz.boot/efi/boot ]
+ basename /builder/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-22.03.7-5d4c2bdbd5a2-x86-64-generic-ext4-combined-efi.img.gz.boot/efi/boot
+ baseentry=boot
+ mmd -i /builder/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-22.03.7-5d4c2bdbd5a2-x86-64-generic-ext4-combined-efi.img.gz.kernel ::/efi/boot
+ dos_dircopy /builder/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-22.03.7-5d4c2bdbd5a2-x86-64-generic-ext4-combined-efi.img.gz.boot/efi/boot /efi/boot/
+ local entry
+ local baseentry
+ [ -f /builder/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-22.03.7-5d4c2bdbd5a2-x86-64-generic-ext4-combined-efi.img.gz.boot/efi/boot/bootx64.efi ]
+ mcopy -i /builder/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-22.03.7-5d4c2bdbd5a2-x86-64-generic-ext4-combined-efi.img.gz.kernel /builder/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-22.03.7-5d4c2bdbd5a2-x86-64-generic-ext4-combined-efi.img.gz.boot/efi/boot/bootx64.efi ::/efi/boot/
+ dd if=/builder/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-22.03.7-5d4c2bdbd5a2-x86-64-generic-ext4-combined-efi.img.gz.kernel of=/builder/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-22.03.7-5d4c2bdbd5a2-x86-64-generic-ext4-combined-efi.img.gz bs=512 seek=512 conv=notrunc
32768+0 records in
32768+0 records out
16777216 bytes (17 MB, 16 MiB) copied, 0.0459718 s, 365 MB/s
+ rm -f /builder/build_dir/target-x86_64_musl/linux-x86_64/tmp/openwrt-22.03.7-5d4c2bdbd5a2-x86-64-generic-ext4-combined-efi.img.gz.kernel
/builder/staging_dir/host/bin/grub-bios-setup: warning: Your BIOS Boot Partition is under 1 MiB, please increase its size..
done
Invalid argument (22)

```

I think there is a problem with my [expanded root partition](https://openwrt.org/docs/guide-user/installation/openwrt_x86#expanding_root_partition_and_filesystem) that causes this issue.

As there have been some important security updates in the meantime and I am still stuck on an older verison... any suggestions how I can fix this? Or can/should I just install a (fresh) new image via the luci GUI?

thanks in advance