openwrt/asu #780: Error "No data available (61) " when running uac -b 23.05

paulz-gh, 2024-03-28:
I get the following output when I try to upgrade my Edgerouter-X:

root@wrt:~# auc -b 23.05
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   22.03.6 r20265-f85a79bcb4 on ramips/mt7621 (ubnt,edgerouter-x)
Available: 23.05.3 r23809-234f1a2efa
Requesting package lists...
No data available (61)

martipoe, 2024-03-31:
Same here with Netgear R7800:
```
root@OpenWrt:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on ipq806x/generic (netgear,r7800)
Available: 23.05.3 r23809-234f1a2efa
Requesting package lists...
No data available (61)
```

An attempt via LUCI generates the error message "Error building the firmware image, Server response: init"

Stilzchen, 2024-04-01:
Some users reports the same bug in the last few days in closed [issue 364](https://github.com/openwrt/asu/issues/364).

I have the same error for two different targets:
wrt1:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on realtek/rtl838x (zyxel,gs1900-24e)
Available: 23.05.3 r23809-234f1a2efa
Requesting package lists...
No data available (61)

wrt2:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on ipq40xx/generic (aruba,ap-303h)
Available: 23.05.3 r23809-234f1a2efa
Requesting package lists...
No data available (61)

aparcar, 2024-04-01:
Could you please try again?

Stilzchen, 2024-04-01:
Is working again for my two devices and I updated them successfully.

closed 2024-04-01 by aparcar