openwrt/asu #781: Missing Device

reukiodo, 2024-03-28:
I would like to use the ASU on the Vonets VAR11n-300 which does indeed have an official OpenWrt build:
https://downloads.openwrt.org/releases/19.07.10/targets/ramips/mt7620/openwrt-19.07.10-ramips-mt7620-vonets_var11n-300-initramfs-kernel.bin

However, even after selecting 19.07.10 in the version dropdown, the corresponding https://firmware-selector.openwrt.org/data/19.07.10/overview.json file downloaded doesn't list this device, making it impossible to generate a custom firmware to flash.

aparcar, 2024-04-02:
Heartbreaking but the device is to small to handle ever growing Kernels.

https://github.com/openwrt/openwrt/commit/d7d46da938e3f5c4ace815870c95f67d8b663ebe

closed 2024-04-02 by aparcar

reukiodo, 2024-04-03:
Yeah, that makes sense for newer OpenWrt versions, but I am talking about sysupgrade of older OpenWrt versions... ones where they still run and work fine. The firmware selector is disabled for the older versions as well as the newer versions.

aparcar, 2024-04-03:
The ASU project can't support such old versions, sorry :/