openwrt/asu #782: Attended Upgrade: Error building the firmware image, Impossible package selection

znoteer, 2024-03-28:
Tried upgrading from 22.03.5 to 23.05.3 using Attended Upgrade. I got an error dialogue with the title **Error building the firmware image** and a line saying `Server response: Error: Impossible package selection` and request to report the issue with the following information.

I note that wireguard does not appear in the list of packages even though it is installed and configured and visible in luci under installed software. I'm not sure if there is anything else that is missing. I checked for wireguard because I was bitten once before by it not being reinstalled after flashing an upgrade.

Any advice as to how to pass to 23.05.3 would be much appreciated. Thank you.

Request Data:

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "request_hash": "f351180a58399987e91b754125009110",
    "profile": "tplink,archer-c7-v5",
    "target": "ath79/generic",
    "version": "23.05.3",
    "packages": [
        "ath10k-board-qca988x",
        "ath10k-firmware-qca988x-ct",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb2",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-wireguard",
        "luci-base",
        "luci-i18n-attendedsysupgrade-en",
        "luci-i18n-wireguard-en",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-proto-wireguard",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "swconfig",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}
```

STDERR:

```
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-i18n-attendedsysupgrade-en.
 * opkg_install_cmd: Cannot install package luci-i18n-wireguard-en.
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-mips_24kc_musl/root-ath79/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

znoteer, 2024-03-31:
I just noticed something in the error output. Near the end the output says:

```
* check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-mips_24kc_musl/root-ath79/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
```

However, package `libustream-mbedtls20201210` is not installed on my system, nor have I asked for it to be installed. So it shouldn't be trying to install any files.

Any advice with this new observation?

dannil, 2024-04-02:
The default crypto library was swapped from `wolfssl` to `mbedtls` with 23.05 (see https://forum.openwrt.org/t/openwrt-23-05-0-first-stable-release/174239#switch-from-wolfssl-to-mbedtls-as-default-4), and `luci-ssl` depends on `mbedtls` (it says so in the log), but since you already have `wolfssl` (`libustream-wolfssl` specifically) this package selection isn't possible since they would conflict with each other. You need to swap out `wolfssl` to `mbedtls` in all your packages (that I don't know how to do nicely, I've never used asu between major versions).

dannil, 2024-04-02:
Regarding the language packages (ending with `-en`) those should be able to just be dropped since they're now baked into respective package to my knowledge.

JcMinarro, 2024-04-03:
> The default crypto library was swapped from `wolfssl` to `mbedtls` with 23.05 (see https://forum.openwrt.org/t/openwrt-23-05-0-first-stable-release/174239#switch-from-wolfssl-to-mbedtls-as-default-4), and `luci-ssl` depends on `mbedtls` (it says so in the log), but since you already have `wolfssl` (`libustream-wolfssl` specifically) selected this package selection isn't possible since they would conflict with each other. You need to swap out `wolfssl` to `mbedtls` in all your packages (that I don't know how to do nicely, I've never used asu between major versions).
> 
> If possible (big if, never done it myself), you could maybe force every package that defaults to using `mbedtls` to use the `wolfssl` variant but since `mbedtls` is the default in 23.05 and going forwards, I would advise against that.

Thanks.

It works.

I needed to uninstall `libustream-wolfssl` and install `libustream-mbedtls`.
The only problem is that, whenever you uninstall `libustream-wolfssl`, opkg is not able to update/download any package.
I needed to download `libustream-mbedtls` from a different router (I have two routers in my house) by using `opkg download <package name>`, transfer it to the router I was upgrading, and then install it locally by `opkg install local-package-name.ipk`
After that, I ran the Attended Sysupgrade from the router webpage and it was upgraded properly

znoteer, 2024-04-03:
I had the same problem as [JcMinarro](https://github.com/JcMinarro) when I removed wolfssl.  I didn't have a second router, so I connected my laptop directly to the ISP's cable box and downloaded 23.05.3. I flashed that and restored from backup and installed missing packages including attendedsysupgrade. 

All appears to be back in order. 

Thank you.

closed 2024-04-03 by znoteer