openwrt/asu #783: 23.05.3: opkg_install_cmd: Cannot install package ntpclient.

ulrich-mayer, 2024-03-29:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23630-842932a63d",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "linksys,e8450-ubi",
    "target": "mediatek/mt7622",
    "version": "23.05.3",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "ca-certificates",
        "cgi-io",
        "diffutils",
        "dnsmasq-full",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "irqbalance",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7615-common",
        "kmod-mt7615-firmware",
        "kmod-mt7615e",
        "kmod-mt7622-firmware",
        "kmod-mt7915-firmware",
        "kmod-mt7915e",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb3",
        "less",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-openssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-dawn",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-uhttpd",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-lib-px5g",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "ntpclient",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssl-util",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "vim-full",
        "wireless-regdb",
        "wpad-openssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.306.39494-6f7b129"
}

STDERR:

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/mediatek/mt7622/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/23.05.3/targets/mediatek/mt7622/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package ntpclient.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

dannil, 2024-04-02:
It was removed with https://github.com/openwrt/packages/pull/23569, delete it.

ulrich-mayer, 2024-04-03:
Thank you for pointing me to this change.

closed 2024-04-03 by ulrich-mayer