openwrt/asu #784: "Duplicate src declaration" errors requesting build for RaspberryPi 4, 23.05.3

dzm06, 2024-03-29:
Building 23.05.3 for Raspberry 4 failing:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23630-842932a63d",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "raspberrypi,4-model-b",
    "target": "bcm27xx/bcm2711",
    "version": "23.05.3",
    "packages": [
        "alpine-repositories",
        "apk",
        "attendedsysupgrade-common",
        "base-files",
        "bash",
        "bcm27xx-gpu-fw",
        "bcm27xx-userland",
        "bind-dig",
        "bind-libs",
        "block-mount",
        "brcmfmac-nvram-43455-sdio",
        "busybox",
        "ca-bundle",
        "cfdisk",
        "cgi-io",
        "collectd",
        "collectd-mod-cpu",
        "collectd-mod-cpufreq",
        "collectd-mod-ethstat",
        "collectd-mod-exec",
        "collectd-mod-interface",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-memory",
        "collectd-mod-network",
        "collectd-mod-rrdtool",
        "collectd-mod-sensors",
        "curl",
        "cypress-firmware-43455-sdio",
        "darkstat",
        "diffutils",
        "dnsmasq-full",
        "dropbear",
        "dtc",
        "e2fsprogs",
        "f2fs-tools",
        "f2fsck",
        "fail2ban",
        "firewall4",
        "fstools",
        "fwtool",
        "gcc",
        "gdisk",
        "getrandom",
        "git",
        "git-http",
        "hostapd-common",
        "hostapd-utils",
        "htop",
        "httping",
        "https-dns-proxy",
        "iftop",
        "ip-tiny",
        "ip6tables-zz-legacy",
        "iperf3",
        "ipset",
        "iptables-mod-conntrack-extra",
[... 663 lines cut ...]
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://downloads.openwrt.org/releases/23.05.3/targets/bcm27xx/bcm2711/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/base). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/luci). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/routing). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_telephony https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/telephony). Skipping.

qyz, 2024-03-30:
For what it's worth, I get the same error when trying to upgrade a Raspberry Pi 3 from 23.05.0. Output attached.
[attended-sysupgrade-error-raspi-3.txt](https://github.com/openwrt/asu/files/14811306/attended-sysupgrade-error-raspi-3.txt)

qyz, 2024-03-30:
Problem seems to be in the firmware builder. https://firmware-selector.openwrt.org produces the same error, when I try to build the custom image.

aparcar, 2024-04-01:
Sorry but I'm unable to reproduce this.

dzm06, 2024-04-01:
I can repro it every time with the package list above. 

Here is the package list in copy/pasteable form if you want to plonk it into the firmware-selector to repro it there (same error as attended-sysupgrade):

I spent some time over the weekend seeing if I could identify the offending package (or packages) but was never able to definitively identify one. I did discover that if I go to the firmware-selector and just provide the high-level packages I want (e.g. collectd, statistics) and let the build engine find all the dependencies then I seem to get a clean build. Somehow the problem stems from the package list that attended-sysupgrade is pulling from opkg when it determines what packages it wants baked into the build.

alpine-repositories
apk
ar
attendedsysupgrade-common
base-files
bash
bcm27xx-gpu-fw
bcm27xx-userland
bind-dig
bind-libs
binutils
block-mount
brcmfmac-firmware-usb
brcmfmac-nvram-43455-sdio
busybox
ca-bundle
cfdisk
cgi-io
collectd
collectd-mod-cpu
collectd-mod-cpufreq
collectd-mod-ethstat
collectd-mod-exec
collectd-mod-interface
collectd-mod-iwinfo
collectd-mod-load
collectd-mod-memory
collectd-mod-network
collectd-mod-rrdtool
collectd-mod-sensors
curl
cypress-firmware-43455-sdio
darkstat
dawn
diffutils
dnsmasq-full
dropbear
dtc
e2fsprogs
f2fs-tools
f2fsck
fail2ban
firewall4
fstools
fwtool
gcc
gdisk
getrandom
git
git-http
hostapd-common
hostapd-utils
htop
httping
https-dns-proxy
iftop
ip-tiny
ip6tables-zz-legacy
iperf3
ipset
iptables-mod-conntrack-extra
iptables-mod-ipopt
iptables-zz-legacy
irqbalance
iw
iwinfo
jansson
jq
jshn
jsonfilter
kernel
kmod-brcmfmac
kmod-brcmutil
kmod-cfg80211
kmod-crypto-acompress
kmod-crypto-aead
kmod-crypto-ccm
kmod-crypto-cmac
kmod-crypto-crc32
kmod-crypto-crc32c
kmod-crypto-ctr
kmod-crypto-gcm
kmod-crypto-gf128
kmod-crypto-ghash
kmod-crypto-hash
kmod-crypto-hmac
kmod-crypto-kpp
kmod-crypto-lib-chacha20
kmod-crypto-lib-chacha20poly1305
kmod-crypto-lib-curve25519
kmod-crypto-lib-poly1305
kmod-crypto-manager
kmod-crypto-null
kmod-crypto-rng
kmod-crypto-seqiv
kmod-crypto-sha256
kmod-crypto-sha512
kmod-fixed-phy
kmod-fs-ext4
kmod-fs-f2fs
kmod-fs-vfat
kmod-hid
kmod-hid-generic
kmod-hwmon-core
kmod-hwmon-raspberrypi
kmod-i2c-algo-bit
kmod-i2c-bcm2835
kmod-i2c-core
kmod-i2c-gpio
kmod-input-core
kmod-input-evdev
kmod-ip6tables
kmod-ipt-conntrack
kmod-ipt-conntrack-extra
kmod-ipt-core
kmod-ipt-ipopt
kmod-ipt-ipset
kmod-ipt-raw
kmod-lib-crc-ccitt
kmod-lib-crc16
kmod-lib-crc32c
kmod-lib-lzo
kmod-libphy
kmod-mac80211
kmod-mdio-devres
kmod-mii
kmod-mmc
kmod-nf-conncount
kmod-nf-conntrack
kmod-nf-conntrack-netlink
kmod-nf-conntrack6
kmod-nf-flow
kmod-nf-ipt
kmod-nf-ipt6
kmod-nf-log
kmod-nf-log6
kmod-nf-nat
kmod-nf-reject
kmod-nf-reject6
kmod-nfnetlink
kmod-nft-core
kmod-nft-fib
kmod-nft-nat
kmod-nft-offload
kmod-nls-base
kmod-nls-cp437
kmod-nls-iso8859-1
kmod-nls-utf8
kmod-of-mdio
kmod-phy-microchip
kmod-phy-realtek
kmod-ppp
kmod-pppoe
kmod-pppox
kmod-r8169
kmod-scsi-core
kmod-slhc
kmod-sound-arm-bcm2835
kmod-sound-core
kmod-tun
kmod-udptunnel4
kmod-udptunnel6
kmod-usb-core
kmod-usb-ehci
kmod-usb-hid
kmod-usb-net
kmod-usb-net-cdc-ether
kmod-usb-net-cdc-ncm
kmod-usb-net-lan78xx
kmod-usb-net-rtl8152
kmod-usb-storage
kmod-usb-storage-uas
kmod-usb-xhci-hcd
kmod-usb2
kmod-usb3
kmod-wireguard
libatomic
libbfd
libblkid
libblobmsg-json
libbz2
libc
libcap
libcares
libcomerr
libctf
libcurl
libev
libevdev
libevent2
libevent2-core
libext2fs
libf2fs
libfdisk
libffi
libgcc
libgcrypt
libgmp
libgpg-error
libip4tc
libip6tc
libiperf3
libipset
libiptext
libiptext6
libiwinfo-data
libiwinfo-lua
libiwinfo
libjpeg-turbo
libjson-c
libjson-script
libltdl
liblua
liblucihttp-lua
liblucihttp-ucode
liblucihttp
libmbedtls
libmnl
libmount
libncurses
libnetfilter-conntrack
libnettle
libnfnetlink
libnftnl
libnghttp2
libnl-tiny
libopcodes
libopenssl-conf
libopenssl
libpcap
libpcre
libpcre2
libpng
libpopt
libpthread
libpython3
libqrencode
libreadline
librrd1
librt
libsensors
libsmartcols
libsqlite3
libss
libstdcpp
libsysfs
libtirpc
libubox
libubus-lua
libubus
libuci-lua
libuci
libuclient
libucode
libudev-zero
libusb-1.0
libustream-mbedtls
libuuid
libuv
libwebp
libxtables
libzstd
lm-sensors
logd
lsof
lua
luci
luci-app-advanced-reboot
luci-app-attendedsysupgrade
luci-app-dawn
luci-app-firewall
luci-app-https-dns-proxy
luci-app-mwan3
luci-app-nlbwmon
luci-app-opkg
luci-app-statistics
luci-base
luci-lib-base
luci-lib-ip
luci-lib-json
luci-lib-jsonc
luci-lib-nixio
luci-light
luci-lua-runtime
luci-mod-admin-full
luci-mod-network
luci-mod-status
luci-mod-system
luci-proto-ipv6
luci-proto-ppp
luci-proto-wireguard
luci-ssl
luci-theme-bootstrap
make
mesh11sd
mkf2fs
mtd
mtr-json
mwan3
netifd
nftables-json
nlbwmon
objdump
odhcp6c
odhcpd-ipv6only
openssl-util
openwrt-keyring
opkg
partx-utils
pingcheck
ppp
ppp-mod-pppoe
procd
procd-seccomp
procd-ujail
px5g-mbedtls
python3-base
python3-ctypes
python3-distutils
python3-email
python3-light
python3-logging
python3-pkg-resources
python3-sqlite3
python3-urllib
qrencode
r8152-firmware
r8169-firmware
resize2fs
resolveip
rpcd
rpcd-mod-file
rpcd-mod-iwinfo
rpcd-mod-luci
rpcd-mod-rpcsys
rpcd-mod-rrdns
rpcd-mod-ucode
rrdtool1
rsync
shadow-common
shadow-groupadd
shadow-useradd
shadow-usermod
sudo
sysfsutils
terminfo
tmux
tune2fs
ubox
ubus
ubusd
uci
uclient-fetch
ucode
ucode-mod-fs
ucode-mod-html
ucode-mod-lua
ucode-mod-math
ucode-mod-nl80211
ucode-mod-rtnl
ucode-mod-ubus
ucode-mod-uci
ucode-mod-uloop
uhttpd
uhttpd-mod-ubus
umdns
urandom-seed
usbids
usbutils
usign
wireguard-tools
wireless-regdb
wpad-mesh-mbedtls
xtables-legacy
zlib
zsh

dzm06, 2024-04-01:
I'm sorry, I'm not entirely sure I understand what you're saying.

When you say "the package selection is too big for the raspberry," do you mean that there is a limitation in the number of packages the raspberrypi can run, or that the requested packages exceed the filesystem size that is allocated to store the packages that have been requested? So the solution is to increase that disk image on the build machine?

The build error you provide seems to indicate the latter (that the disk image is too small), but I'm not positive.

Thanks for your help!

aparcar, 2024-04-01:
Per default OpenWrt images have a root filesystem size of 100MB, if allowed by the hardware itself. For devices with storage cards this limitation still applies per default. There is an extra, not fully expoed API call called `rootfs_size_mb` which allows to increase the size, however it comes with a bit of a drawback. You can somewhat overload the server by allowing arbitrary root filesystem sizes, so ideally you'd use something like [extroot](https://openwrt.org/docs/guide-user/additional-software/extroot_configuration) and only install the base packages you need.

Maybe @dangowrt got a better idea how to handle this?

dzm06, 2024-04-01:
Makes sense. Thanks @aparcar .

Seems like a good opportunity for me to revisit whether I use all the things I've installed at one time or another.

aparcar, 2024-04-02:
Sounds good to me.

dzm06, 2024-04-02:
Your call as to whether this is closable or not. To me it seems as though there is still an underlying problem: small file system + bad error reporting. It would be really nice if, at a minimum, an error message could be returned that indicates that the size of the build is what's causing the problem. It would be even better if the system offered:

* Error message indicating size problems
* Size of each package
* Dependency graph of each package (e.g. these packages depend on this one and will be de-selected if you de-select this one)
* De-select package, and package groups

A fella can dream, right?

appleimperio, 2024-04-16:
Having  the same issue.
````
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23630-842932a63d",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "raspberrypi,4-compute-module",
    "target": "bcm27xx/bcm2711",
    "version": "23.05.3",
    "packages": [
        "adguardhome",
        "auc",
        "base-files",
        "bcm27xx-gpu-fw",
        "block-mount",
        "brcmfmac-nvram-43455-sdio",
        "busybox",
        "ca-bundle",
        "cfdisk",
        "cloudflared",
        "collectd-mod-sensors",
        "collectd-mod-thermal",
        "curl",
        "cypress-firmware-43455-sdio",
        "ddns-scripts-cloudflare",
        "dnsmasq",
        "docker",
        "docker-compose",
        "dockerd",
        "dropbear",
        "e2fsprogs",
        "f2fs-tools",
        "fdisk",
        "firewall4",
        "fstools",
        "iperf3",
        "iwinfo",
        "kernel",
        "kmod-brcmfmac",
        "kmod-fs-ext4",
        "kmod-fs-f2fs",
        "kmod-fs-vfat",
        "kmod-igc",
        "kmod-nft-offload",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-r8169",
        "kmod-sound-arm-bcm2835",
        "kmod-sound-core",
        "kmod-usb-hid",
        "kmod-usb-net-lan78xx",
        "libc",
        "libcurl",
        "libgcc",
        "libustream-mbedtls",
        "logd",
        "losetup",
        "lsblk",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-cloudflared",
        "luci-app-ddns",
        "luci-app-dockerman",
        "luci-app-statistics",
        "luci-base",
        "luci-mod-network",
        "luci-mod-status",
[... 1058 lines cut ...]
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/aarch64_cortex-a72/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/dev".
Ignoring, exclude it (-e/-ef) to override.
error: ext4_allocate_best_fit_partial: failed to allocate 395 blocks, out of space?
make[3]: *** [/builder/include/image.mk:350: /builder/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/root.ext4] Error 1
make[2]: *** [Makefile:208: build_image] Error 2
make[1]: *** [Makefile:146: _call_image] Error 2
make: *** [Makefile:262: image] Error 2
````

maxwowpow, 2024-04-28:
Same here, all packages are needed. 
Plus 1 big block list or similar and it's only the matter of time the 100mb limit will be hit.
For raspberrys it seems reasonable.
Thanks.