openwrt/asu #790: [raspberrypi,4-model-b] | Error building the firmware image

mrinmaydhar, 2024-04-09:
Server response: Error: Selected packages exceed device storage

Request Data: 
[Request Data.json](https://github.com/openwrt/asu/files/14918941/Request.Data.json)

STDOUT: 
[STDOUT.txt](https://github.com/openwrt/asu/files/14918943/STDOUT.txt)

STDERR: 
[STDERR.txt](https://github.com/openwrt/asu/files/14918946/STDERR.txt)

`error: ext4_allocate_best_fit_partial: failed to allocate 2246 blocks, out of space?`
`make[3]: *** [/builder/include/image.mk:350: /builder/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/root.ext4] Error 1`
`make[2]: *** [Makefile:208: build_image] Error 2`
`make[1]: *** [Makefile:146: _call_image] Error 2`
`make: *** [Makefile:262: image] Error 2`

mrinmaydhar, 2024-04-09:
#784 similar error. If required, can close this as duplicate (however the error is different in this case)

Moreover, current squashfs usage is 94.13MiB / 1.68GiB (5%). If this is cause for issue, please do let me know. Thank you

RicardoJCMarques, 2024-05-10:
I've got the same error trying to upgrade Snapshot on a RP5.. I'm just getting used to openwrt but I expanded the default partition size on my sd-card. Guessing that might have something to do with this? Does it need a bigger boot partition too?

mrinmaydhar, 2024-05-11:
As mentioned by @aparcar [here](https://github.com/openwrt/asu/issues/784#issuecomment-2030546498), the default root allocation is for 100MB, and any expand operations that are performed are after the fact (after initial boot, etc.). So, expanding the boot partition would not be of help (90% sure).

If an attended sysupgrade is something of a priority, then you could remove some packages that you don't need.

I, however, have a custom build environment setup to deal with this limitation.

mrinmaydhar, 2024-09-28:
Sorry I won't be able to help you with this anymore. I've moved on to opnsense for my needs.