openwrt/asu #795: Stuck on version r25899-98235e60ed

Soromeister, 2024-04-15:
Received below error when using the API:

```
HTTP/1.1 500 Internal Server Error
Access-Control-Allow-Headers: *
Access-Control-Allow-Methods: POST, GET, OPTIONS
Access-Control-Allow-Origin: *
Connection: close
Content-Length: 10266
Content-Type: application/json
Date: Mon, 15 Apr 2024 07:08:22 GMT
Server: Caddy

{
  "detail": "Error: Received inncorrect version r25899-98235e60ed (requested r25933-cab2e1de0d)",
  "enqueued_at": "2024-04-15T07:08:13.040405Z",
  "error": "Traceback (most recent call last):\n  File \"/usr/local/lib/python3.11/site-packages/rq/worker.py\", line 1431, in perform_job\n    rv = job.perform()\n         ^^^^^^^^^^^^^\n  File \"/usr/local/lib/python3.11/site-packages/rq/job.py\", line 1280, in perform\n    self._result = self._execute()\n                   ^^^^^^^^^^^^^^^\n  File \"/usr/local/lib/python3.11/site-packages/rq/job.py\", line 1317, in _execute\n    result = self.func(*self.args, **self.kwargs)\n             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/app/asu/build.py\", line 70, in build\n    report_error(\n  File \"/app/asu/common.py\", line 315, in report_error\n    raise\nRuntimeError: No active exception to reraise\n",
  "request": {
    "base_container": "ghcr.io/openwrt/imagebuilder",
    "branch": "SNAPSHOT",
    "branch_data": {
      "enabled": true,
      "extra_keys": [],
      "extra_repos": {},
      "git_branch": "master",
      "name": "SNAPSHOT",
      "package_changes": [
        {
          "revision": 21994,
          "source": "libustream-wolfssl",
          "target": "libustream-mbedtls"
        },
        {
          "revision": 21994,
          "source": "px5g-wolfssl",
          "target": "px5g-mbedtls"
        },
        {
          "revision": 21994,
          "source": "wpad-basic-wolfssl",
          "target": "wpad-basic-mbedtls"
        },
        {
          "revision": 21994,
          "source": "libustream-wolfssl",
          "target": "libustream-mbedtls"
        },
        {
          "mandatory": true,
          "revision": 20282,
          "source": "kmod-nft-nat6"
        },
        {
          "revision": 18611,
          "source": "firewall",
          "target": "firewall4"
        }
      ],
      "path": "snapshots",
      "path_packages": "snapshots/packages",
      "pubkey": "RWS1BD5w+adc3j2Hqg9+b66CvLR7NlHbsj7wjNVj0XGt/othDgIAOJS+",
      "repos": [
        "base",
        "packages",
        "luci",
        "routing",
        "telephony"
      ],
      "snapshot": true,
      "updates": "dev",
[... 149 lines cut ...]
      "wireless-regdb",
      "wpad-wolfssl",
      "xtables-nft"
    ],
    "profile": "xiaomi_ax9000",
    "public_path": "/mnt/store/asu",
    "repository_allow_list": [
      "http://downloads.openwrt.org",
      "https://downloads.openwrt.org",
      "http://feed.libremesh.org",
      "https://feed.libremesh.org"
    ],
    "request_hash": "7cfd526daaba2c809bf17968da4c635e",
    "target": "qualcommax/ipq807x",
    "version": "SNAPSHOT",
    "version_code": "r25933-cab2e1de0d"
  },
  "request_hash": "7cfd526daaba2c809bf17968da4c635e",
  "status": 500,
  "stderr": "",
  "stdout": "Current Target: \"qualcommax/ipq807x\"\nCurrent Architecture: \"aarch64\"\nCurrent Revision: \"r25899-98235e60ed\"\nDefault Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-mbedtls logd mtd netifd opkg uci uclient-fetch urandom-seed urngd busybox procd procd-ujail procd-seccomp kmod-usb3 kmod-usb-dwc3 kmod-usb-dwc3-qcom kmod-leds-gpio kmod-gpio-button-hotplug kmod-qca-nss-dp kmod-ath11k-ahb wpad-basic-mbedtls uboot-envtools e2fsprogs kmod-fs-ext4 losetup kmod-phy-aquantia ath11k-firmware-ipq8074 dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe\nAvailable Profiles:\n\narcadyan_aw1000:\n    Arcadyan AW1000\n    Packages: ipq-wifi-arcadyan_aw1000 kmod-spi-gpio kmod-gpio-nxp-74hc164 kmod-usb-serial-option uqmi\n    hasImageMetadata: 1\n    SupportedDevices: arcadyan,aw1000\nbuffalo_wxr-5950ax12:\n    Buffalo WXR-5950AX12\n    Packages: ipq-wifi-buffalo_wxr-5950ax12\n    hasImageMetadata: 1\n    SupportedDevices: buffalo,wxr-5950ax12\ncmcc_rm2-6:\n    CMCC RM2-6\n    Packages: ipq-wifi-cmcc_rm2-6 kmod-hwmon-gpiofan\n    hasImageMetadata: 1\n    SupportedDevices: cmcc,rm2-6\ncompex_wpq873:\n    Compex WPQ873\n    Packages: ipq-wifi-compex_wpq873\n    hasImageMetadata: 1\n    SupportedDevices: compex,wpq873\ndynalink_dl-wrx36:\n    Dynalink DL-WRX36\n    Packages: ipq-wifi-dynalink_dl-wrx36\n    hasImageMetadata: 1\n    SupportedDevices: dynalink,dl-wrx36\nedgecore_eap102:\n    Edgecore EAP102\n    Packages: ipq-wifi-edgecore_eap102\n    hasImageMetadata: 1\n    SupportedDevices: edgecore,eap102\nedimax_cax1800:\n    Edimax CAX1800\n    Packages: ipq-wifi-edimax_cax1800\n    hasImageMetadata: 1\n    SupportedDevices: edimax,cax1800\nlinksys_mx4200v1:\n    Linksys MX4200 v1\n    Packages: kmod-leds-pca963x ipq-wifi-linksys_mx4200 kmod-bluetooth\n    hasImageMetadata: 1\n    SupportedDevices: linksys,mx4200v1\nlinksys_mx4200v2:\n    Linksys MX4200 v2\n    Packages: kmod-leds-pca963x ipq-wifi-linksys_mx4200 kmod-bluetooth\n    hasImageMetadata: 1\n    SupportedDevices: linksys,mx4200v2\nlinksys_mx5300:\n    Linksys MX5300\n    Packages: kmod-leds-pca963x kmod-rtc-ds1307 ipq-wifi-linksys_mx5300 kmod-ath10k-ct ath10k-firmware-qca9984-ct\n    hasImageMetadata: 1\n    SupportedDevices: linksys,mx5300\nnetgear_rax120v2:\n    Netgear RAX120v2\n    Packages: ipq-wifi-netgear_rax120v2 kmod-spi-gpio kmod-spi-bitbang kmod-gpio-nxp-74hc164 kmod-hwmon-g761\n    hasImageMetadata: 1\n    SupportedDevices: netgear,rax120v2\nnetgear_wax218:\n    Netgear WAX218\n    Packages: kmod-spi-gpio kmod-spi-bitbang kmod-gpio-nxp-74hc164 ipq-wifi-netgear_wax218\n    hasImageMetadata: 1\n    SupportedDevices: netgear,wax218\nnetgear_wax620:\n    Netgear WAX620\n    Packages: kmod-spi-gpio kmod-gpio-nxp-74hc164 ipq-wifi-netgear_wax620\n    hasImageMetadata: 1\n    SupportedDevices: netgear,wax620\nnetgear_wax630:\n    Netgear WAX630\n    Packages: kmod-spi-gpio ipq-wifi-netgear_wax630\n    hasImageMetadata: 1\n    SupportedDevices: netgear,wax630\nprpl_haze:\n    prpl Foundation Haze\n    Packages: ath11k-firmware-qcn9074 ipq-wifi-prpl_haze kmod-ath11k-pci mkf2fs f2fsck kmod-fs-f2fs kmod-leds-lp5562\n    hasImageMetadata: 1\n    SupportedDevices: prpl,haze\nqnap_301w:\n    QNAP 301w\n    Packages: ipq-wifi-qnap_301w\n    hasImageMetadata: 1\n    SupportedDevices: qnap,301w\nredmi_ax6:\n    Redmi AX6\n    Packages: ipq-wifi-redmi_ax6\n    hasImageMetadata: 1\n    SupportedDevices: redmi,ax6\nxiaomi_ax3600:\n    Xiaomi AX3600\n    Packages: ipq-wifi-xiaomi_ax3600 kmod-ath10k-ct-smallbuffers ath10k-firmware-qca9887-ct\n    hasImageMetadata: 1\n    SupportedDevices: xiaomi,ax3600\nxiaomi_ax9000:\n    Xiaomi AX9000\n    Packages: ipq-wifi-xiaomi_ax9000 kmod-ath11k-pci ath11k-firmware-qcn9074 kmod-ath10k-ct ath10k-firmware-qca9887-ct\n    hasImageMetadata: 1\n    SupportedDevices: xiaomi,ax9000\nyuncore_ax880:\n    Yuncore AX880\n    Packages: ipq-wifi-yuncore_ax880\n    hasImageMetadata: 1\n    SupportedDevices: yuncore,ax880\nzte_mf269:\n    ZTE MF269\n    Packages: ipq-wifi-zte_mf269\n    hasImageMetadata: 1\n    SupportedDevices: zte,mf269\nzyxel_nbg7815:\n    ZYXEL NBG7815\n    Packages: ipq-wifi-zyxel_nbg7815 kmod-ath11k-pci kmod-bluetooth kmod-hwmon-tmp103\n    hasImageMetadata: 1\n    SupportedDevices: zyxel,nbg7815\n"
}
```

Mirror must be stuck or something. It's been like this for the past days.

aparcar, 2024-04-15:
https://github.com/openwrt/docker/pull/134

Soromeister, 2024-04-15:
Fixed by https://github.com/openwrt/docker/pull/134

closed 2024-04-15 by Soromeister

Yax, 2024-05-25:
@Soromeister this issue has returned:
```
auc -n -b 23.05 -B 23.05-SNAPSHOT
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.3 r23809-234f1a2efa on ramips/mt7621 (ubnt,edgerouter-x-sfp)
Available: 23.05-SNAPSHOT r23880-28bf0137b4
Requesting package lists...
 kmod-nft-fib: 5.15.150-1 -> 5.15.158-1
[................]
 kmod-slhc: 5.15.150-1 -> 5.15.158-1
Are you sure you want to continue the upgrade process? [N/y] y
Current Target: "ramips/mt7621"
Current Architecture: "mipsel"
Current Revision: "r23893-08becaf62f"
Default Packages: base-files ca-bundle dropbear fstools libc libgcc libustream-mbedtls logd mtd netifd opkg uci uclient-fetch urandom-seed urngd busybox procd procd-ujail procd-seccomp kmod-leds-gpio kmod-gpio-button-hotplug wpad-basic-mbedtls uboot-envtools dnsmasq firewall4 nftables kmod-nft-offload odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe
Available Profiles:

adslr_g7:
    ADSLR G7
    Packages: kmod-mt7615-firmware -uboot-envtools
    hasImageMetadata: 1
    SupportedDevices: adslr,g7
[................]
zyxel_wsm20:
    ZyXEL WSM20
    Packages: kmod-mt7915-firmware
    hasImageMetadata: 1
    SupportedDevices: zyxel,wsm20
Error: Received inncorrect version r23893-08becaf62f (requested r23880-28bf0137b4)
Bad message (77)
```