openwrt/asu #800: Cannot determine currently running branch/invalid argument (22)

benhedrington, 2024-04-20:
Unsure how to resolve this issue.

```
root@openwrt:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on x86/64 (generic)
WARNING: cannot determing currently running branch.
Invalid argument (22)
```

Danacus, 2024-04-21:
After some digging, I think I figured it out. This may be a regression of https://github.com/openwrt/asu/commit/85392c0e44e05916a16d437940639ce819f409c3.

@aparcar You removed `path_packages` from the branches API, as it is supposedly never really used. However, if I understood correctly, `auc` will not process any branches that lack this field, as it is considered mandatory: https://github.com/openwrt/packages/blob/master/utils/auc/src/auc.c#L1203-L1207.

Side-note: you added the latest releases in https://github.com/openwrt/asu/commit/6ce5023d32410c82752e043ed2e126eee03b9422, but this does not seem to be live on https://sysupgrade.openwrt.org/json/v1/branches.json yet.

sumerland, 2024-04-21:
Thank you for digging, @Danacus . That probably explains why I could upgrade to 23.05.2 but not to 23.05.3.

benhedrington, 2024-04-21:
@Danacus thank you for helping progress on this bug.

Are you aware of anything I can do on my side to help auc understand my current branch like some sort of argument or setting?

Danacus, 2024-04-22:
I don't think there's anything we can do, since `auc` will not process any of the branches from the sysupgrade server.

I think the only options are:
- fix `auc` to ignore the `path_packages` field
- fix the `asu` server by reverting the commit that removes this field from the branches endpoint, and host your own `asu` server
- wait for a fix to be applied on sysupgrade.openwrt.org
- use the luci interface, which might still work

opty77, 2024-05-04:
Still a thing:

```
root@tplink:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on ath79/generic (tplink,tl-wdr3600-v1)
WARNING: cannot determing currently running branch.
Invalid argument (22)
```

aparcar, 2024-05-04:
Sorry will try to take care of it soonish

aparcar, 2024-05-04:
I think this fixed it, now I just have to deploy the fresh docker containers which takes a bit since the CI is bloated again https://github.com/openwrt/asu/commit/c0104cd849bf5f8d13edc1b7175665615e89d58e

aparcar, 2024-05-05:
Should be fixed, please try again and close this issue if successful.

Danacus, 2024-05-05:
This solved the issue for me. Thanks a lot!

closed 2024-05-05 by aparcar

woeisme, 2024-07-19:
I'm still getting this error

```
# auc -r
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   21.02.1 r16325-88151b8303 on x86/64 (generic)
WARNING: cannot determing currently running branch.
Invalid argument (22)
```

jcgoette, 2025-01-25:
Same:

```
# auc
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   22.03.7 r20341-591b7e93d3 on ipq40xx/generic (netgear,rbr50)
WARNING: cannot determing currently running branch.
Invalid argument (22)
```

Is this potentially a 0.3.1 vs. 0.3.2 issue?

hadifarnoud, 2025-02-17:
same error:
```
root@OpenWrt:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.0-rc1 r23069-e2701e0f33 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
```

hadifarnoud, 2025-02-17:
> same error:
> 
> ```
> root@OpenWrt:~# auc
> auc/0.3.2-1
> Server:    https://sysupgrade.openwrt.org
> Running:   23.05.0-rc1 r23069-e2701e0f33 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
> Invalid argument (22)
> ```

no. I have the same issue