openwrt/asu #841: Error building the firmware image / Server response: Unsupported profile: linksys,wrt1900acs

davidlday, 2024-06-01:
Error building the firmware image

Server response: Unsupported profile: linksys,wrt1900acs

Request Data:
```json
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19803-9a599fee93",
    "advanced_mode": "0",
    "branch": "22.03",
    "efi": null,
    "profile": "linksys,wrt1900acs",
    "target": "mvebu/cortexa9",
    "version": "22.03.5",
    "packages": [
        "acme-dnsapi",
        "addrwatch",
        "arp-scan",
        "arp-scan-database",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "ddns-scripts-cloudflare",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-i18n-acme-en",
        "luci-i18n-ddns-en",
        "luci-i18n-pbr-en",
        "luci-i18n-uhttpd-en",
        "luci-i18n-upnp-en",
        "luci-i18n-wireguard-en",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "mwlwifi-firmware-88w8864",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}
```

davidlday, 2024-06-01:
I tried the approach tried here: https://github.com/openwrt/asu/issues/708

I enabled "Advanced Config" and was able to select 23.05.3, then received a different error.

Request Data:
```json
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19803-9a599fee93",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "request_hash": "b522182cee75d4d1c66095792820c8a0",
    "profile": "linksys,wrt1900acs",
    "target": "mvebu/cortexa9",
    "version": "23.05.3",
    "packages": [
        "acme-dnsapi",
        "addrwatch",
        "arp-scan",
        "arp-scan-database",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "ddns-scripts-cloudflare",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
[... 108 lines cut ...]
Downloading https://downloads.openwrt.org/releases/23.05.3/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package acme-dnsapi.
 * opkg_install_cmd: Cannot install package luci-i18n-acme-en.
 * opkg_install_cmd: Cannot install package luci-i18n-ddns-en.
 * opkg_install_cmd: Cannot install package luci-i18n-pbr-en.
 * opkg_install_cmd: Cannot install package luci-i18n-uhttpd-en.
 * opkg_install_cmd: Cannot install package luci-i18n-upnp-en.
 * opkg_install_cmd: Cannot install package luci-i18n-wireguard-en.
 * check_data_file_clashes: Package libustream-mbedtls20201210 wants to install file /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-wolfssl20201210
 * opkg_install_cmd: Cannot install package luci-ssl.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2
```

I think I understand what's going on. Initially, I had selected an upgrade to 22.05.3, and maybe that's just too old for the build server since v23 is out. But, for whatever reason, without advanced config enabled I didn't even notice the default and only selection was for v22, not v23. Then, when I was able to select v23, the new error tells me the package structure between v22 and v23 for the packages I have installed changed enough to that they cant be automatically mapped between the two.

I don't think either is technically an error or a bug, but the feedback could be more clear. Pushing out the request data in the first instance with no attempt to explain the problem confused me. Pushing out the request data and the STDERR in the second instance is better, but still could be more direct.

davidlday, 2024-06-02:
I made some changes to the package list based on a little research and combing through change logs. Will test it out later.

Seems to me these substitutions should be known and captured somewhere. Seven of these are just package name changes. The rest is due to `luci-ssl` moving from `-wolfssl` to `-mbedtls`, I think.

Replaced:
* acme-dnsapi -> acme-acmesh-dnsapi
* libustream-wolfssl -> libustream-mbedtls
* libwolfssl -> libmbedtls12
* luci-i18n-acme-en -> luci-app-acme
* luci-i18n-ddns-en -> luci-app-ddns
* luci-i18n-pbr-en -> luci-app-pbr
* luci-i18n-uhttpd-en -> luci-app-uhttpd
* luci-i18n-upnp-en -> luci-app-upnp
* luci-i18n-wireguard-en -> luci-app-wireguard
* px5g-wolfssl -> px5g-mbedtls
* wpad-basic-wolfssl -> wpad-basic

davidlday, 2024-11-15:
Quick update - I finally got around to upgrading (yeah, yeah), and with the package substitutions I made above, everything worked like a charm. Closing this out.

tlaurion, 2024-11-17:
@davidlday is this fixed in code or you closed because you changed packages list yourself? If the later, not sure this issue should be closed?

davidlday, 2024-11-19:
@tlaurion - I closed because I updated the package list manually. Reopening, as suggested.

tlaurion, 2024-11-20:
> I made some changes to the package list based on a little research and combing through change logs. Will test it out later.
> 
> Seems to me these substitutions should be known and captured somewhere. Seven of these are just package name changes. The rest is due to `luci-ssl` moving from `-wolfssl` to `-mbedtls`, I think.
> 
> Replaced:
> 1. acme-dnsapi -> acme-acmesh-dnsapi
> 1. libustream-wolfssl -> libustream-mbedtls
> 1. libwolfssl -> libmbedtls12
> 1. luci-i18n-acme-en -> luci-app-acme
> 1. luci-i18n-ddns-en -> luci-app-ddns
> 1. luci-i18n-pbr-en -> luci-app-pbr
> 1. luci-i18n-uhttpd-en -> luci-app-uhttpd
> 1. luci-i18n-upnp-en -> luci-app-upnp
> 1. luci-i18n-wireguard-en -> luci-app-wireguard
> 1. px5g-wolfssl -> px5g-mbedtls
> 1. wpad-basic-wolfssl -> wpad-basic-mbedtls
> 

@aparcar this will need mapping somehow somewhere to fix broken dependencies reported when upgrading firmware between versions of openwrt. 

If not @aparcar, who should be tagged here for which component if not here?