openwrt/asu #848: auc - updates not possible with SNAPSHOTS

GoetzGoerisch, 2024-06-03:
Related issue:
https://github.com/openwrt/openwrt/issues/15619

brada4, 2024-06-08:
Long thread in forums.
https://forum.openwrt.org/t/auc-sysupgrade-does-not-work-for-me/196782/34

What happens is that client checks new version, gets back nice "23.05-SNAPSHOT" and release tag.
The release tag intended is never posted to server, so server generates best image it can, which can TOCTOU being different than initially suggested.
It needs API enhancement to build for particular release tag and client enhancement to actually post release tag intended.
Thank you for considering @aparcar

pko76, 2024-06-09:
Neither auc nor Luci attended sysupgrade work perfectly OK. To upgrade 23.05-snapshot using Luci I have to use Advsnxe Mode, remove Or add some packages and than I am sometimes able to get the newrst version than is detected. Thet are cases when software package update indicates new base file, the attended upgrade does not recognise it. Add9ng or removing packages allows to get the newest version in spite of disolaying that the newest is already installed. It needs some more development.

pko76, 2024-06-16:
Is there any roadmap/hope to fix it?

(LuCI Attended Sysupgrade does not work neither)
another EXAMPLE:

root@OpenWrt:~# auc -f
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05-SNAPSHOT r23900-d8415fa270 on ipq40xx/generic (zte,mf286d)
Available: 23.05-SNAPSHOT r23900-d8415fa270
Requesting package lists...
Are you sure you want to continue the upgrade process? [N/y] y
Current Target: "ipq40xx/generic"
Current Architecture: "arm"
Current Revision: "r23898-d0620706f0"
Default Packages: (...)
Available Profiles:
(...)
zte_mf286d:
    ZTE MF286D
    Packages: kmod-usb-net-qmi-wwan kmod-usb-serial-option uqmi
    hasImageMetadata: 1
    SupportedDevices: zte,mf286d
Error: Received incorrect version r23898-d0620706f0 (requested r23900-d8415fa270)
Bad message (74)

GoetzGoerisch, 2024-06-28:
solved with [owut](https://github.com/efahl/owut/), the replacement for `auc` going forward.

Thank you @efahl ! ❤️

GoetzGoerisch, 2024-07-11:
auc is failing on 23.05-SNAPSHOT currently

pko76, 2024-07-11:
Confirmed...

auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05-SNAPSHOT r23987-d53f1caeb0 on ipq40xx/generic (zte,mf286d)
WARNING: cannot determing currently running branch.
Invalid argument (22)

pko76, 2024-09-25:
Why the r24106 is not available for MF286D?

auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05-SNAPSHOT r24104-a2a67804fc on ipq40xx/generic (zte,mf286d)
Available: 23.05-SNAPSHOT r24100-66055df3e0
Requesting package lists...

aparcar, 2024-09-25:
Seems to work with the staging firmware-selector, can't test this right now on real devices sorry...

https://firmware-selector.staging.openwrt.org/?version=23.05-SNAPSHOT&target=bmips%2Fbcm6368&id=actiontec_r1000h

pko76, 2024-09-25:
Right, the 23.05.5 is avaialable on the r24106, and the snapshot is still on 
23.05-SNAPSHOT (r24104-a2a67804fc)

pko76, 2024-09-25:
Right, 
the 23.05.5 is available as 23.05.5 (r24106-10cc5fcd00)
and the 23.05-SNAPSHOT is available as 23.05-SNAPSHOT (r24104-a2a67804fc)

closed 2025-03-26 by GoetzGoerisch