openwrt/asu #855: Error building the firmware image (SNAPSHOT ath79/generic/WNDRMAC v2)

lorand-horvath, 2024-06-17:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r26620-6b03050192",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/24.148.86114~d174bb8",
    "packages": {
        "libc": "1.2.5-r4",
        "wpad-basic-mbedtls": "2024.03.09~695277a5-r1",
        "busybox": "1.36.1-r1",
        "swconfig": "12",
        "procd-ujail": "2024.03.30~946552a7-r1",
        "base-files": "1602~6b03050192",
        "libustream-mbedtls": "2024.04.19~524a76e5-r1",
        "firewall4": "2024.05.21~4c01d1eb-r1",
        "uboot-envtools": "2024.01-r3",
        "kmod-usb-ohci": "6.1.92-r1",
        "dnsmasq": "2.90-r2",
        "procd": "2024.03.30~946552a7-r1",
        "kmod-leds-reset": "6.1.92-r1",
        "kmod-usb2": "6.1.92-r1",
        "odhcp6c": "2023.05.12~bcd28363-r20",
        "kmod-ath9k": "6.1.92.6.6.15-r2",
        "uci": "2023.08.10~5781664d-r1",
        "dropbear": "2024.85-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-owl-loader": "6.1.92.6.6.15-r2",
        "procd-seccomp": "2024.03.30~946552a7-r1",
        "urandom-seed": "3",
        "ppp": "2.4.9_git20210104-r5",
        "kmod-gpio-button-hotplug": "6.1.92-r3",
        "logd": "2024.04.26~85f10530-r1",
        "luci-app-attendedsysupgrade": "24.148.86114~d174bb8",
        "auc": "0.3.2-r1",
        "ca-bundle": "20240203-r1",
        "luci": "23.051.66410~a505bb1",
        "kmod-switch-rtl8366s": "6.1.92-r1",
        "kmod-usb-ledtrig-usbport": "6.1.92-r1",
        "kernel": "6.1.92~3de2e6158397f652c6826e71225e501b-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.4.9_git20210104-r5"
    },
    "profile": "netgear,wndrmac-v2",
    "target": "ath79/generic",
    "version": "SNAPSHOT",
    "diff_packages": true,
    "filesystem": "squashfs"
}

Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/snapshots/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/ath79/generic/kmods/6.1.92-1-3de2e6158397f652c6826e71225e501b/Packages.gz
Updated list of available packages in /builder/build_dir/target-mips_24kc_musl/root-ath79/../../../../builder/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/ath79/generic/kmods/6.1.92-1-3de2e6158397f652c6826e71225e501b/Packages.sig
[... 52 lines cut ...]
 * 	kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1)
 * opkg_install_cmd: Cannot install package kmod-switch-rtl8366s.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1) for kmod-usb-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1) for kmod-usb-ledtrig-usbport
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb-ledtrig-usbport:
 * 	kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1)
 * opkg_install_cmd: Cannot install package kmod-usb-ledtrig-usbport.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1) for kmod-usb-ohci
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1) for kmod-usb2
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-usb2:
 * 	kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1)
 * opkg_install_cmd: Cannot install package kmod-usb2.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1) for kmod-lib-crc-ccitt
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ppp:
 * 	kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1)
 * opkg_install_cmd: Cannot install package ppp.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ppp-mod-pppoe:
 * 	kernel (= 6.6.33~184d29dbd3adf6671eeca14d585a7c1a-r1)
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
make[2]: *** [Makefile:220: package_install] Error 255
make[1]: *** [Makefile:161: _call_manifest] Error 2
make: *** [Makefile:322: manifest] Error 2
```

aparcar, 2024-06-17:
Yea it's odd but I think something broke upstream...

lorand-horvath, 2024-06-17:
Looks like https://firmware-selector.openwrt.org has been fixed - I've just got a new build 26657 for ath79/generic (this is supposed to be kernel 6.6).

LuCI ASU still showing the same error as posted above...

closed 2024-07-11 by lorand-horvath