openwrt/asu #859: Error building the firmware image

Lyapsus, 2024-06-18:
Server response: Error: Received incorrect version r26684-7ccd7dac7e (requested r26340-9bdaebaff3)

```
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r26340-9bdaebaff3",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/24.168.60561~8860ca0",
    "packages": {
        "opkg": "2022.02.24~d038e5b6-r2",
        "wpad-basic-mbedtls": "2024.03.09~695277a5-r1",
        "kmod-usb-core": "6.6.30-r1",
        "libpthread": "1.2.5-r4",
        "busybox": "1.36.1-r1",
        "kmod-usb-net-qmi-wwan": "6.6.30-r1",
        "kmod-nft-fib": "6.6.30-r1",
        "kmod-nfnetlink": "6.6.30-r1",
        "kmod-nf-reject6": "6.6.30-r1",
        "kmod-usb-dwc3": "6.6.30-r1",
        "kmod-nf-flow": "6.6.30-r1",
        "kmod-lib-crc-ccitt": "6.6.30-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-pppoe": "6.6.30-r1",
        "ipq-wifi-zte_mf287plus": "2024.04.26~644ba9ea-r1",
        "kmod-pppox": "6.6.30-r1",
        "kmod-nf-reject": "6.6.30-r1",
        "procd-ujail": "2024.03.30~946552a7-r1",
        "base-files": "1591~9bdaebaff3",
        "kmod-nf-nat": "6.6.30-r1",
        "kmod-crypto-crc32c": "6.6.30-r1",
        "libustream-mbedtls": "2024.04.19~524a76e5-r1",
        "kmod-usb-dwc3-qcom": "6.6.30-r1",
        "firewall4": "2023.11.03~698a5335-r1",
        "uboot-envtools": "2024.01-r2",
        "dnsmasq": "2.90-r2",
        "procd": "2024.03.30~946552a7-r1",
        "kmod-crypto-acompress": "6.6.30-r1",
        "kmod-lib-crc32c": "6.6.30-r1",
        "kmod-nft-nat": "6.6.30-r1",
        "kmod-usb-serial-option": "6.6.30-r1",
        "kmod-usb3": "6.6.30-r1",
        "libmbedtls": "3.6.0-r1",
        "libmnl": "1.0.5-r1",
        "jansson": "2.14-r3",
        "odhcp6c": "2023.05.12~bcd28363-r20",
        "uclient-fetch": "2024.04.19~e8780fa7-r1",
        "uci": "2023.08.10~5781664d-r1",
        "ath10k-firmware-qca4019-ct": "2020.11.08-r1",
        "luci-ssl": "23.035.26083~7550ad6",
        "dropbear": "2024.85-r1",
        "libnftnl": "1.2.6-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "ath10k-board-qca4019": "20240220-r1",
        "procd-seccomp": "2024.03.30~946552a7-r1",
        "libjson-script": "2024.03.29~eb9bcb64-r1",
        "kmod-nf-log": "6.6.30-r1",
        "urandom-seed": "3",
        "ppp": "2.4.9_git20210104-r5",
        "kmod-leds-gpio": "6.6.30-r1",
        "kmod-gpio-button-hotplug": "6.6.30-r3",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-nf-log6": "6.6.30-r1",
        "luci-app-attendedsysupgrade": "24.168.60561~8860ca0",
[... 348 lines cut ...]
    Packages: kmod-usb-net-qmi-wwan kmod-usb-serial-option uqmi ipq-wifi-zte_mf287plus
    hasImageMetadata: 1
    SupportedDevices: zte,mf287plus
zte_mf287:
    ZTE MF287
    Packages: kmod-usb-net-qmi-wwan kmod-usb-serial-option uqmi ipq-wifi-zte_mf287
    hasImageMetadata: 1
    SupportedDevices: zte,mf287
zte_mf287pro:
    ZTE MF287Pro
    Packages: kmod-usb-net-qmi-wwan kmod-usb-serial-option uqmi ipq-wifi-zte_mf287plus
    hasImageMetadata: 1
    SupportedDevices: zte,mf287pro
zte_mf289f:
    ZTE MF289F
    Packages: kmod-usb-net-qmi-wwan kmod-usb-serial-option uqmi ath10k-firmware-qca9984-ct
    hasImageMetadata: 1
    SupportedDevices: zte,mf289f
zyxel_nbg6617:
    ZyXEL NBG6617
    Packages: kmod-usb-ledtrig-usbport
    hasImageMetadata: 1
    SupportedDevices: zyxel,nbg6617

```

dannil, 2024-06-18:
https://github.com/openwrt/asu/issues/853

Lyapsus, 2024-06-18:
nvm it’s bricked now