openwrt/asu #862: Requesting an image for raspberrypi,3-model-b gives an rpi-2 image

DanaGoyette, 2024-06-21:
When I attempted attendedsysupgrade on my Raspberry Pi 3 Model B, the server built me an image for the rpi-2 profile.  Is that safe to flash, or is the server building something that won't work?  It seems like there is an rpi-3 server profile, so I'd expect it to use that.

The image request:
```json
{
    "profile": "raspberrypi,3-model-b",
    "target": "bcm27xx/bcm2709",
    "version": "23.05.3",
    "packages": [
        "atop",
        "auc",
        "base-files",
        "bash",
        "bcm27xx-gpu-fw",
        "bcm27xx-userland",
        "brcmfmac-nvram-43430-sdio",
        "brcmfmac-nvram-43455-sdio",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "cypress-firmware-43430-sdio",
        "cypress-firmware-43455-sdio",
        "dnsmasq",
        "dropbear",
        "dump1090",
        "e2fsprogs",
        "ethtool",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "htop",
        "iperf3",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-brcmfmac",
        "kmod-crypto-acompress",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-fs-vfat",
        "kmod-hid",
        "kmod-hid-generic",
        "kmod-hwmon-raspberrypi",
        "kmod-input-core",
        "kmod-input-evdev",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-ppp",
        "kmod-pppoe",
[... 805 lines cut ...]
    "target": "bcm27xx/bcm2709",
    "titles": [
        {
            "model": "2B/2B 1.2",
            "variant": "(32bit)",
            "vendor": "Raspberry Pi"
        },
        {
            "model": "3B/3B+/CM3",
            "variant": "(32bit)",
            "vendor": "Raspberry Pi"
        },
        {
            "model": "4B/400/CM4",
            "variant": "(32bit)",
            "vendor": "Raspberry Pi"
        }
    ],
    "version_code": "r23809-234f1a2efa",
    "version_number": "23.05.3"
}
```

[stderr.txt](https://github.com/user-attachments/files/15930691/stderr.txt)
[stdout.txt](https://github.com/user-attachments/files/15930688/stdout.txt)

aparcar, 2024-06-21:
I think we had that before where a single image is working on multiple pi generations and both ends get confused on what they are and what they build.

Is the image working? If you look at the titles at the end of the json response, it also includes a raspberry 3

aparcar, 2024-06-21:
I think this is an issue if it silently downgrade a 64Bit system to 32Bit. @dangowrt do you know ow more about this?

DanaGoyette, 2024-06-23:
I didn't test the image, so I left it sitting on the old version.  That Pi 3 is only really in use to run an RTL-SDR, so it's not critical for it to be up to date.

DanaGoyette, 2024-07-26:
I finally got around to trying the image, and it seems to boot just fine; the pi3 one was armv7l, according to a pre-upgrade `uname -a`.  I wonder how much difference there actually is between the two images?

Before: `Linux pi3 5.15.137 #0 SMP Tue Nov 14 13:38:11 2023 armv7l GNU/Linux`
After: `Linux pi3 5.15.162 #0 SMP Mon Jul 15 22:14:18 2024 armv7l GNU/Linux`

If using a pi2 image doesn't create any real differences in the filesystem or kernel, then the only issue here is just that it's a bit confusing.