openwrt/asu #876: NanoPI R2S Unsupported profile: friendlyelec,nanopi-r2

donbowman, 2024-07-02:
from https://openwrt.org/toh/friendlyarm/nanopi_r2s, there is a version avail of 23.05.03, but the below error comes out when using the stock friendlyelec firmware (which is also sort of 23.05.03)

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23630-842932a63d",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "friendlyelec,nanopi-r2",
    "target": "rockchip/armv8",
    "version": "23.05.3",
    "packages": [
        "adblock",
        "aria2",
        "attr",
        "auc",
        "avahi-dbus-daemon",
        "base-files",
        "bash",
        "batctl-default",
        "bind-dig",
        "bind-libs",
        "blkdiscard",
        "blkid",
        "block-mount",
        "blockd",
        "brcmfmac-firmware-usb",
        "bsdiff",
        "bspatch",
        "btrfs-progs",
        "busybox",
        "bzip2",
        "ca-bundle",
        "ca-certificates",
        "cfdisk",
        "cgi-io",
        "cgroupfs-mount",
        "chat",
        "collectd",
        "collectd-mod-cpu",
        "collectd-mod-cpufreq",
        "collectd-mod-df",
        "collectd-mod-interface",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-memory",
        "collectd-mod-network",
        "collectd-mod-rrdtool",
        "collectd-mod-thermal",
        "collectd-mod-uptime",
        "collectd-mod-wireless",
        "comgt",
        "coremark",
        "coreutils",
        "coreutils-base64",
        "coreutils-basename",
        "coreutils-cat",
        "coreutils-chgrp",
        "coreutils-chmod",
        "coreutils-chown",
        "coreutils-chroot",
        "coreutils-cp",
        "coreutils-cut",
        "coreutils-date",
        "coreutils-dd",
        "coreutils-df",
        "coreutils-dirname",
        "coreutils-du",
[... 1009 lines cut ...]
        "uqmi",
        "urandom-seed",
        "urngd",
        "usb-modeswitch-official",
        "usbids",
        "usbutils",
        "usign",
        "uuidgen",
        "vim-full",
        "vsftpd",
        "watchcat",
        "wget-ssl",
        "wireless-regdb",
        "wpad-mini",
        "wsdd2",
        "wwan",
        "xtables-nft",
        "xz",
        "xz-utils",
        "zlib"
    ],
    "diff_packages": true,
    "client": "luci/git-23.339.51123-138595a"
}
```

Darin755, 2024-07-23:
I do not think friendlyelec is supported by OpenWRT

donbowman, 2024-07-24:
this is the upstream openwrt i think.

Darin755, 2024-07-24:
The profile is friendlyelec,nanopi-r2 which is not openwrt. OpenWRT does not officially support the device.