openwrt/asu #877: Error building the firmware image Server response: Error: Impossible package selection

James62370, 2024-07-03:
Please report the error message and request
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r26733-2b6772c82c",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/24.171.68415~188de6f",
    "packages": {
        "iwinfo": "2024.03.23~79a96150-r1",
        "libopenssl": "3.0.14-r2",
        "libc": "1.2.5-r4",
        "iw": "6.9-r1",
        "busybox": "1.36.1-r1",
        "ipq-wifi-dynalink_dl-wrx36": "2024.04.26~644ba9ea-r1",
        "nano": "8.0-r1",
        "kmod-usb-dwc3": "6.6.34-r1",
        "ucode-mod-uloop": "2024.06.18~4dd98370-r1",
        "ucode-mod-ubus": "2024.06.18~4dd98370-r1",
        "kmod-phy-aquantia": "6.6.34-r1",
        "procd-ujail": "2024.03.30~946552a7-r1",
        "base-files": "1606~2b6772c82c",
        "ath11k-firmware-ipq8074": "2024.03.14~795809c7-r2",
        "ucode-mod-uci": "2024.06.18~4dd98370-r1",
        "libustream-mbedtls": "2024.04.19~524a76e5-r1",
        "kmod-usb-dwc3-qcom": "6.6.34-r1",
        "libsmartcols": "2.40.1-r1",
        "firewall4": "2024.05.21~4c01d1eb-r1",
        "uboot-envtools": "2024.01-r3",
        "dnsmasq": "2.90-r2",
        "procd": "2024.03.30~946552a7-r1",
        "ucode-mod-math": "2024.06.18~4dd98370-r1",
        "kmod-usb3": "6.6.34-r1",
        "libopenssl-legacy": "3.0.14-r2",
        "odhcp6c": "2023.05.12~bcd28363-r20",
        "uci": "2023.08.10~5781664d-r1",
        "ucode-mod-fs": "2024.06.18~4dd98370-r1",
        "kmod-fs-ext4": "6.6.34-r1",
        "losetup": "2.40.1-r1",
        "dropbear": "2024.85-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "procd-seccomp": "2024.03.30~946552a7-r1",
        "ucode-mod-nl80211": "2024.06.18~4dd98370-r1",
        "ucode": "2024.06.18~4dd98370-r1",
        "urandom-seed": "3",
        "ppp": "2.4.9_git20210104-r5",
        "kmod-leds-gpio": "6.6.34-r1",
        "kmod-gpio-button-hotplug": "6.6.34-r3",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-qca-nss-dp": "6.6.34.2024.04.16~5bf8b91e-r1",
        "luci-app-attendedsysupgrade": "24.171.68415~188de6f",
        "e2fsprogs": "1.47.0-r2",
        "libopenssl-conf": "3.0.14-r2",
        "auc": "0.3.2-r1",
        "wpad-mesh-openssl": "2024.03.09~695277a5-r1",
        "ucode-mod-rtnl": "2024.06.18~4dd98370-r1",
        "ca-bundle": "20240203-r1",
        "luci": "23.051.66410~a505bb1",
        "kmod-ath11k-ahb": "6.6.34.6.6.15-r2",
        "kernel": "6.6.34~02fea4c5a81de007604bc46d7cf9bac7-r1",
        "libucode": "2024.06.18~4dd98370-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.4.9_git20210104-r5"
    },
    "profile": "dynalink,dl-wrx36",
    "target": "qualcommax/ipq807x",
    "version": "SNAPSHOT",
    "diff_packages": true,
    "filesystem": "squashfs"
}
STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.6.35-1-f37c2fdbdb753b455ef8cac06b859604/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/kmods/6.6.35-1-f37c2fdbdb753b455ef8cac06b859604/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-qualcommax/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package auc.
make[2]: *** [Makefile:220: package_install] Error 255
make[1]: *** [Makefile:161: _call_manifest] Error 2
make: *** [Makefile:322: manifest] Error 2

aparcar, 2024-07-03:
Sorry auc was removed via owut

James62370, 2024-07-03:
Thank you for your reply.  Can you advise if this is a permanent change or
just a temporary removal?

Appreciate your assistance in this matter.

SpandrelPanel, 2024-07-15:
@aparcar Could you explain further - is auc deprecated in favour of owut? Is it possible to migrate from auc to owut or do I need to do a sysupgrade and reinstall all extra packages?

aparcar, 2024-07-15:
Yes it's deprecated but there is a server issue right now. I'm pls ing to fix and update the server tomorrow