openwrt/asu #885: Build is failing with: Pseudo file "dev" exists in source filesystem ....

zazizou, 2024-07-09:
Hi 

I have the build failing with this error message:

`Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/dev".
Ignoring, exclude it (-e/-ef) to override.`

The target is the *linksys wrt32x*.

Could you please help solve this ?

Regards

zazizou, 2024-07-09:
Here is the full error log:

`Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/dev".
Ignoring, exclude it (-e/-ef) to override.
make[3]: *** No rule to make target 'linksys_wrt32x-initramfs-images', needed by '/builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/tmp/openwrt-016ac06e0ad9-mvebu-cortexa9-linksys_wrt32x-squashfs-sysupgrade.bin'.  Stop.
make[2]: *** [Makefile:247: build_image] Error 2
make[1]: *** [Makefile:153: _call_image] Error 2
make: *** [Makefile:310: image] Error 2
`

zazizou, 2024-07-10:
I did not do anything but the build passes today.

Maybe this kind of issue should be pinned as a common build issue - telling people to just try again.

pcartwright81, 2024-07-11:
Same error on a different target.  Tried for 2 days this the same error.

opty77, 2025-05-02:
> `Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/dev". Ignoring, exclude it (-e/-ef) to override.`

https://github.com/openwrt/openwrt/issues/18684#issuecomment-2847249532

lucky-coder-21, 2025-05-27:
Hi I'm experiencing the same issue in my version of the build. What do i do?

> `Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/dev".
Ignoring, exclude it (-e/-ef) to override.`