openwrt/asu #894: ASU not working

senposage, 2024-07-18:
as of 7/17 ASU has stopped working attempting to perform a upgrade just results in nothing happening 
visting the firmware selector webpage and requesting a build also doesn't work

lorand-horvath, 2024-07-18:
@aparcar firmware-selector.openwrt.org also doesn't work. Can you please fix it?

aparcar, 2024-07-18:
https://github.com/openwrt/asu/issues/663

cherinyy, 2024-07-18:
https://github.com/openwrt/asu/issues/663#issuecomment-2235843620

> A new round, I'll have to reconfigure multiple things, server is down until further notice, should be done by tomorrow.
> 
> Sorry for the inconvenience

tlaurion, 2024-07-18:
Would suggest that this is reported in newer version of luci app, to give context. Maybe even provide a crafted url for users to click on to post in openwrt version + model for easier bug report and users to see actual duplicates of issues to reduce your maintainer burden.

aparcar, 2024-07-18:
That would be awesome! Do you mind implementing something like that?

tlaurion, 2024-07-18:
> That would be awesome! Do you mind implementing something like that?

Really not my expertise @aparcar :( Just thought it would be a great idea and lower your burden of maintenance long term.

hardcorp74, 2024-07-19:
build firmware works, but cann't download it now

KeepingItCuttingEdge, 2024-07-20:
Same!  Asking for the wrong snapshop version that was built to download.

[image]